blueman-project/blueman #273: Cannot set iPhone as NAP with bluez-4 and udhcpc

juanitotc, 2015-05-01:
Note that I can set an iPhone as a network access point with bluez-5 and udhcpc.

After scanning for the iPhone, pairing with it and setting it as trusted, I cannot set it as a NAP.

Whereas with bluez-5, two pop-ups showed blueman attempting to make the connection and then that the connection had succeeded, with bluez-4 there are no pop-ups and blueman times out.

$ blueman-applet
blueman-applet version 1.99.alpha2 starting
/usr/local/lib/python2.7/site-packages/gi/overrides/__init__.py:175: Warning: The property GSettings:sch ema is deprecated and shouldn't be used anymore. It will be removed in a future version.
  return super_init_func(self, **new_kwargs)
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:60)
['TransferService', 'StatusIcon', 'StandardItems', 'ShowConnected', 'SerialManager', 'RecentConns', 'Pow erManager', 'PPPSupport', 'Networking', 'NetUsage', 'NMPANSupport', 'NMDUNSupport', 'Menu', 'KillSwitch' , 'Headset', 'GameControllerWakelock', 'ExitItem', 'DiscvManager', 'DhcpClient', 'DBusService', 'AuthAge nt', 'AppIndicator'] 
_________
get_interface_version (/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 4 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
[... 1176 lines cut ...]
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:161)
list: device_prop_ch Connected 0 /org/bluez/7693/hci0/dev_54_E4_3A_22_48_E9 () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 0 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating down 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:184)
stopping monitor (not connected) 
_________
unset_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0xb5f0edec (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x8ec5658)> 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
j-frankish iPhone 
_________
unset_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0xb5eb3f54 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x8ec57f0)> 
_________
fail (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:121)
fail (DBusException(dbus.String(u'Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.'),),) 
/usr/local/lib/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 6391 was not found when attempting to remove it
  return fn(*args, **kwargs)

infirit, 2015-05-01:
Can you try with the GameControllerWakelock and SerialManager plugins disabled?

juanitotc, 2015-05-01:
Sorry, but that didn't change anything :(

cschramm, 2015-05-11:
Please use dbus-monitor to check what's going on on the system bus when trying to activate the service.

juanitotc, 2015-05-11:
starting the monitor after pairing, I get:

`signal sender=org.freedesktop.DBus -> dest=:1.13 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.13"
signal sender=:1.0 -> dest=(null destination) serial=390 path=/org/bluez/7670/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Device; member=PropertyChanged
   string "Connected"
   variant       boolean false`

cschramm, 2015-05-11:
You need to run it with root privileges.

cschramm, 2015-05-11:
Expecting issues on the bluetoothd side you can also check its logs for anything interesting when clicking on the NAP service.

juanitotc, 2015-05-11:
The last parts of bluetoothd and dbus-monitor output (blueman-{applet, manager} output in pastebin above.

sudo dbus-monitor --system

...
signal sender=:1.0 -> dest=(null destination) serial=83 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Device; member=PropertyChanged
   string "UUIDs"
   variant       array [
         string "00000000-deca-fade-deca-deafdecacafe"
         string "00001000-0000-1000-8000-00805f9b34fb"
         string "0000110a-0000-1000-8000-00805f9b34fb"
         string "0000110c-0000-1000-8000-00805f9b34fb"
         string "0000110e-0000-1000-8000-00805f9b34fb"
         string "00001116-0000-1000-8000-00805f9b34fb"
         string "0000111f-0000-1000-8000-00805f9b34fb"
         string "0000112f-0000-1000-8000-00805f9b34fb"
         string "00001132-0000-1000-8000-00805f9b34fb"
         string "00001200-0000-1000-8000-00805f9b34fb"
      ]
signal sender=:1.0 -> dest=(null destination) serial=95 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.AudioSource; member=PropertyChanged
   string "State"
   variant       string "connecting"
signal sender=:1.0 -> dest=(null destination) serial=99 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.HandsfreeGateway; member=PropertyChanged
   string "State"
   variant       string "connecting"
signal sender=:1.0 -> dest=(null destination) serial=101 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Device; member=PropertyChanged
   string "Trusted"
   variant       boolean true
signal sender=:1.0 -> dest=(null destination) serial=109 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Control; member=Connected
signal sender=:1.0 -> dest=(null destination) serial=110 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Control; member=PropertyChanged
   string "Connected"
   variant       boolean true
signal sender=:1.0 -> dest=(null destination) serial=111 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.HandsfreeGateway; member=PropertyChanged
   string "State"
   variant       string "disconnected"
signal sender=:1.0 -> dest=(null destination) serial=112 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.AudioSource; member=PropertyChanged
   string "State"
   variant       string "disconnected"
signal sender=:1.0 -> dest=(null destination) serial=113 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Control; member=Disconnected
signal sender=:1.0 -> dest=(null destination) serial=114 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Control; member=PropertyChanged
   string "Connected"
   variant       boolean false
signal sender=:1.0 -> dest=(null destination) serial=127 path=/org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9; interface=org.bluez.Device; member=PropertyChanged
   string "Connected"
   variant       boolean false

sudo /usr/local/sbin/bluetoothd -nd

...
bluetoothd[4796]: src/adapter.c:adapter_create_device() 54:E4:3A:22:48:E9
bluetoothd[4796]: src/device.c:device_create() Creating device /org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[4796]: src/device.c:btd_device_ref() 0xb7be9870: ref=1
bluetoothd[4796]: src/device.c:device_set_temporary() temporary 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_create_bonding() hci0 bdaddr 54:E4:3A:22:48:E9 io_cap 0x04
bluetoothd[4796]: src/device.c:bonding_request_new() Requesting bonding for 54:E4:3A:22:48:E9
bluetoothd[4796]: src/device.c:bonding_request_new() Temporary agent registered for 54:E4:3A:22:48:E9 at :1.1:/org/blueman/agent/temp/54E43A2248E9
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() cond 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() Received 43 bytes from management socket
bluetoothd[4796]: plugins/mgmtops.c:mgmt_device_connected() hci0 device 54:E4:3A:22:48:E9 connected eir_len 24
bluetoothd[4796]: src/adapter.c:adapter_get_device() 54:E4:3A:22:48:E9
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() cond 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() Received 18 bytes from management socket
bluetoothd[4796]: plugins/mgmtops.c:mgmt_user_confirm_request() hci0 54:E4:3A:22:48:E9 confirm_hint 0
bluetoothd[4796]: src/adapter.c:adapter_get_device() 54:E4:3A:22:48:E9
bluetoothd[4796]: src/device.c:device_request_authentication() Requesting agent authentication for 54:E4:3A:22:48:E9
bluetoothd[4796]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.1, path=/org/blueman/agent/temp/54E43A2248E9, passkey=203851
bluetoothd[4796]: plugins/mgmtops.c:mgmt_confirm_reply() index 0 addr 54:E4:3A:22:48:E9 success 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() cond 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() Received 16 bytes from management socket
bluetoothd[4796]: plugins/mgmtops.c:mgmt_cmd_complete() 
[... 42 lines cut ...]
bluetoothd[4796]: audio/manager.c:gateway_auth_cb() Accepted AG connection from 54:E4:3A:22:48:E9 for /org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[4796]: audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 54:E4:3A:22:48:E9
bluetoothd[4796]: audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=672
bluetoothd[4796]: audio/avctp.c:avctp_connect_cb() AVCTP: connected to 54:E4:3A:22:48:E9
bluetoothd[4796]: Can't open input device: No such file or directory (2)
bluetoothd[4796]: AVRCP: failed to init uinput for 54:E4:3A:22:48:E9
bluetoothd[4796]: audio/avctp.c:avctp_set_state() AVCTP Connected
[... cut ...]
bluetoothd[4796]: audio/avctp.c:session_cb() AVCTP session 0xb7bfd000 got disconnected
bluetoothd[4796]: audio/avctp.c:avctp_set_state() AVCTP Disconnected
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() cond 1
bluetoothd[4796]: plugins/mgmtops.c:mgmt_event() Received 14 bytes from management socket
bluetoothd[4796]: plugins/mgmtops.c:mgmt_device_disconnected() hci0 device 54:E4:3A:22:48:E9 disconnected
bluetoothd[4796]: src/event.c:btd_event_disconn_complete() 
bluetoothd[4796]: src/adapter.c:adapter_remove_connection()

cschramm, 2015-05-13:
This is strange. There should be a dbus call to the Connect method on interface org.bluez.Network1 - the one that times out at the end of your blueman output. Are you sure you tried to connect the service while capturing?

juanitotc, 2015-05-13:
Absolutely sure -  I tried several times.

I just updated blueman on tinycorelinux, you can try yourself if you'd like ;)

cschramm, 2015-05-13:
So when you click on the NAP service no messages appear in either dbus or bluetoothd and you only get the dbus timeout in blueman? :open_mouth:

juanitotc, 2015-05-13:
Affirmative (I double-checked just now).

Note that, apart from some minor version differences in gtk components, the main difference between the bluez-4.101 system and the bluez-5.28 system I'm using (on the same machine) is that the bluez-4 is x86 and the bluez-5 is x86_64, but I doubt that changes anything.

cschramm, 2015-05-13:
Can you check the session bus? Maybe the message is completely misdirected.

You can try to issue the call manually with something like `dbus-send --print-reply --system /org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9 org.bluez.Network1.Connect string:00001116-0000-1000-8000-00805f9b34fb`.

juanitotc, 2015-05-13:
Here's what I get:

`$ dbus-send --print-reply --system /org/bluez/4796/hci0/dev_54_E4_3A_22_48_E9 org.bluez.Ne
twork1.Connect string:00001116-0000-1000-8000-00805f9b34fb
Error org.freedesktop.DBus.Error.UnknownMethod: Method "Connect" with signature "s" on interface "org.bluez.Network1" doesn't exist`

and...

$ dbus-monitor --session
signal sender=org.freedesktop.DBus -> dest=:1.9 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.9"
method call sender=:1.9 -> dest=org.freedesktop.DBus serial=3 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "eavesdrop=true,type='method_call'"
method call sender=:1.9 -> dest=org.freedesktop.DBus serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "eavesdrop=true,type='method_return'"
method call sender=:1.9 -> dest=org.freedesktop.DBus serial=5 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "eavesdrop=true,type='error'"
method call sender=:1.6 -> dest=org.blueman.Applet serial=29 path=/; interface=org.freedesktop.DBus.Introspectable; member=Introspect
method call sender=:1.6 -> dest=org.blueman.Applet serial=30 path=/; interface=org.freedesktop.DBus.Introspectable; member=Introspect
method return sender=:1.3 -> dest=:1.6 reply_serial=29
   string "<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node name="/">
  <interface name="org.blueman.Applet">
    <method name="CancelDeviceCreation">
      <arg direction="in"  type="s" name="adapter_path" />
      <arg direction="in"  type="s" name="address" />
    </method>
    <method name="SetPluginConfig">
      <arg direction="in"  type="s" name="plugin" />
      <arg direction="in"  type="b" name="value" />
    </method>
    <method name="connect_service">
      <arg direction="in"  type="o" name="object_path" />
      <arg direction="in"  type="s" name="uuid" />
    </method>
    <method name="CreateDevice">
      <arg direction="in"  type="s" name="adapter_path" />
      <arg direction="in"  type="s" name="address" />
      <arg direction="in"  type="b" name="pair" />
      <arg direction="in"  type="u" name="time" />
    </method>
    <method name="QueryAvailablePlugins">
      <arg direction="out" type="as" />
    </method>
    <method name="open_plugin_dialog">
    </method>
    <method name="DhcpClient">
      <arg direction="in"  type="s" name="interface" />
    </method>
    <method name="GetBluetoothStatus">
      <arg direction="out" type="b" />
    </method>
    <method name="DisconnectDevice">
      <arg direction="in"  type="o" name="obj_path" />
    </method>
    <method name="SetBluetoothStatus">
      <arg direction="in"  type="b" name="status" />
    </method>
    <method name="QueryPlugins">
      <arg direction="out" type="as" />
    </method>
    <method name="disconnect_service">
      <arg direction="in"  type="o" name="object_path" />
      <arg direction="in"  type="s" name="uuid" />
      <arg direction="in"  type="d" name="port" />
    </method>
    <signal name="BluetoothStatusChanged">
      <arg type="b" name="status" />
    </signal>
[... 339 lines cut ...]
    <method name="SetBluetoothStatus">
      <arg direction="in"  type="b" name="status" />
    </method>
    <method name="QueryPlugins">
      <arg direction="out" type="as" />
    </method>
    <method name="disconnect_service">
      <arg direction="in"  type="o" name="object_path" />
      <arg direction="in"  type="s" name="uuid" />
      <arg direction="in"  type="d" name="port" />
    </method>
    <signal name="BluetoothStatusChanged">
      <arg type="b" name="status" />
    </signal>
    <method name="SetTimeHint">
      <arg direction="in"  type="u" name="time" />
    </method>
  </interface>
  <interface name="org.freedesktop.DBus.Introspectable">
    <method name="Introspect">
      <arg direction="out" type="s" />
    </method>
  </interface>
</node>
"

cschramm, 2015-05-13:
Sorry, must be Network, without the 1, for BlueZ 4.

juanitotc, 2015-05-13:
The error is the same for Network or Network1 :(

cschramm, 2015-05-15:
OK, I had a closer look. It seems like we cannot make dbus-monitor show the bluez messages due to dbus security policies and dbus-send needs the parameter `--dest=org.bluez`. Please try to call the Connect method with dbus-send and see what's the result (`--print-reply`) and what happens in the bluetoothd log.

juanitotc, 2015-05-15:
Sorry, but:

`dbus-send --system --dest=org.bluez --print-reply /org/bluez/6069/hci0/dev_54_E4_3A_22_48_E9 org.bluez.Network.Connect string:00001116-0000-1000-8000-00805f9b34fb

Error org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "s" on interface "org.bluez.Network" doesn't exist`

The interface is definitely there:

`bluetoothd[6069]: src/device.c:device_probe_drivers() Probing drivers for 54:E4:3A:22:48:E9
bluetoothd[6069]: network/manager.c:network_probe() path /org/bluez/6069/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[6069]: src/device.c:btd_device_ref() 0xb90e4108: ref=3
bluetoothd[6069]: network/connection.c:create_peer() Registered interface org.bluez.Network on path /org/bluez/6069/hci0/dev_54_E4_3A_22_48_E9`

Is it that `org.bluez.Network.Connect` does not exist in the bluez-4 API?

cschramm, 2015-05-15:
:open_mouth: Try calling org.freedesktop.DBus.Introspectable.Introspect to check what's available on the object.

org.bluez.Network.Connect not being available would not explain the dbus timeout in blueman. It should reply with "doesn't exist" then as well.

juanitotc, 2015-05-15:
Ref:
`$ dbus-send --system --dest=org.bluez --print-reply /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9 org.bluez.Network.Connect

Error org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.`

It seems to be the "string:00001116-0000-1000-8000-00805f9b34fb" that is causing the problem?

As per bluetoothd, the serial interfaces get assigned 000011xx-0000-1000-8000-00805f9b34fb, but not the network interface:
`bluetoothd[5764]: src/device.c:device_probe_drivers() Probing drivers for 54:E4:3A:22:48:E9
bluetoothd[5764]: network/manager.c:network_probe() path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[5764]: src/device.c:btd_device_ref() 0xb8d00870: ref=3
bluetoothd[5764]: network/connection.c:create_peer() Registered interface org.bluez.Network on path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[5764]: serial/manager.c:serial_probe() path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9: 00001132-0000-1000-8000-00805f9b34fb
bluetoothd[5764]: serial/port.c:create_serial_device() Registered interface org.bluez.Serial on path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9
bluetoothd[5764]: serial/manager.c:serial_probe() path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9: 00000000-deca-fade-deca-deafdecacafe
bluetoothd[5764]: serial/manager.c:serial_probe() path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9: 0000112f-0000-1000-8000-00805f9b34fb
bluetoothd[5764]: serial/manager.c:serial_probe() path /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9: 0000111f-0000-1000-8000-00805f9b34fb`

Here's the Introspect stuff:
`$ dbus-send --system --dest=org.bluez --print-reply /org/bluez/5764/hci0/dev_54_E4_3A_22_48_E9 org.freedesktop.DBus.Introspectable.Introspect
method return sender=:1.2 -> dest=:1.9 reply_serial=2
   string "<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node>
        <interface name="org.freedesktop.DBus.Introspectable">
                <method name="Introspect">
                        <arg name="xml" type="s" direction="out"/>
                </method>
        </interface>
        <interface name="org.bluez.Device">
                <method name="GetProperties">
                        <arg name="properties" type="a{sv}" direction="out"/>
                </method>
                <method name="SetProperty">
                        <arg name="name" type="s" direction="in"/>
                        <arg name="value" type="v" direction="in"/>
                </method>
                <method name="DiscoverServices">
                        <arg name="pattern" type="s" direction="in"/>
                        <arg name="services" type="a{us}" direction="out"/>
                </method>
                <method name="CancelDiscovery"/>
                <method name="Disconnect"/>
                <signal name="PropertyChanged">
                        <arg name="name" type="s"/>
                        <arg name="value" type="v"/>
                </signal>
                <signal name="DisconnectRequested"/>
        </interface>
        <interface name="org.bluez.Network">
                <method name="Connect"/>
                <method name="Disconnect"/>
                <method name="GetProperties">
                        <arg name="properties" type="a{sv}" direction="out"/>
                </method>
                <signal name="PropertyChanged">
                        <arg name="name" type="s"/>
                        <arg name="value" type="v"/>
                </signal>
        </interface>
        <interface name="org.bluez.Serial">
                <method name="Connect">
                        <arg name="pattern" type="s" direction="in"/>
                        <arg name="tty" type="s" direction="out"/>
                </method>
                <method name="ConnectFD">
                        <arg name="pattern" type="s" direction="in"/>
                        <arg name="fd" type="s" direction="out"/>
                </method>
                <method name="Disconnect">
[... 38 lines cut ...]
                        <arg name="name" type="s"/>
                        <arg name="value" type="v"/>
                </signal>
        </interface>
        <interface name="org.bluez.Control">
                <method name="IsConnected">
                        <arg name="connected" type="b" direction="out"/>
                </method>
                <method name="GetProperties">
                        <arg name="properties" type="a{sv}" direction="out"/>
                </method>
                <method name="VolumeUp"/>
                <method name="VolumeDown"/>
                <signal name="Connected">
                        <annotation name="org.freedesktop.DBus.Deprecated" value="true"/>
                </signal>
                <signal name="Disconnected">
                        <annotation name="org.freedesktop.DBus.Deprecated" value="true"/>
                </signal>
                <signal name="PropertyChanged">
                        <arg name="name" type="s"/>
                        <arg name="value" type="v"/>
                </signal>
        </interface>
</node>`

cschramm, 2015-05-15:
OK... Something's completely wrong here. According to your introspection org.bluez.Network.Connect does not have any parameters. According to the BlueZ 4 docs and my experience it does.

Does Bluez' `test-network` script work?

Anything happening in bluetoothd when you get the "Did not receive a reply"?

juanitotc, 2015-05-15:
There's nothing at all from bluetoothd when I get the "Did not receive a reply" message.

`Traceback (most recent call last):
  File "./test-network", line 42, in <module>
    device = adapter.FindDevice(address)
  File "/usr/local/lib/python2.7/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/local/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.DoesNotExist: Does Not Exist`

I recompiled bluez-4.101 to double-check, but got the same result :(

cschramm, 2015-05-15:
:unamused: It's a bug in BlueZ 4.100 and 4.101. Patch is here: http://git.kernel.org/cgit/bluetooth/bluez.git/diff/network/connection.c?id=57170b311f1468330f4a9961dc0b3ac45f97bc13

closed 2015-05-15 by cschramm

juanitotc, 2015-05-16:
Arrrrrrrgh.

Sorry for all of the trouble caused, but I was unaware of the patch - udhcpc networking works now.
[image]