blueman-project/blueman #274: Error: GLib-GIO-ERROR: Settings schema 'org.blueman.general' is not installed

hlechner, 2015-05-03:
The last commit that I can use is: e698736a174658498d82e6304f97cc039c18b485 (Mar 5)

Here the error I'm facing on latest git version (abf2b30a9b82d705b73bb07f2d528c8d2de4c9a7):

```
/usr/bin/blueman-manager:54: Warning: The property GtkAlignment:left-padding is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.Builder.add_from_file(UI_PATH + "/manager-main.ui")
/usr/bin/blueman-manager:54: Warning: The property GtkAlignment:right-padding is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.Builder.add_from_file(UI_PATH + "/manager-main.ui")
/usr/bin/blueman-manager:54: Warning: The property GtkAlignment:xalign is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.Builder.add_from_file(UI_PATH + "/manager-main.ui")
/usr/bin/blueman-manager:54: Warning: The property GtkAlignment:xscale is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.Builder.add_from_file(UI_PATH + "/manager-main.ui")
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services'] 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:338)
1 
/usr/lib/python2.7/site-packages/blueman/gui/MessageArea.py:43: Warning: The property GtkMisc:xpad is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.icon.props.xpad = 4
/usr/lib/python2.7/site-packages/blueman/gui/MessageArea.py:54: Warning: The property GtkSettings:gtk-button-images is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.b_more.set_image(im)
blueman-manager version 1.99.alpha2 starting
Stale PID, overwriting
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:338)
2 
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:96)
org.bluez owner changed to  :1.1 
/usr/lib/python2.7/site-packages/gi/overrides/__init__.py:306: Warning: The property GSettings:schema is deprecated and shouldn't be used anymore. It will be removed in a future version.
  return super_init_func(self, **new_kwargs)

(blueman-manager:21001): GLib-GIO-ERROR **: Settings schema 'org.blueman.general' is not installed

Trace/breakpoint trap (core dumped)
```

System:
Arch Linux
Gnome 3.16.1

Thanks!

cschramm, 2015-05-03:
The gsettings schema needs to be installed. make install should do that four you.

hlechner, 2015-05-03:
I'm not sure if it should be this package:
```extra/gsettings-desktop-schemas 3.16.1-1 [installed]```

But the PKGBUILD from aur package already do the make install:
```
prepare() {
    cd "${srcdir}/${pkgname}"
    # Replace the python interpreter with python2
    for PY in apps/blueman-*
    do
        sed -i 's/python/python2/' "${PY}"
    done
    NOCONFIGURE=1 ./autogen.sh
}

build() {
    cd "${srcdir}/${pkgname}"
    export PYTHON=/usr/bin/python2
    export CYTHONEXEC=/usr/bin/cython2
    ./configure \
        --prefix=/usr \
        --sysconfdir=/etc \
        --libexecdir=/usr/lib/${_pkgname} \
        --with-gtk=3.0 \
        --disable-hal \
        --disable-sendto \
        --disable-static
    make
}

package() {
    cd "${srcdir}/${pkgname}"
    make DESTDIR="${pkgdir}" install
```

Should something be different?

hlechner, 2015-05-03:
I found a solution:

need to run it as root:
```/usr/bin/glib-compile-schemas /usr/share/glib-2.0/schemas/```

cschramm, 2015-05-03:
There's probably something wrong with the PKGBUILD or our make system then, so let's take a closer look. Do you have the data/Makefile from the build?

Mine includes
````make
GLIB_COMPILE_SCHEMAS = /usr/lib/x86_64-linux-gnu/glib-2.0/glib-compile-schemas

gsettings_SCHEMAS = org.blueman.gschema.xml

all-am: $(gsettings_SCHEMAS:.xml=.valid)

%.gschema.valid: %.gschema.xml $(gsettings__enum_file)
        $(AM_V_GEN) $(GLIB_COMPILE_SCHEMAS) --strict --dry-run $(addprefix --schema-file=,$(gsettings__enum_file)) --schema-file=$< && mkdir -p $(@D) && touch $@
````
which should run that command on make install.

infirit, 2015-05-03:
The distro's build system will run this in the build environment, most likely sandboxed or a jail of some sorts and the ```gschemas.compiled``` file will only contain the compiled blueman schemas. If this is included in the binary package it will overwrite the existing one with all the already installed schemas.

Normally distro's should pass ```--disable-schemas-compile``` to configure and run ```glib-compile-schemas``` in their pre/post-rm and post-install phases :smile:.

cschramm, 2015-05-03:
True (it's not that explicit in the Debian world due to triggers :wink: ). Then the PKGBUILD is probably missing the step in its post_install script (or some definition to invoke something similar to Debian triggers?).

infirit, 2015-05-03:
Most pkgbuilds I know of use an install file with something like the below.
```
post_install() {
  glib-compile-schemas /usr/share/glib-2.0/schemas
}

post_upgrade() {
  post_install $1
}
```

cschramm, 2015-05-07:
@flexiondotorg

closed 2015-05-07 by cschramm

Markismus, 2015-09-27:
@hlechner The problem still persists. Thanks for your  `sudo /usr/bin/glib-compile-schemas /usr/share/glib-2.0/schemas/`. It solved the problem for me.

Where could we change the PKGBUILD to include this command?

rwst, 2016-09-28:
Recompiling schemas had no effect for me (both 2.0.4/master on OpenSuSE/Xfce) but when I did `configure --prefix=/usr` and installed I got past that step. From this it seems `glib-schemas` does not always look by default into `/usr/local`, so it has to be told. I don't know glib, isn't there a user config where it can be told?

Secondly after passing this I get:
```
blueman-applet version 2.1.alpha1 starting
Stale PID, overwriting
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:61)
['Headset', 'AuthAgent', 'DBusService', 'DhcpClient', 'DiscvManager', 'ExitItem', 'GameControllerWakelock', 'KillSwitch', 'Menu', 'NetUsage', 'Networking', 'NMDUNSupport', 'NMPANSupport', 'PowerManager', 'PPPSupport', 'RecentConns', 'SerialManager', 'ShowConnected', 'StatusIcon', 'StandardItems', 'TransferService', 'AppIndicator'] 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:61)
Unable to load plugin module Headset
cannot import name raise_dbus_error 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:61)
Unable to load plugin module AppIndicator
 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:134)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:134)
Failed to load StatusIcon
'BluemanApplet' object has no attribute 'plugin_run_state_changed' 
```
The fact is puzzling because I know I could run blueman a week before on the same system modulo updates. Did they change glib?