blueman-project/blueman #275: Can send, but not receive files

juanitotc, 2015-05-06:
Using bluez-5.28.

Following #84 I created /usr/local/etc/dbus-1/system.d/obex-data-server.conf as the source package does not install the file.

Although I can send files to a windows laptop without problems, I cannot receive them.

On the windows machine I get:
`discovering services available on...
no services found`

I can see that dbus has started obex-data-server:
`$ ps aux | grep obex
 8062 tc       /usr/local/bin/obex-data-server --no-daemon`

..but:
`bluetoothd[7963]: src/adapter.c:trigger_passive_scanning() 
bluetoothd[7963]: src/adapter.c:connected_callback() hci0 device C4:46:19:E9:C1:EC connected eir_len 5
bluetoothd[7963]: src/adapter.c:dev_disconnected() Device C4:46:19:E9:C1:EC disconnected, reason 3
bluetoothd[7963]: src/adapter.c:adapter_remove_connection() 
bluetoothd[7963]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[7963]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr C4:46:19:E9:C1:EC type 0 status 0xe
bluetoothd[7963]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[7963]: src/device.c:device_bonding_failed() status 14
bluetoothd[7963]: src/adapter.c:resume_discovery()`

blueman-applet:
`__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_C4_46_19_E9_C1_EC 
_________
Destroy (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_C4_46_19_E9_C1_EC`

blueman-manager:
`update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:184)
stopping monitor (not connected) 
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:161)
list: device_prop_ch Connected 1 /org/bluez/hci0/dev_C4_46_19_E9_C1_EC () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 1 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PAULINE 
_________
monitor_power_levels (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:183)
starting monitor 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating up 
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:161)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_C4_46_19_E9_C1_EC () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 0 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PAULINE 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating down 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:184)
stopping monitor (not connected) `

plaaplaa, 2015-05-06:
1. obex-data-server can not connect to bluez sdp server, edit bluez sources
2. obexd is preventing obex-data-server to receive, so obexd needs to killed, edit blueman sources.
3. openopex, obex-data-server and blueman needs to be compile againts bluez 5.28 library.

How to fix, see:https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=757633#106

juanitotc, 2015-05-06:
Ref 1, what needs to be done with bluez sources, all I could see was some obscure debian permissions stuff (chmod(SDP_UNIX_PATH, S_IRUSR | S_IWUSR | S_IRGRP | S_IWGRP | S_IROTH | S_IWOTH)?
Ref 2, obexd was not running - I checked.
Ref 3, all of them were compiled against bluez-5.28.

cschramm, 2015-05-06:
No need to edit any sources.

Do you see /var/run/sdp when bluetoothd is running? You need to start it with --compat.

What are the permission mask and owner of /var/run/sdp? Is the user running obex-data-server able to access it?

Do you see "Couldn't connect to SDP server" in the output from obex-data-server (or syslog)?

cschramm, 2015-05-06:
You could also test the obexd branch. BlueZ 5 support should be ready. :smiley:

juanitotc, 2015-05-06:
so the obexd branch is for obex send and receive?

with or without obex-data-server?

cschramm, 2015-05-06:
The obexd branch ports receiving from obex-data-server to BlueZ' obexd. No more need for obex-data-server then.

juanitotc, 2015-05-07:
To close out the obex-data-server thing, /var/log/sdp is created as root:root, but as obex-data-server going to be phased out, let's leave that.

Using the obexd branch as of today with bluez-5.28.

Immediately on pairing the windows laptop installed 3-4 additional bluetooth drivers, so something changed for the good :)

Before attempting the transfer I ran xdg-user-dirs-update, but where does blueman expect transferred files to be copied to?

I'd say the problem is this:
`May  7 10:57:09 box daemon.debug bluetoothd[6033]: src/service.c:change_state() 0x1069200: device C4:46:19:E9:C1:EC profile Object Push state changed: disconnected -> connected (0)
May  7 10:57:09 box daemon.debug obexd[6059]: CONNECT(0x0), (null)(0xffffffff)
May  7 10:57:09 box daemon.debug obexd[6059]: CONNECT(0x0), (null)(0x0)
May  7 10:57:09 box daemon.err obexd[6059]: disconnected: Unicode conversion failed: Conversion from character set 'UTF-16BE' to 'UTF-8' is not supported`

..but it could be that blueman does not find the folder in which to place the received file?

Does the locale need to be set to en_US.UTF-8 or similar (it is set to C)? What is required to convert from 'UTF-16BE' to 'UTF-8'?

cschramm, 2015-05-07:
You can configure the destination folder with blueman-services.

Looks like your iconv lacks support for those character sets. You can check with iconv -l.

juanitotc, 2015-05-07:
Setting my locale to en_US.UTF-8 made things work :)

This being said, that makes several MB of unnecessary stuff and why would a file transfer need the locale to be set?

By default the received file went to the root of the filesystem ("/"), maybe it would be better if the default was $HOME or /tmp?

Anyway, very good - audio, file send/receive and udhcpc/networkmanager networking all work with bluez-5 :D

cschramm, 2015-05-07:
You're right. I think we're missing a sane default for shared-path in the obexd branch.

juanitotc, 2015-05-07:
..and the locale thing?

cschramm, 2015-05-07:
OBEX uses UCS-2 in headers. obexd uses glib to convert between UTF-32 and UTF-8. glib uses iconv and iconv somehow seems to depend on the locale. No idea why, but at least there's nothing we can do about that in blueman.

juanitotc, 2015-05-07:
Thanks for the reply.

Yeah - no idea why people feel the need to do that, gnome-terminal > 3.8 being another example...

infirit, 2015-05-07:
>I think we're missing a sane default for shared-path in the obexd branch.

Previously the path was set in ``blueman/applet/Transfer.py`` but this is [removed](https://github.com/blueman-project/blueman/pull/263/files#diff-f9af48d642100eb4d659c2a3b5ba2fbfL47). I think ``TransferService.py`` is a good place to move in to?

cschramm, 2015-05-07:
Yes, I agree. Either `on_load` or `_on_transfer_completed` (before `shutil.move`).

infirit, 2015-05-07:
Here is a [commit](https://github.com/infirit/blueman/commit/94d5646bd58e9620dbea5b6a458cd684e42586a8) that add the code in ``on_load``. I put it there because I want to make sure the user is informed when the shared-path does not exist before a transfer is even started.

cschramm, 2015-05-07:
`on_load` makes sense, yes. I missed the fact that there's a possible dialog.

I think we should not call it share directory in the message anymore as it's only for incoming files (initially browsable from a remote device). I also changed the wording in the services dialog.

Is it on purposed that you did not add the "directory does not exist" exception?

infirit, 2015-05-07:
>no idea why people feel the need to do that, gnome-terminal > 3.8 being another example...

My guess is that this has to do with the file-name. If we receive an unicode file-name there is no proper way to convert this to ASCII (C locale).

infirit, 2015-05-07:
>Is it on purposed that you did not add the "directory does not exist" exception?

Yes, if we raise an exception we do not load the plugin and blueman-services does not allow to change the path.

cschramm, 2015-05-08:
> I think we should not call it share directory in the message anymore as it's only for incoming files (initially browsable from a remote device).

For the same reason we could think of using another default now. Maybe the user's downloads directory?

infirit, 2015-05-08:
>Maybe the user's downloads directory?

Yep, ``GLib.UserDirectory.DIRECTORY_DOWNLOAD`` make more sense now.

cschramm, 2015-05-10:
BlueZ 4 support, shared-path default, wording and default changes arrived in #263. :smiley: 

Anything left in this thread other than #84?

juanitotc, 2015-05-10:
is obex-data-server required to test with bluez-4?

cschramm, 2015-05-10:
No

closed 2015-05-10 by cschramm

juanitotc, 2015-05-11:
works fine for me with bluez-4.101 and obexd-0.48

..and the file transferred appears in home - thanks