blueman-project/blueman #276: Not in Fedora 21 Repos

norpol, 2015-05-07:
Hi, does anyone knows why blueman is not in the Fedora repositories? Next release is 22, maybe it's possible to get it until there into the repos.

infirit, 2015-05-07:
You can use the git builds, https://copr.fedoraproject.org/coprs/heikoada/blueman-git/

When 2.0 is released I suspect then it will end up in the repositories but this is mostly up to the fedora folks.

norpol, 2015-05-07:
Thank's for your fast reply,
so it's nice to see that I'm not alone with 'receiving files doesn't work ATM but you still can send files via bluetooth'. I;d guess you can close the issue?

cschramm, 2015-05-07:
https://github.com/blueman-project/blueman/wiki/Packaged-versions#fedora

closed 2015-05-07 by cschramm

asl97, 2015-05-07:
@norpol 

> does anyone knows why blueman is not in the Fedora repositories?

seeing how nobody has answer your first question, see 

1. https://fedoraproject.org/wiki/Changes/Bluez5
2. https://lists.fedoraproject.org/pipermail/devel/2013-August/187223.html (this link is in the link above)