blueman-project/blueman #277: Struct() argument 1 must be string, not unicode

yars068, 2015-05-07:
Hi folks,
I have a issue when trying to run latest blueman snapshot (since may-07-2015). There are:
```
$ blueman-applet 
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 31, in <module>
    from blueman.Functions import *
  File "/usr/lib64/python2.7/site-packages/blueman/Functions.py", line 52, in <module>
    in_fg = os.getpgrp() == struct.unpack('h', fcntl.ioctl(0, termios.TIOCGPGRP, "  "))[0]
TypeError: Struct() argument 1 must be string, not unicode
```
That is on Slackware64-41, Bluez-5.29 and python-2.7.5. UTF8 is system encoding.

infirit, 2015-05-07:
Hmm, I thought I fixed that.. Anyway, the below should fix the problem, please report if this solved it.

```diff
diff --git a/blueman/Functions.py b/blueman/Functions.py
index 29494a1..8e924e7 100644
--- a/blueman/Functions.py
+++ b/blueman/Functions.py
@@ -49,7 +49,7 @@ YELLOW = lambda x: "\x1b[33;01m" + x + "\x1b[39;49;00m"
 import fcntl, struct, termios
 
 try:
-    in_fg = os.getpgrp() == struct.unpack('h', fcntl.ioctl(0, termios.TIOCGPGRP, "  "))[0]
+    in_fg = os.getpgrp() == struct.unpack(str('h'), fcntl.ioctl(0, termios.TIOCGPGRP, "  "))[0]
 except IOError:
     in_fg = 'DEBUG' in os.environ
```

yars068, 2015-05-07:
Thanks for patch. But I have a new issue:
```
$ LANG=en blueman-applet 

(process:28890): Gtk-WARNING **: Locale not supported by C library.
	Using the fallback 'C' locale.
blueman-applet version 1.99.alpha2 starting
Stale PID, overwriting
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
['RecentConns', 'NMDUNSupport', 'AuthAgent', 'PPPSupport', 'Menu', 'StatusIcon', 'ExitItem', 'StandardItems', 'SerialManager', 'ShowConnected', 'KillSwitch', 'TransferService', 'DhcpClient', 'PowerManager', 'GameControllerWakelock', 'DiscvManager', 'NMPANSupport', 'DBusService', 'Networking', 'Headset', 'AppIndicator', 'NetUsage'] 
_________
get_interface_version (/usr/lib64/python2.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
ERROR:root:Could not find any typelib for AppIndicator3
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 

(blueman-applet:28890): GLib-GIO-ERROR **: Settings schema 'org.blueman.plugins.powermanager' is not installed
```

yars068, 2015-05-07:
Sorry, I fix it by running `glib-compile-schemas`.

infirit, 2015-05-07:
See https://github.com/blueman-project/blueman/issues/274#issuecomment-98499950

infirit, 2015-05-07:
>Sorry, I fix it by running glib-compile-schemas.

Ok, I pushes the fix to master so closing :smile:

closed 2015-05-07 by infirit