blueman-project/blueman #278: UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 15: ordian not in range

yars068, 2015-05-08:
Hi!
After [bug #277](https://github.com/blueman-project/blueman/issues/277), I managed to get working blueman, but I occured an error:
```
$ blueman-applet 
blueman-applet version 1.99.alpha2 starting
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
['RecentConns', 'NMDUNSupport', 'AuthAgent', 'PPPSupport', 'Menu', 'StatusIcon', 'ExitItem', 'StandardItems', 'SerialManager', 'ShowConnected', 'KillSwitch', 'TransferService', 'DhcpClient', 'PowerManager', 'GameControllerWakelock', 'DiscvManager', 'NMPANSupport', 'DBusService', 'Networking', 'Headset', 'AppIndicator', 'NetUsage'] 
_________
get_interface_version (/usr/lib64/python2.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
ERROR:root:Could not find any typelib for AppIndicator3
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because it's conflict has higher priority 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NetUsage.NetUsage'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
[... 113 lines cut ...]
!                 dprint("Unable to load plugin module %s\n%s" % (plugin, e))
  
          for cls in self.plugin_class.__subclasses__():
              self.__classes[cls.__name__] = cls
--- 89,95 ----
              try:
                  __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
              except ImportError as e:
!                 dprint("Unable to load plugin module %s\n%s" % (str(plugin).encode('UTF-8'), e)
  
          for cls in self.plugin_class.__subclasses__():
              self.__classes[cls.__name__] = cls
```
But after this patch applying, I have a issue again:
```
$ blueman-applet 
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 43, in <module>
    from blueman.main.PluginManager import PersistentPluginManager
  File "/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py", line 94
    for cls in self.plugin_class.__subclasses__():
                                                 ^
SyntaxError: invalid syntax
```
I'm is not a guru of Python, and I don't know how to fix it, unfortunately.

infirit, 2015-05-08:
I'll push the struct.unpack fix in KillSwitch plugin.

But, the second fix is not right, even though we use ``unicode_literals`` the ``__file__`` attribute still returns a byte string. Please try the below instead.

````diff
diff --git a/blueman/main/PluginManager.py b/blueman/main/PluginManager.py
index 924fadf..9da82fe 100644
--- a/blueman/main/PluginManager.py
+++ b/blueman/main/PluginManager.py
@@ -78,6 +78,9 @@ class PluginManager(GObject.GObject):
             return
 
         path = os.path.dirname(self.module_path.__file__)
+        if hasattr(path, "decode") and hasattr(path, "upper"):
+            path = path.decode("UTF-8")
+
         plugins = []
         for root, dirs, files in os.walk(path):
             for f in files:
````

yars068, 2015-05-08:
No, this patch does not help :(

infirit, 2015-05-08:
That is unfortunate. Please keep the patch but also apply additional debug patch below. Please post the output from blueman-applet again (in a gist or paste website :smile: ).

```diff
diff --git a/blueman/main/PluginManager.py b/blueman/main/PluginManager.py
index aeb20f2..c99d2ec 100644
--- a/blueman/main/PluginManager.py
+++ b/blueman/main/PluginManager.py
@@ -94,6 +94,8 @@ class PluginManager(GObject.GObject):
             try:
                 __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
             except ImportError as e:
+                dprint("plugin: %s" % plugin)
+                dprint("error: %s" % e)
                 dprint("Unable to load plugin module %s\n%s" % (plugin, e))
 
         for cls in self.plugin_class.__subclasses__():
```

infirit, 2015-05-08:
Also as I can not reproduce this on gentoo or debian can you post instructions how you currently build blueman for slackware? I will setup a vm with slackware64.

yars068, 2015-05-09:
Here is output which you asked: [http://pastebin.com/pEr2L3J6](http://pastebin.com/pEr2L3J6).
A number of experiments with these patch indicates that the problem is in variable 'e'.

yars068, 2015-05-09:
This is biult package and all need to rebuild: [https://yadi.sk/d/C265JHGrgXFUT](https://yadi.sk/d/C265JHGrgXFUT).

infirit, 2015-05-09:
>A number of experiments with these patch indicates that the problem is in variable 'e'.

Sigh, I was afraid of that :cry:. I'll see if I can figure out how to install and build on slackware.

yars068, 2015-05-09:
How I can print 'raw' value of 'e' variable?

infirit, 2015-05-09:
You can try to use ``print`` and ``repr`` instead of blueman's ``dprint``. Also try looking at ``e.message`` and print that.

Does the following also cause the problem for you?
```python
try:
	import abcde
except ImportError as e:
	print("error: %s" % e)
```

yars068, 2015-05-09:
This code does not cause problems, but instead of expected "error: No module named abcde" I get just "No module named abcde".

yars068, 2015-05-09:
That code works fine.

infirit, 2015-05-09:
Sorry, I messed up the code :disappointed:.  I updated it with ``from __future__ import unicode_literals`` so please try this again with that :smile:

yars068, 2015-05-09:
That also with no problem. :(

infirit, 2015-05-09:
>Maybe I'll should try with another version of python?

The plan is to support Python 2.7 and 3.2 from the same codebase. You have 2.7 so you are good on the python version.

>but every time, when I open the it's window, I see

This is expected, all blueman-* programs do this.

The manager tries to load two plugins, PulseAudioProfile and Services and it is choking on one of them.

I noticed ``__import__`` was not given globals and locals which __may__ be related to the problem you are seeing. There is a better way in 2.7 to load modules at runtime so I changed the plugin load with this [commit](https://github.com/infirit/blueman/commit/68cf7bf6d9ea24a24dfb6cd56ac9c646630799a1.diff). It works ok for me so until I have a vm please try this long-shot :smiley:

yars068, 2015-05-09:
Perhaps it is PulseAudio plugin, because PA is not part of Slackware. Ok, that commit is applied, but again I see: [http://pastebin.com/nmcNA6Kt](http://pastebin.com/nmcNA6Kt).

yars068, 2015-05-09:
About `e.message`: that is in blueman-browse: [http://pastebin.com/nRBckFGS](http://pastebin.com/nRBckFGS).

infirit, 2015-05-09:
>Perhaps it is PulseAudio plugin

That may be it, can you remove ``blueman/plugins/manager/PulseAudioProfile.py`` and see if that helps? I am on gentoo so I can easily rebuild without pulseadio and try to reproduce.

infirit, 2015-05-09:
>About e.message: that is in blueman-browse

Sorry I confused you with ``e.message`` please ignore this in ``blueman-browse``. The variable e in PluginManager is an Exception object and ``e.message`` contains the text what the exception is about.

yars068, 2015-05-09:
Removal of PulseAudioProfile is working, so we need to modify them to check if PA is installed, and if so, to load the plugin, or not load if not. :)

infirit, 2015-05-09:
The original gentoo ebuild writer/maintainer most likely ran into this or similar problem with pulseaudio  and removed the file when pulseaudio use flag (think support) was disabled.

@yars068, Now that I figured this out please try https://github.com/infirit/blueman/commit/30ec5947a622295b5212a2e250981c5c203c35b2 which should handle pulseaudio libraries missing more gracefully. The previous patches are harmless but should not be needed anymore :smile:.

@cschramm, I'll most likely push the [importlib](https://github.com/infirit/blueman/commit/68cf7bf6d9ea24a24dfb6cd56ac9c646630799a1) commit as well as we should avoid using [``__import__``](https://docs.python.org/2/library/functions.html#__import__) when possible. The ``__import__`` code is used to import modules from a string during normal operation, do you see any other reason why ``__import__`` was used?

cschramm, 2015-05-10:
Replacing __import__ with importlib makes sense to me.

So what's the issue here? Does unicode_literals break __import__ for Python 2.7?

infirit, 2015-05-10:
>So what's the issue here? Does unicode_literals break import for Python 2.7?

No, nothing breaks and unrelated to this issue. Ignore this for now and I'll put it in a PR where we can discuss further and not add noise to this issue :smile:

yars068, 2015-05-10:
@infirit, [your commit](30ec5947a622295b5212a2e250981c5c203c35b2) is working, but need to be extended, because execution of PulseAudioUtils does not stopped when exception is generated.

infirit, 2015-05-10:
>because execution of PulseAudioUtils does not stopped when exception

I do not understand what you mean with that. When we raise an exception loading of the module is stopped. Do you see something in the user interface perhaps?

Anyway, I was looking in PulseAudioProfile.py and I noticed this is already handled. However now I know what is causing your problem I just need you to try one more path, see below :smile:.

```diff
diff --git a/blueman/plugins/manager/PulseAudioProfile.py b/blueman/plugins/manager/PulseAudioProfile.py
index e5c1235..06310fc 100644
--- a/blueman/plugins/manager/PulseAudioProfile.py
+++ b/blueman/plugins/manager/PulseAudioProfile.py
@@ -7,7 +7,7 @@ from blueman.plugins.ManagerPlugin import ManagerPlugin
 try:
     from blueman.main.PulseAudioUtils import PulseAudioUtils, EventType
 except OSError as e:
-    raise ImportError(e)
+    raise ImportError("%s" % e)
 from blueman.main.SignalTracker import SignalTracker
 from blueman.gui.manager.ManagerDeviceMenu import ManagerDeviceMenu
 from blueman.gui.MessageArea import MessageArea
```

yars068, 2015-05-10:
> Do you see something in the user interface perhaps?

No, but I would to see a messages about loading pulse's shared libraries, when PulseAudioUtils tried to load them. But you're right, my patch is not need anyway.

yars068, 2015-05-11:
I think, this issue we can close. Your [commit](b444452bddcdf3ea118cd2a00da5550e0b242130) as I think, does help, but when I have a'browse-command' (Local services > Extended) is empty, I get an error. I'm going to create a new issue about that error...
Thanks you very much, and many thanks you for patience.

yars068, 2015-05-11:
Strange... I confused with this bug :( I cloned a 'master' branch (`git clone https://github.com/blueman-project/blueman.git`), build the package without any patches applied, but the same error appeared again :(
Here is [output.](http://pastebin.com/uwKjqgkB)

infirit, 2015-05-12:
>but the same error appeared again

I finished the install and I can __not__ reproduce this problem. Blueman's applet and manager come up fine with the default bluez package (version 4) and blueman git master from today.

What is the output of ``locale`` (user and system) and provide the output of blueman-manager (note, not the applet!) from a terminal.

I'll re-open but it is very strange..

infirit, 2015-05-12:
@yars068 Can you provide me with the bluez5 build files, I could not find it on slackbuilds..

yars068, 2015-05-12:
Please: [https://yadi.sk/d/FdaIpTi0gaP3y](https://yadi.sk/d/FdaIpTi0gaP3y). Here is bulit package and slackbuild.
Locale: 
```
# locale
LANG=ru_RU.UTF-8
LC_CTYPE="ru_RU.UTF-8"
LC_NUMERIC="ru_RU.UTF-8"
LC_TIME="ru_RU.UTF-8"
LC_COLLATE=C
LC_MONETARY="ru_RU.UTF-8"
LC_MESSAGES="ru_RU.UTF-8"
LC_PAPER="ru_RU.UTF-8"
LC_NAME="ru_RU.UTF-8"
LC_ADDRESS="ru_RU.UTF-8"
LC_TELEPHONE="ru_RU.UTF-8"
LC_MEASUREMENT="ru_RU.UTF-8"
LC_IDENTIFICATION="ru_RU.UTF-8"
LC_ALL=
```
That is system, user-specific `locale` is same, because that is system-wide setting (look in /etc/rc.d/rc.font for console font, and /etc/profile.d/lang.sh for locale).
blueman-manager just saying:
```
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services'] 
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 279, in <module>
    Blueman()
  File "/usr/bin/blueman-manager", line 59, in __init__
    self.Plugins.Load()
  File "/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py", line 90, in Load
    __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/manager/PulseAudioProfile.py", line 7, in <module>
    from blueman.main.PulseAudioUtils import PulseAudioUtils, EventType
  File "/usr/lib64/python2.7/site-packages/blueman/main/PulseAudioUtils.py", line 15, in <module>
    raise ImportError("%s" % e)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 15: ordinal not in range(128)
```

infirit, 2015-05-12:
The main problem is that the exception message is still a byte string under python 2 with unicode_literals. So python implicitly decodes the message to ASCII and we die horribly. This is a problem for __all__ exception that contains non iso-latin1 characters. In this case a localized error message in Cyrillic from ctypes that the shared library can not be found.

@cschramm, I do not know how to handle in a sane manner across python versions. So for now my sugestion would be just to workaround this particular problem and do not take the error message from ctypes and create our own (feel free to suggest a better message :wink:). See below.

@yars068, I am 99% sure the below will fix the problem but I would be good if you can confirm :smile:.

```diff
diff --git a/blueman/main/PulseAudioUtils.py b/blueman/main/PulseAudioUtils.py
index 76126a0..efaa2e8 100644
--- a/blueman/main/PulseAudioUtils.py
+++ b/blueman/main/PulseAudioUtils.py
@@ -11,8 +11,8 @@ from blueman.Functions import YELLOW, dprint
 try:
     libpulse = CDLL("libpulse.so.0")
     libpulse_glib = CDLL("libpulse-mainloop-glib.so.0")
-except OSError as e:
-    raise ImportError("%s" % e)
+except OSError:
+    raise ImportError("Could not load pulseaudio library")
 
 pa_glib_mainloop_new = libpulse_glib.pa_glib_mainloop_new
 pa_glib_mainloop_new.argtypes = [c_void_p]
```

yars068, 2015-05-12:
Yes, this works fine :)

yars068, 2015-05-12:
I think that issue can be completely closed :)

closed 2015-05-13 by infirit

drzraf, 2016-12-04:
In case this helps, I got consistently this error this way:
* Use fr_FR system where Download is called "Téléchargement" (\xe9)
* Run blueman-applet with LANG=C
* Try to enable the Transfer Service
=> results in the same error (with \xe9 byte error)
Solution: fix the locale

commit fde4df8b (2015-05-11): PulseAudio: Fix UnicodeDecodeError with missing pulse library

Push the check down into PulseAudioUtils and force python2
to implicitly decode to codepoints.

closes #278
commit 72fd5a84 (2015-05-13): PulseAudio: Fix UnicodeDecodeError with missing pulse library

Under python2 with unicode_literals Exceptions still contain
byte strings. When we try to load the pulse shared libraries
we may get a localised error message that contains non ASCII
characters. Python 2 will then implicitly decode the byte
string to codepoints with ASCII and fail.

There is no clean cross python version way to handle this but
we are pretty sure what went wrong so use our own error
message instead of the localised one.

Also push the the check down PulseAudioUtils where it belongs.

closes #278