blueman-project/blueman #285: Stream Setup Failed

lewoberst, 2015-05-12:
I am using Linux Mint with
* blueman 1.23-git201403102151-1ubuntu1
* pulseaudio-module-bluetooth 1:4.0-0ubuntu11.1
* pulseaudio 1:4.0-0ubuntu11.1

When I'm trying to connect a bluetooth audio device through blueman, pairing works fine but a connection always fails, more precisely all devices I tried seem to connect successfully and then immediately disconnect. Blueman logged the following events, "###" lines are my comments:

<pre>
### opening config dialogue:

Loading configuration plugins
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['Services', 'PulseAudioProfile'] 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
1 
blueman-manager version 1.23 starting
There is an instance already running
_________
child_closed (/usr/lib/python2.7/dist-packages/blueman/Functions.py:153)
/usr/bin/blueman-manager closed 

### switching on paired device
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 

### manually issuing "Connect to audio sink"
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
notify (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:237)
<Device object at 0x7fb3cc7ebf00 (blueman+main+Device+Device at 0x2c2aea0)> org.bluez.AudioSink dbus.Array([], signature=dbus.Signature('s')) 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:136)
rebuilding menu 
_________

### Device notifies about connect and immediate disconnect

__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
</pre>

I am not sure how to debug this further. I can compile custom versions or grab more debug output if that helps.

cschramm, 2015-05-12:
Is the device paired?

You could check the bluetoothd output (should be in /var/log/daemon.log).

lewoberst, 2015-05-12:
Yes, pairing was successful.

I don't have the log of the previous run, unfortunately. However, retrying, the output is:

<pre>
May 12 17:09:35 raphael dbus[618]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
May 12 17:09:35 raphael blueman-mechanism: Starting blueman-mechanism 
May 12 17:09:35 raphael dbus[618]: [system] Successfully activated service 'org.blueman.Mechanism'
May 12 17:09:35 raphael blueman-mechanism: loading Ppp 
May 12 17:09:35 raphael blueman-mechanism: loading RfKill 
May 12 17:09:35 raphael blueman-mechanism: loading Config 
May 12 17:09:35 raphael blueman-mechanism: loading Network 
May 12 17:10:05 raphael blueman-mechanism: Exiting 
May 12 17:09:27 raphael bluetoothd[3566]: hci0: Remove UUID (0x0011) failed: Busy (0x0a)
May 12 17:10:14 raphael bluetoothd[3566]: Host is down (112)
</pre>

lewoberst, 2015-05-12:
Sorry, the device wasn't turned on during the last test. The actual output is:

<pre>
May 12 17:17:14 raphael bluetoothd[3566]: Unable to select SEP
May 12 17:17:14 raphael kernel: [ 5325.957464] input: 00:02:3C:2E:1C:93 as /devices/virtual/input/input28
May 12 17:17:16 raphael blueman-mechanism: Exiting 
</pre>

cschramm, 2015-05-12:
That's some issue with the BlueZ 4 configuration. Do you have an `Enable=` setting in `/etc/bluetooth/audio.conf`? Try adding `Socket`.

lewoberst, 2015-05-12:
I found that suggestion online as well and had it added to the config
without any effects before asking here.

So you're saying I'd better go searching for help somewhere around
bluez.org, did I get that correctly?

cschramm, 2015-05-12:
There are also references that the error may be triggered by an unloaded `module-bluetooth-discover` module which is the case for your blueman version. You can check with `pactl list`. If it is missing, issue `pactl load module-bluetooth-discover` and try again. If that helps, you can get a fixed version at http://snapshot.debian.org/package/blueman/1.23-git201406261335-deb-1/ to make it work persistently.

lewoberst, 2015-05-13:
Still no luck with your last suggestion, the result remains the same, at least from my end user's perspective. Connect and immediate disconnect.

Btw, just in case someone else stumbles across and wants to try: The command should read `pactl load-module module-bluetooth-discover`.

I'm attaching the log once more because it looks a bit different this time:

<pre>
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Creative WP-300 Headphones 
_________
set_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:75)
op: regenerating instance <ManagerDeviceMenu object at 0x7facf8e88460 (GtkMenu at 0x182f5f0)> 
_________
set_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:75)
op: regenerating instance <ManagerDeviceMenu object at 0x7facfaf65b90 (GtkMenu at 0x182f790)> 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
notify (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:237)
<Device object at 0x7fd052159f50 (blueman+main+Device+Device at 0x2054e40)> org.bluez.AudioSink dbus.Array([], signature=dbus.Signature('s')) 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:136)
rebuilding menu 
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:168)
list: device_prop_ch Connected 1 /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 () {} 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:286)
row update event Connected 1 
_________
monitor_power_levels (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:190)
starting monitor 
_________
level_setup_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:347)
animating up 
_________
unset_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:89)
op: regenerating instance <ManagerDeviceMenu object at 0x7facf8e88460 (GtkMenu at 0x182f5f0)> 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Creative WP-300 Headphones 
_________
unset_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:89)
op: regenerating instance <ManagerDeviceMenu object at 0x7facfaf65b90 (GtkMenu at 0x182f790)> 
_________
fail (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:119)
fail (DBusException(dbus.String(u'Stream setup failed'),),) 

/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerProgressbar.py:174: Warning: Source ID 1498 was not found when attempting to remove it
  gobject.source_remove(self.gsource)
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:168)
list: device_prop_ch Connected 0 /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 () {} 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:286)
row update event Connected 0 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/3566/hci0/dev_00_02_3C_2E_1C_93 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Creative WP-300 Headphones 
_________
level_setup_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:347)
animating down 
_________
update (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:191)
stopping monitor (not connected) 
</pre>

cschramm, 2015-05-13:
Still `Unable to select SEP` in bluetoothd? Hrm..

It's some issue with pulseaudio and the BlueZ 4 settings.

Did you try https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=725610#89 yet?

lewoberst, 2015-05-13:
The system log only has this line once per attempt, with the input number increasing:

<pre>May 13 09:41:22 raphael kernel: [14807.023948] input: 00:02:3C:2E:1C:93 as /devices/virtual/input/input35</pre>

No more `Unable to select SEP`.

That said, the problem should have been solved with your earlier `Enable=Socket` suggestion, only now I realized I had that line added to the `[Headset]` section, but had better added it to the `[General]` section.

Thanks a lot for your help!

closed 2015-05-13 by lewoberst

cschramm, 2015-05-13:
That's probably the HFP or something getting picked up by your input subsystem. :smiley: