blueman-project/blueman #286: Aesthetic issue on Fedora 21 Mate

bjrosen, 2015-05-13:
Selecting a device makes it disappear from the display, see attached screenshot
[image]

cschramm, 2015-05-13:
:astonished: Do you get any messages when when you start blueman-manager in a terminal and click on a device?

infirit, 2015-05-13:
It is not likely gone but the text colour the same as the background colour. How does it look with the default gtk+3 theme adwaita?

bjrosen, 2015-05-13:
I;m using Mate which is GTK2, Adwaita isn't available in Yumex, it might be because I'm using Mate. However using Clearlooks seems to fix the problem.

You can close this, the solution is to find a theme that's compatible.

cschramm, 2015-05-13:
Hmm... Could it be that we're showing white text on a theme defined background (so that it could get white on white)?

infirit, 2015-05-13:
Adwaita is in gnome-themes-standard, but you can also try the MATE menta/traditional themes as they include a Gtk3 theme.

Also which theme are you using so we can have a look?

>Could it be that we're showing white text on a theme defined background.

Yes, both background and foreground colours in both selected and unselected versions are defined in the theme. But I would like to have a look at the theme to verify what is going on.

bjrosen, 2015-05-13:
I stepped through the available themes, most owrk, the one that doesn't is CurvyLooks

infirit, 2015-05-13:
Looks like Gtk3 CurvyLooks has not been updated in a while (~4 years) so broken with recent Gtk3 releases. There is unfortunately not much we can do here.

closed 2015-05-13 by infirit