blueman-project/blueman #288: Deprecated GTK+ / GLib properties

cschramm, 2015-05-21:
We should take care of all those deprecation warnings. Deprecated properties include:

* GtkAlignment:left-padding
* GtkAlignment:right-padding
* GtkAlignment:top-padding
* GtkAlignment:xalign
* GtkAlignment:xscale
* GtkAlignment:yscale
* GtkMisc:xpad
* GtkMisc:ypad
* GtkSettings:gtk-button-images
* GSettings:schema
* GtkContainer:resize-mode

leigh123linux, 2015-07-07:
Can you add PyGI Deprecated fixes

rough example

http://pastebin.com/raw.php?i=Cj6DT64D

and what appears to be gi changes

http://pastebin.com/raw.php?i=Bxg9zCct

infirit, 2015-08-07:
@leigh123linux, your pastes are gone, can you put them in a [gist](https://gist.github.com/) perhaps?

I just took care of the gsettings one with 5ab15125dcfe5ebf5dbf578e50903816eb10c340

leigh123linux, 2015-08-07:
@infirit 

http://pkgs.fedoraproject.org/cgit/blueman.git/tree/fix_gi_import.patch

and

http://pkgs.fedoraproject.org/cgit/blueman.git/tree/PyGIDeprecationWarning.patch

infirit, 2015-08-07:
@leigh123linux, thanks.

I pushed s couple of commits to take care of those. Let me know if folks run into issues :smile:

infirit, 2015-08-07:
@cschramm, which Gtk+3 version is the taget? I ask because ``margin-left`` is deprecated in 3.12 and replaced with ``margin-start``. But if I use the new property this means we drop support for earlier versions.

cschramm, 2015-08-08:
Basically I'd say we stick with the deprecated one for now if the replacement is not available in older versions. But... I could not find `margin-left` in our code. Do you actually want to introduce it (as a replacement for some other deprecated property)?

infirit, 2015-08-08:
I am moving/replacing the alignment/padding set by the ``GtkAlignment`` widget to the ``GtkBox`` widgets, see [here](https://github.com/infirit/blueman/commit/7d661020bceda9d5a88ef7712e1b1b8bbd2b6f14) for what I am talking about.

So yes I am introducing it to replace ``left-padding`` from the ``GtkAlignment`` and in the process figured out that it was also deprecated. I'll keep it for now but let me know if you want me to use the new ``margin-start``.

infirit, 2015-08-08:
The alignment properties and viewport resize-mode deprecations are sorted.

infirit, 2015-08-11:
All but one left, GtkSettings:gtk-button-images.

There are still warnings but these are for the ``margin-left`` and ``margin-right`` properties discussed earlier.

cschramm, 2015-08-11:
Great! :smiley: 

That warning is triggered when calling `set_image` on a `Gtk.Button`. The strange thing is... There's no deprecation notice in the docs...

infirit, 2015-08-11:
Indeed they hid this well, see https://github.com/GNOME/gtk/blob/master/gtk/gtksettings.c#L1313

However, we do not set it anywhere in blueman but I know desktops like MATE and Xfce do fiddle with GtkSettings. I'll poke around MATE to see if I can make the warning go away.

infirit, 2015-08-11:
We do set it in mate-settings-daemon [here](https://github.com/mate-desktop/mate-settings-daemon/blob/master/plugins/xsettings/msd-xsettings-manager.c#L199). However un-setting make no difference to the warning while it does remove the button images for both Gtk versions :confused:..

infirit, 2015-08-11:
Ok, this is what is what is going on. The GtkSetting is deprecated, and when we set the image on a button with text the image can be switched on or off via this setting. It seems this behaviour is deprecated, because reasons (insert your favourite Gtk3 rant). I can force the image on with ``self.b_more.props.always_show_image = True`` and the warning goes away. Which means the image is always shown which may or may not be what was originally intended for this button...

The question is, force to always show the image or let it be handled by GtkSettings and it probably will get forced eventually when the brilliant ui designers of gnome decide deprecation time is over...

leigh123linux, 2015-08-23:
I get this warning from blueman git

```
(blueman-applet:2441): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
```

```
$ rpm -q blueman
blueman-2.1-0.1.git0a5defd.fc23.x86_64
```

cschramm, 2015-08-24:
@leigh123linux How is that related to this change? Did you bisect it?

yars068, 2015-10-21:
I have current blueman-git-07afca4, that is about deprecation warnings: http://pastebin.com/nB5ntQH4

infirit, 2015-10-21:
Regarding the margin-* properties see above. In short, Glade (gui) does not support them and if we use them we drop compatibility with < Gtk+ 3.12. I suggest to wait for Glade to catch up before we look at these.

The images on buttons see [this](https://github.com/blueman-project/blueman/issues/288#issuecomment-129951027)

The GSettings warning is fixed already.

cschramm, 2015-12-17:
I think we can close this, as the current state is the best we can currently achieve. Objections?

closed 2015-12-17 by cschramm

infirit, 2015-12-17:
We can do another round for 2.2 :smile:
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.