blueman-project/blueman #296: Blueman in Tiny Core

geroysasha, 2015-05-31:
Im using Blueman in Tiny Core.
All devices connected successfully.
After reboot system connected devices are disappearing.
Tell me which file stores information about the connected devices, please.

cschramm, 2015-06-01:
I think connected devices are not persisted by BlueZ, just paired ones.

If the paired devices do not show up anymore, you're having some issue with BlueZ. A user reported that this happens with 5.30. Are you using that version?

geroysasha, 2015-06-01:
In Ubuntu 12.04 is not the problem. There is installed Bluez 4.98.
Tiny Core is loaded in RAM and after reboot system connected devices are disappearing.
To save files or settings need to make a backup but I am not know where are settings Blueman. And after every reboot is necessary to reconnect a device:(
For example WIFI connection settings I can save.

cschramm, 2015-06-01:
That's probably in the kernel part of BlueZ. blueman does not store devices itself.

closed 2015-06-01 by cschramm