blueman-project/blueman #298: Random passkey does not appear when attempting to pair device

retired1, 2015-06-04:
Blueman: v2.0
Bluez5: v5.30
Distribution: Vector Linux Vlocity 7.1
Desktop Environment: XFCE

When attempting to pair a device via Blueman, the random passkey confirmation does not appear on the desktop. The device can, however, be paired via bluetoothctl in terminal. 

on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discoverable 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop DiscoverableTimeout 60 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
RequestConfirmation (/usr/lib64/python2.7/site-packages/blueman/main/applet/BluezAgent.py:202)
Agent.RequestConfirmation 
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
    *message.get_args_list()))
  File "/usr/lib64/python2.7/site-packages/blueman/bluez/Device.py", line 33, in err
    raise exception
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discoverable 0 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
Stop 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
RequestConfirmation (/usr/lib64/python2.7/site-packages/blueman/main/applet/BluezAgent.py:202)
Agent.RequestConfirmation 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 

** (blueman-assistant:3596): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(blueman-assistant:3596): Gtk-WARNING **: Theme parsing error: gtk.css:1:58: Failed to import: The resource at '/org/gnome/adwaita/gtk-main.css' does not exist
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
RequestConfirmation (/usr/lib64/python2.7/site-packages/blueman/main/applet/BluezAgent.py:202)
Agent.RequestConfirmation 
Authentication Failed
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_30_F3_1D_F2_B4_B1 

Blueman was compiled from git clone with --prefix=/usr passed to .configure

retired1, 2015-06-09:
Geee. I'm beginning to get a complex here. :(

No comments at all?

cschramm, 2015-06-09:
Sorry, I haven't had a look at this yet. "No agent available for request type 2" sound concerning, but unfortunately I do not know what "type 2" means.

What are you doing in bluetoothctl to pair?

retired1, 2015-06-09:
Pretty much the standard stuff. 

bluetoothctl
agent on
default-agent
scan on
then pair the phone's mac address when it appears. 
A confirmation for the random passkey appears both in bluetoothctl and on the phone. 

Once the device is paired, files can be transferred in both directions and the blueman notices work just fine and appear on the desktop.  For some reason, it's just not working for the passkey confirmation. 

I should probably mention that Vector Linux is a Slackware derivative.

cschramm, 2015-06-09:
So if you do not set a default agent in bluetoothctl you get the "No agent available for request type 2" when trying to pair? Sounds like blueman fails to register the agent.

Please paste the output from blueman-applet's initialization.

retired1, 2015-06-09:
Here's the output from start to an attempted pair. 

http://pastebin.com/UcL3Yada

Interestingly enough, the syslog errors did not appear this time. 

Here's the bluetoothctl output: 

http://pastebin.com/23NnnnME

cschramm, 2015-06-10:
Looks like the agent is registered successfully and the expected callback even happens, but you're missing the notification. Are you getting any notifications at all? What notification daemon do you have? Please fire up a python shell and see if you get a notification with the following:

````python
from blueman.gui.Notification import Notification
Notification("Bluetooth", "Test")
````

retired1, 2015-06-10:
Got a nice little notice at the top right of the screen. 

[code]retired1:$ python
Python 2.7.8 (default, Apr 29 2015, 15:15:32) 
[GCC 4.7.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from blueman.gui.Notification import Notification

** (.:3127): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
>>> Notification("Bluetooth", "Test")
<Notification object at 0x7f97ac9eb6e0 (NotifyNotification at 0x993d00)>
>>> 
[/code]

cschramm, 2015-06-10:
Your output includes

````
RequestConfirmation (/usr/lib64/python2.7/site-packages/blueman/main/applet/BluezAgent.py:202)
Agent.RequestConfirmation
````

I assume this is appearing when you try to pair, right? That means that the blueman agent's callback method is triggered and should show a notification. Maybe you have issues with notification actions.

Please try
````python
Notification("Bluetooth", "Test", 0, [["confirm", _("Confirm")], ["deny", _("Deny")]])
````

retired1, 2015-06-10:
Curious..

```
retired1:$ python
Python 2.7.8 (default, Apr 29 2015, 15:15:32) 
[GCC 4.7.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> Notification("Bluetooth", "Test", 0, [["confirm", _("Confirm")], ["deny", _("Deny")]])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Notification' is not defined
>>> 
```

cschramm, 2015-06-10:
You need to use the same import statement before that.

retired1, 2015-06-10:
I just realized that. I shouldn't be doing this before coffee. LOL

```
retired1:$ python
Python 2.7.8 (default, Apr 29 2015, 15:15:32) 
[GCC 4.7.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from blueman.gui.Notification import Notification

** (.:3256): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
>>> Notification("Bluetooth", "Test", 0, [["confirm", _("Confirm")], ["deny", _("Deny")]])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib64/python2.7/site-packages/blueman/gui/Notification.py", line 167, in __new__
    return klass(summary, message, timeout, actions, actions_cb, pixbuf, status_icon)
  File "/usr/lib64/python2.7/site-packages/blueman/gui/Notification.py", line 138, in __new__
    self.add_action(action[0], action[1], on_action, None)
TypeError: add_action() takes exactly 6 arguments (5 given)
>>> 
```

cschramm, 2015-06-10:
Bam, bug! I'll look at it. :wink:

cschramm, 2015-06-10:
Seems like you're having an incompatible version of libnotify. I have 0.7.6 and poking around in the python shell I get the following:

````python
>>> from gi.repository import Notify
>>> Notify
<gi.module.DynamicModule 'Notify' from '/usr/lib/girepository-1.0/Notify-0.7.typelib'>
>>> Notify.Notification.add_action.get_arguments()
(gi.ArgInfo(action), gi.ArgInfo(label), gi.ArgInfo(callback), gi.ArgInfo(user_data), gi.ArgInfo(free_func))
````

retired1, 2015-06-10:
Interesting. 0.7.5 is installed currently. I'll update and see what happens on this end.

retired1, 2015-06-10:
We updated the blueman package along with libnotify and everything is working as it should. Many thanks for your assistance in helping us track this issue down.

closed 2015-06-10 by retired1

cschramm, 2015-06-10:
Fine. :smiley: Looks like the add_action method is broken for introspection in 0.7.5: https://git.gnome.org/browse/libnotify/commit/?id=2b4ab4d22f42dd264a4ddfa68b02bba0d6c71b0b

jhmartin, 2016-07-16:
For Google searches: Bluez 5.40 sources comes with a command-line python based agent that works with this:
bluez-5.40/test/simple-agent

Sample:
```
RequestConfirmation (/org/bluez/hci0/dev_xx_xx_xx_xx_xx_xx, 731265)
Confirm passkey (yes/no): yes
```