blueman-project/blueman #300: Error when connecting Sphero

thesbros, 2015-06-07:
When I try to connect my Sphero 2.0, I get "Connection Failed: Traceback (most recent call last)":

```
Connection Failed: Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/services/meta/SerialService.py", line 29, in connect
    port_id = create_rfcomm_device(Adapter(props['Adapter']).get_properties()['Address'], props['Address'], 1)
  File "_blueman.pyx", line 212, in _blueman.create_rfcomm_device (_blueman.c:1935)
TypeError: expected bytes, dbus.String found
```

cschramm, 2015-06-09:
Please try this patch: https://github.com/blueman-project/blueman/commit/19a3ccc3e66e02f34cde341d2da5446b3ff8f75a#diff-869c6f1418f71215a47bd024d4aade38

thesbros, 2015-06-09:
That error is gone, now I'm getting:

```python
Connection Failed: Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/services/meta/SerialService.py", line 29, in connect
    port_id = create_rfcomm_device(Adapter(props['Adapter']).get_properties()['Address'], props['Address'], 1)
  File "_blueman.pyx", line 218, in _blueman.create_rfcomm_device (_blueman.c:2053)
Exception: Can't connect RFCOMM socket
```

cschramm, 2015-06-09:
Is it possibly already connected? Do you see a disconnect item in the device menu?

thesbros, 2015-06-09:
There is no option to disconnect. The Sphero does not connect with bluetoothctl either. If I use BlueZ 4 it connects fine, so this is probably a problem with BlueZ 5 rather than Blueman.

cschramm, 2015-06-10:
What are you trying in bluetoothctl? I don't think it provides a way to create an rfcomm device. Anyway, bluez ships with an rfcomm binary. You could try its connect command to see if you get any further with it, but it basically does the same as blueman.

thesbros, 2015-06-10:
This is my problem with BlueZ 5:
```bash
❯ sudo bluetoothctl
[NEW] Controller 5C:F3:70:68:FA:03 desktop [default]
[bluetooth]# power on
[CHG] Controller 5C:F3:70:68:FA:03 Class: 0x1c0104
Changing power on succeeded
[CHG] Controller 5C:F3:70:68:FA:03 Powered: yes
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller 5C:F3:70:68:FA:03 Discovering: yes
[CHG] Device 68:86:E7:02:AA:76 RSSI: -46 # this is the sphero
[bluetooth]# pair 68:86:E7:02:AA:76
Attempting to pair with 68:86:E7:02:AA:76
Failed to pair: org.bluez.Error.AlreadyExists # i already paired, ignore this
[bluetooth]# connect 68:86:E7:02:AA:76
Attempting to connect to 68:86:E7:02:AA:76
Failed to connect: org.bluez.Error.Failed # this is the problem
[bluetooth]# 
```

cschramm, 2015-06-10:
I would not expect connect to succeed if the serial service is the only service provided. BlueZ 5's generic connect method only connects to basic service types like HID or audio, not types like network or serial services. Try the rfcomm tool to see if it performs any better than blueman.

thesbros, 2015-06-12:
I tried rfcomm, no errors but the serial port doesn't work.

cschramm, 2016-10-05:
Closing. Looks like this is not a blueman issue.

closed 2016-10-05 by cschramm