blueman-project/blueman #303: connection.py:651:call_blocking:DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken

leigh123linux, 2015-06-13:
Forwarded from https://bugzilla.redhat.com/show_bug.cgi?id=1231462

```
Description of problem:
I was looking through the various options on "blueman-manager" to see if I could find the battery level of the bluetooth mouse that was connected.  At no particular special time, this error occured.

Version-Release number of selected component:
blueman-2.0-6.fc21

Additional info:
reporter:       libreport-2.3.0
cmdline:        python /usr/bin/blueman-manager
dso_list:       dbus-python-1.2.0-7.fc21.x86_64
executable:     /usr/bin/blueman-manager
kernel:         4.0.4-202.fc21.x86_64
runlevel:       N 5
type:           Python
uid:            1000

Truncated backtrace:
connection.py:651:call_blocking:DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerMenu.py", line 92, in <lambda>
    item.connect('activate', lambda *args: self.blueman.Applet.open_plugin_dialog())
  File "/usr/lib64/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Local variables in innermost frame:
byte_arrays: False
self: <dbus._dbus.SessionBus (session) at 0x7f5c113b7c50>
args: ()
object_path: u'/'
signature: None
bus_name: u'org.blueman.Applet'
get_args_opts: {'byte_arrays': False, 'utf8_strings': False}
timeout: -1.0
kwargs: {}
dbus_interface: u'org.blueman.Applet'
message: <dbus.lowlevel.MethodCallMessage path: /, iface: org.blueman.Applet, member: open_plugin_dialog dest: org.blueman.Applet>
method: 'open_plugin_dialog'
```

cschramm, 2015-06-13:
* Is / was blueman-applet running as well?
* If not, is / was it started automatically?
* Any connected output in blueman-applet?

leigh123linux, 2015-06-13:
I have no idea I only forwarded it from redhat bugzilla

cschramm, 2015-08-19:
Cannot do anything here without information.

closed 2015-08-19 by cschramm