blueman-project/blueman #304: Cannot connect to bluetooth headset

hobarrera, 2015-06-15:
Selecting either `headset` or `handfree` results in the headset not being properly connected:

 * The headset lights up as connected.
 * Pulseaudio doesn't show any new devices.
 * All blueman says is: `Connection Failed: blueman.bluez.errors.DBusFailedError: No such file or directory...`

The logs themselves just show:

    Jun 15 11:21:09 athena.barrera.io bluetoothd[374]: a2dp-sink profile connect failed for 00:07:04:CE:2C:3B: Protocol not available

I'm running the latest from git. Pulseaudio 6.0, on ArchLinux amd64.
Any hints? I can run custom builds, or provide any additional information.

cschramm, 2015-06-15:
Looks like you're running into

````c
        /* Return protocol not available if no record/endpoint exists */
        if (server->source_record_id == 0)
                return -ENOPROTOOPT;
````

in BlueZ. Does you system actually provide an audio source that could be connected to the sink?

hobarrera, 2015-06-21:
I don't quite understand what you mean. :(

What would my system have to provide for headphone to work? I recall this working on prior versions of blueman, so I don't think I'm lacking anything.

cschramm, 2015-06-22:
I'm not really sure what's the problem BlueZ encounters, but it sounds a little bit to me like it's trying to connect to an external audio sink and fails to find a matching audio source (on your local system). Don't know what to check or try to get behind that. Do you have the pulseaudio-module-bluetooth?

hobarrera, 2015-07-12:
I apologize for the slow reply - looks like pulseaudio was recently split into several packages in my distro, and I needed to install pulseaudio-blueblooth, so this was probably a PEBCAK.

Regrettably, some other recent change made the bluetooth driver non-funcional (maybe a kernel upgrade?), and I haven't gotten around to fixing it, so haven't *really* made sure everything works, but since I didn't have pulseaudio-bluetooth installed in the first place, I'm going to attribute my former problems to that and go ahead and close this. I'll get back to you if the problem persists once I fix my new, unrelated issue.

closed 2015-07-12 by hobarrera

bserem, 2016-07-06:
In my case `aptitude install pulseaudio-module-bluetooth` did the trick