blueman-project/blueman #306: Use GNotification

cschramm, 2015-06-19:
Using GNotification provided by GLib 2.39+ we can get rid of the libnotify dependency. Also, GTK+ 3.14 deprecated GtkStatusIcon in favor of it, but I'm not yet sure how to do anything in the system tray with a GNotification.

leigh123linux, 2015-06-20:
Would using libnotify be optional as some DE's use it (MATE)?

cschramm, 2015-06-20:
If I get GNotification correct, it probably uses libnotify itself if available, but to be honest I have not had a closer look yet. If it does not make use of notification daemons like MATE's, it's not an option.

infirit, 2015-06-29:
>GNotification integrates nicely in the GTK+ application framework. It expects you to use GApplication and GAction

Note GNotification requires GApplication see [here](https://developer.gnome.org/GNotification/), so tightly coupled to the new Gnome3 way of doing applications.

>Would using libnotify still be an optional as some DE's use it (MATE)?

It is my understanding that it will fall back to sending a dbus message to ```org.freedesktop.Notifications``` but I can't find a reference for this atm.

cschramm, 2016-11-01:
So using GNotification does not make sense for us and dropping GtkStatusIcon will not be necessary as long as we stick to GTK+ 3. For the future see #610.

closed 2016-11-01 by cschramm