blueman-project/blueman #307: Unable to share internet connection using PAN, it says "Can't add bnep0 to the bridge pan1: Invalid argument(22)"

azazar, 2015-06-19:
```# tail -f syslog 
Jun 19 21:33:34 portable dnsmasq[24720]: started, version 2.71 cachesize 150
Jun 19 21:33:34 portable dnsmasq[24720]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC
Jun 19 21:33:34 portable dnsmasq-dhcp[24720]: DHCP, IP range 10.127.165.2 -- 10.127.165.254, lease time 1h
Jun 19 21:33:34 portable dnsmasq-dhcp[24720]: DHCP, sockets bound exclusively to interface pan1
Jun 19 21:33:34 portable dnsmasq[24720]: reading /etc/resolv.conf
Jun 19 21:33:34 portable dnsmasq[24720]: using nameserver 127.0.1.1#53
Jun 19 21:33:34 portable dnsmasq[24720]: read /etc/hosts - 9 addresses
Jun 19 21:33:35 portable avahi-daemon[669]: Joining mDNS multicast group on interface pan1.IPv6 with address fe80::10:b6ff:fe84:8429.
Jun 19 21:33:35 portable avahi-daemon[669]: New relevant interface pan1.IPv6 for mDNS.
Jun 19 21:33:35 portable avahi-daemon[669]: Registering new address record for fe80::10:b6ff:fe84:8429 on pan1.*.
Jun 19 21:34:25 portable bluetoothd[22651]: Added new connection: bnep0
Jun 19 21:34:25 portable bluetoothd[22651]: Can't add bnep0 to the bridge pan1: Invalid argument(22)
Jun 19 21:34:25 portable NetworkManager[1034]: <warn> /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/bluetooth/hci0/hci0:42/net/bnep0: couldn't determine device driver; ignoring...
Jun 19 21:34:25 portable NetworkManager[1034]:    SCPlugin-Ifupdown: devices added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/bluetooth/hci0/hci0:42/net/bnep0, iface: bnep0)
Jun 19 21:34:25 portable NetworkManager[1034]:    SCPlugin-Ifupdown: device added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/bluetooth/hci0/hci0:42/net/bnep0, iface: bnep0): no ifupdown configuration found.
Jun 19 21:35:10 portable avahi-daemon[669]: Withdrawing workstation service for bnep0.
Jun 19 21:35:10 portable NetworkManager[1034]:    SCPlugin-Ifupdown: devices removed (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/bluetooth/hci0/hci0:42/net/bnep0, iface: bnep0)
Jun 19 21:35:10 portable ModemManager[978]: <info>  (net/bnep0): released by modem /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2
```

cschramm, 2015-06-20:
It looks like the pan1 interface will only be created if you apply the network settings once. Please try to open the local services dialog, go to the "Network" tab and hit apply (just retype the IP address or something to get the button active). You'll probably get a pan1 interface then and the bnep0 setup should succeed.

azazar, 2015-06-20:
It doesn't help.

```
# ifconfig pan1
pan1      Link encap:Ethernet  HWaddr C2:75:B2:90:04:52  
          inet addr:10.244.169.1  Bcast:10.244.169.255  Mask:255.255.255.0
          inet6 addr: fe80::c075:b2ff:fe90:452/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:95 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:15600 (15.2 KiB)
```

```
# ifconfig bnep0
bnep0     Link encap:Ethernet  HWaddr 01:36:05:21:2A:3D  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:5 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1890 (1.8 KiB)  TX bytes:0 (0.0 B)

# brctl addif pan1 bnep0
brctl: bridge pan1: Invalid argument
```

azazar, 2015-06-20:
It looks like it's a bluez problem, not a blueman's.

cschramm, 2015-06-20:
Seems like it as there's nothing blueman could do wrong when creating the bridge...

azazar, 2015-06-23:
Blueman however is affected by that bug. I didn't manage to find any relevant bugtracker. Where should I submit bug report?

cschramm, 2015-06-23:
For BlueZ? I think their user mailing list is the right place: http://www.bluez.org/development/lists/

azazar, 2015-07-16:
I'm sorry, but I didn't post that bug report there. I'm not familiar with mailing lists (I don't know, how to post messages there)

closed 2015-07-22 by cschramm