blueman-project/blueman #311: Can't pair with device in GUI

DavidGriffith, 2015-06-20:
Using Debian 8 (Jessie)...  When starting the pairing process from my phone, I should see a dialog on the Debian box asking for the passcode that I entered into the phone.  This never appears.  I am forced, then, to do the pairing at the command line.  This will severely irritate my non-cli-using users.  What went wrong?

cschramm, 2015-06-20:
What notification daemon do you use?

Please provide the output of blueman-applet (see https://github.com/blueman-project/blueman/wiki/Troubleshooting) when trying to pair.

yars068, 2015-06-20:
I have same problem on Slackware64-14.1, using blueman's snapshot since jun-20-2015 and bluez-5.31. I start blueman-manager in console, try to pair my BT headset and get big flood:
`$ blueman-manager 2>&1 | tee manager.log`: [click](http://pastebin.com/K1bMmPhe). From bluetooth I have at this moment: [click](http://pastebin.com/5q8cwgu3). The dialog box which should be opened does not appear when I'll click on "Search" button in main window. However, this dialog appears when I doing right click on blueman's tray icon and select "Add new device..." in popup menu. But when searching, my headset does not shown. I close and re-open main window again and now see the unpaired headset. I now attempt to pair, but now get fail.

cschramm, 2015-06-21:
@yars068 Looks like the refactorings in master broke the signals stuff. I already fixed the specific issue with the recursion, but there seems to be another problem with the end of the discovery phase I'll have to look into.

cschramm, 2015-06-22:
OK, I fixed both issue in master now.

@DavidGriffith I just now realize that you were referring to master too. Are your issues fixed in 969228c?

DavidGriffith, 2015-06-22:
Pairing will now work if you select "Setup New Device" and go through the motions.  Issue #313 still prevents me from pairing starting with the "Bluetooth Devices" dialog.

yars068, 2015-06-26:
At this time if I open "Bluetooth Devices" window (left-click on tray icon, and the window is appeared), select "Adapter > Search", nothing is not added. But, if I close the window and open it again, I see the new found device. The same if I will click on "Search" button. Also, time-by-time I see the adapter in a device list. To test, I stop blueman, stop bluetoothd, remove the content of /var/lib/bluetooth directory, and start bluetoothd and blueman-manager again.

cschramm, 2015-06-26:
Makes sense. The dbus refactorings (in master, after 2.0) probably broke some signal handler there. I haven't found the time to look into it.

cschramm, 2015-07-16:
Should be done with #313, right?

closed 2015-07-16 by cschramm

yars068, 2015-07-18:
That is fixed, yes :) But, when I send the file to a phone, I successfully receive a file, but on terminal, I see [this](http://pastebin.com/6AkKT4R1).

cschramm, 2015-07-20:
Please check 2.0 and open a new thread for that issue. Thanks!