blueman-project/blueman #312: Can't accept file transfer initiated from device

DavidGriffith, 2015-06-20:
I managed to get a phone paired with my Debian 8 box using bluetoothctl.  Now, I want to select several files on the phone and upload them to the Debian box through bluetooth.  If my box was running Debian 7, a little dialog would pop up from the bluetooth icon asking me if it's okay to accept the transfer.  Now in Debian 8, I instead see the rune in the icon turn green for a few seconds.  Then the phone reports that the transfer failed.  It's not clear what I'm supposed to do when the rune turns green.

cschramm, 2015-06-20:
What notification daemon do you use?

Please provide the output of blueman-applet (see https://github.com/blueman-project/blueman/wiki/Troubleshooting) when trying to send a file.

plaes, 2015-06-21:
I'm running into similar issue when trying to receive the file. The remote device is also shown for 1-2 seconds in the devices dialog. And following traceback appears in the console from blueman-applet:
```
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:16)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
    handler(name, value, **kwargs)
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 48, in on_device_property_changed
    d = Device(path)
  File "/usr/lib64/python2.7/site-packages/blueman/main/Device.py", line 39, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/lib64/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

DavidGriffith, 2015-06-21:
Using Bluez 5.23 in Debian Jessie.  I believe the notification daemon I'm using is from MATE.

cschramm, 2015-06-22:
Do the current fixes (see https://github.com/blueman-project/blueman/issues/311#issuecomment-113864730) help?

DavidGriffith, 2015-06-22:
The rune no longer turns green.  If the device is trusted, Blueman will silently accept the incoming file.  With the version of Blueman in Debian 7 (1.23), a balloon would pop up asking if it's okay to accept the file transfer.  From here it looks like someone decided to replace this behavior with the idea of "if this device is trusted, silently accept the transfer".  The thought is good, but it you really need to retain the option to have Blueman ask permission every time.  Furthermore, consider whether or not to say anything if something UNtrusted tries to send a file.  Maybe something like this: "An untrusted device, $name, is trying to send a file.  Do you want to accept it?  Ignore further attempts from $name?".

cschramm, 2015-06-22:
You should of course alway get notifications no matter if the device is trusted or not. This only determines the text and if there are buttons or not. I'll look into why that happens. We refactored a lot signals stuff in the current master version, so there's probably something broken.

cschramm, 2015-06-24:
Unfortunately I currently don't have the time to take a closer look at this. Could you please test if you have the same issue with release 2.0?

DavidGriffith, 2015-06-24:
With the 2.0-tagged release, things will work as expected (balloons, green runes, etc) except that no permission is asked if the device is "trusted".  IMHO, there should be an option to ask permission even if the device is "trusted".  Or an I misunderstanding the meaning of "trust" as it applies to Blueman?

cschramm, 2015-06-24:
Trusted means that the device does not need permission to connect services or send files.

cschramm, 2015-07-16:
Works for me with current master.

closed 2015-07-22 by cschramm

yars068, 2015-07-22:
On my system, that works, but when I started a transfer, I see dialog about it's state, the progress graph is always empty, nothing reported and dialog does not disappeared at end of transfer. The phone reports succesfully upload. I will attach a log in short time.

cschramm, 2015-07-22:
Not sure what progress graph you're meaning. The dialog is expected not to disappear as it informs you about the received file and provides a button to open it.

timothybarus77, 2019-09-11:
hello my name is timo, I have a problem with Blueman and I want to ask, is the "Accept file from trusted device" feature (on local setting -> transfer tab) having problems? I want to use this feature in my project, but when I try to send a file from my cellphone (Android), i got notification fails to send the file on my phone. can you help me? @cschramm 

My device already set in trusted device but didn't work

cschramm, 2019-09-15:
@timothybarus77: Please open a new issue filling in the base information in the issue template and providing a log from blueman-applet as described in the troubleshooting information it provides.