blueman-project/blueman #313: Unable to scan for devices

DavidGriffith, 2015-06-21:
Attempting to scan for devices fails.  Error log at http://pastebin.com/ACMzWkVq.  Open a Devices dialog and try to scan for devices.  The progress bar appears at the bottom, but nothing will actually happen.  No devices show up.

Using the latest from the repo (741463dac268e25c13379d75234e1470c1e2c076).  Debian Jessie, MATE, Bluez 5,

cschramm, 2015-06-22:
Do the current fixes (see https://github.com/blueman-project/blueman/issues/311#issuecomment-113864730) help?

DavidGriffith, 2015-06-22:
The progress bar moves and finishes, but nothing is displayed.

cschramm, 2015-06-24:
Unfortunately I currently don't have the time to take a closer look at this. Could you please test if you have the same issue with release 2.0?

DavidGriffith, 2015-06-24:
The 2.0 tag works.  The latest revision that will detect and display devices is c57bd1ecf6f340698066e40ced5894910c6d469e.

cschramm, 2015-06-24:
OK, thanks for the info. https://github.com/blueman-project/blueman/commit/4d12346508adf224ad25a484ee0ae9623c4410ad is bad. ;) But you say https://github.com/blueman-project/blueman/commit/488b2038c80a3e3a16f37496d005e86ac8f89116 does not work as well? Hm. I'll have a look as soon as I have some time for it.

DavidGriffith, 2015-06-24:
Correct.

DavidGriffith, 2015-06-24:
Incidentally, if I pair up with my device using bluetoothctl while blueman-manager is running, the latter will notice this and update itself as expected.

commit 7fa19b18 (2015-07-16): Fix device-created / device-removed signals

Closes #313
closed 2015-07-16 by cschramm