blueman-project/blueman #314: Blueman icon briefly vanishes when turning bluetooth on

DavidGriffith, 2015-06-22:
If Bluetooth is off and you turn it on using a right-click on the icon and selecting "Turn Bluetooth On", the icon will vanish for a quarter of a second and then reappear.  The icon should change from the off type to the on type without doing this vanishing act.

cschramm, 2015-07-15:
I just had a look at this.

The icon is basically visible whenever bluetooth is soft-locked or adapters are available.

Before taking the action, bluetooth is soft-locked, so the icon is visible.

When enabling it and thus removing the soft lock no adapters are available when querying, so the icon is removed.

As soon as adapters become available, it will be visible again.

yars068, 2015-08-11:
I think removal of icon is bad idea, IMO we need to change its color, for example, when bluetooth is hard-locked or no adapters available, the icon is gray, with red "x" at bottom-rigt corner, when adapters availabe, but soft-locked, the icon is gray, and so on...

cschramm, 2015-08-11:
The x icon would be of no use. You cannot do anything with blueman if there's a hard lock or no adapters. And in a desktop environment with blueman pre-installed I would not want to have a Bluetooth widget if I either do not have our disabled Bluetooth.

PR #599 (2017-02-04): Delay hiding the status icon when ...
...bluetooth gets re-enabled.

Right after re-enabling bluetooth no adapters are available so the icon vanishes for a short moment until adapters are presented. To avoid that, just wait a second for adapters to show up before hiding the icon if none are present at all.

Fixes #314
commit 2bb615cf (2017-02-04): Delay hiding the status icon when ...

...bluetooth gets re-enabled.

Right after re-enabling bluetooth no adapters are available so the icon vanishes for a short moment until adapters are presented. To avoid that, just wait a second for adapters to show up before hiding the icon if none are present at all.

Fixes #314
closed 2017-02-04 by cschramm