blueman-project/blueman #317: blueman-manager[24053]: segfault at 30 ip 00007fd8391252f0 sp 00007fff89a36b00 error 4 in libgtk-3.so.0.800.2[7fd838f15000+4c0000]

yars068, 2015-06-23:
Hi, I was get this in dmesg, when open main window, select the device and right-click on them, then try to select a connection profile. Latest blueman (since Jun-23-2015), bluez-5.31, slackware64-14.1, gtk+3-3.8.2, Glib2-2.36.4, pygobject-3.10.2, Cython-0.21.1, dbus-python-1.2.0-x86_64-1, dbus-glib-0.100.2-x86_64-1, dbus-1.6.12-x86_64-1.

cschramm, 2015-06-24:
Looking at your GTK+ version, it's probably #42. What theme do you use?

yars068, 2015-06-25:
I use oxygen-gtk3 on Xfce4.12 theme. I'll tried to change theme to Adwaita, but that does not help.

infirit, 2015-06-29:
You are indeed suffering from #42 which seems to be fixed in Gtk >= 3.10.7. Closing as duplicate.

closed 2015-06-29 by infirit