blueman-project/blueman #319: Use a logging framework?

infirit, 2015-06-30:
This is to get some discussion going :smile:.

Several times I got weird issues to then find out blueman-X was throwing exceptions but because I did not ran it in a console I never saw them. Or sometimes the whole exception is silently ignored because of overly broad try/except blocks.

And every-time I thought we can do better than dprint logging to a console. For example Python's  logging module can log to console, syslog and a file separately or all at the same time. We can have a central log that contains the output from all of blueman.

I think it will also help users if they can just toggle a log switch (in gsettings perhaps?) instead of having to run from a console and then copy/paste the output.

Opinions?

cschramm, 2015-07-01:
:+1: Can't see any downside.

yars068, 2015-07-01:
IMO if that is better, is a good idea to use logging module. That is possible to enable logging with environment and, for example, with gsettings, separately?

infirit, 2015-07-01:
I have not looked into or thought about how to enable and set the levels yet. I have only experimented a bit with the format and handlers see the following if you are interested in my brain droppings   https://gist.github.com/infirit/b938cb54964336a97158.

yars068, 2015-07-01:
Yes, It is interesting for me, and I will do some experiments with it

cschramm, 2015-07-22:
I really like this especially as we're getting levels and our current output is really bloated.

I think we should set up a file similar to @infirit's experiment and import applet_logger, manager_logger, mechanism_logger from it. What's missing there is a way for the user to set the debug level. I'm not sure how to provide that. I think the most comfortable way would be gconf configuration, because parameters or environment variables are hard to set at least in case of blueman-mechanism.

infirit, 2015-07-22:
I have not looked into this much but.. some background on logging module. The levels, format, handlers and options set on the __root__ logger propagate down to children. Taking my my brain-droppings as example ``blueman.applet`` get it's initial level from ``blueman``.

My initial thought was to have a global gsettings key to set the level of the root logger. And perhaps additional per logger through a command line option or env variable.

infirit, 2017-01-21:
This was done. Leave this open for a bit and raise issues that come up here or just close it?

yars068, 2017-01-21:
Just close. It was interesting, but I have no time to learn it, sorry.

closed 2017-02-02 by cschramm