blueman-project/blueman #321: Turning off bluetooth via applet does not turn off physical bluetooth LED

petter3k, 2015-07-14:
As topic says.

Happens after latest updates.

Arch Linux x64
Zenbook UX32LN

local/blueman-git 2.0.r2.g9bd5a20-1
extra/bluez 5.32-1
extra/bluez-libs 5.32-1
extra/bluez-utils 5.32-1

cschramm, 2015-07-15:
I'd say this is expected as blueman only sets a soft killswitch, no hard one. Was it different before? Can you render "latest updates" more precisly? Do you mean changes in blueman master?

petter3k, 2015-07-15:
I do not know the version I had prior to the update, but yes, it was different before, but it might not be related to blueman at all (I wouldnt know).

The reason I know this started after the updates is because the icon in the applet (in Cinnamon atleast) has changed when bluetooth is enabled, while the "disabled" icon is the same as before.

cschramm, 2015-07-16:
Could also have been a driver / kernel update I guess. Could you test an older version of blueman to see if it show a different behavior? Otherwise I don't see anything we could do here...

petter3k, 2015-07-16:
Well, in older versions, toggling bluetooth on/off through the applet would give a statement in dmesg showing that rfkill was toggled. Now there is nothing in dmesg after enabling/disabling bluetooth through the applet. I guess if the code is not altered in the current version of blueman, its caused by something else on my part, and if so, I apologize for the false "issue".

frikisama, 2015-07-18:
I can confirm that enabling and disabling Bluetooth using Blueman doesn't actually change the status in rfkill. If the adapter is on, it stays on although Blueman seems to think it is switching it on and off, as shown by the icon, and if the adapter is off it's impossible to switch it on with Blueman.

Arch Linux with latest bluez and blueman from repos.

petter3k, 2015-07-18:
Then something is definitely wrong, as Im 100% sure it did change rfkill before.

cschramm, 2015-07-19:
The output changed in master, but there should still be some for the kill switch events, tagged with `io_event`.

I think it's not possible that you switch bluetooth on and off in blueman without kill switch events as blueman itself relies on those events to show up to update the icon.

petter3k, 2015-07-21:
Switching to blueberry and turning off bluetooth with applet, toggling the following event in dmesg:

```
[  187.258151] usb 1-4: USB disconnect, device number 5
```

Retoggling it, gives the following:

```
[  485.959109] usb 1-4: new full-speed USB device number 6 using xhci_hcd
[  486.139394] Bluetooth: hci0: read Intel version: 370710018002030d00
[  486.139461] Bluetooth: hci0: Intel Bluetooth firmware file: intel/ibt-hw-37.7.10-fw-1.80.2.3.d.bseq
[  486.319581] Bluetooth: hci0: Intel Bluetooth firmware patch completed and activated

```

closed 2015-09-13 by cschramm