blueman-project/blueman #322: there's no such locale as „he” in std locale in my system. Is my Linux distro developers fault, or will some standards are not taken into account?

LordBlick, 2015-07-16:
I've found non-packetable locale file(no parent dir in system packaging hierarchy):
/usr/share/locale/he/LC_MESSAGES/blueman.mo
while reports locale me:
$ locale -a | grep --text "he"
hebrew
he_IL
he_IL.iso88598
he_IL.utf8
the_NP
the_NP.utf8
Build from last (in the moment) commit:
$ git log --pretty=oneline -1
3ce97e90c9f1dbe53cb3363245343fdccf216592 Functions: handle IOError in get_pid function
$  locale -V
locale (GNU libc) 2.21
Copyright (C) 2015 Free Software Foundation, Inc.
[…] (Translated licence staff etc.)

cschramm, 2015-07-16:
Not sure what's the problem here... Your `locale` output shows that `he` is not present on your system, right. I don't have it either, but I have a couple of files for it:

    % ls /usr/share/locale/he/LC_MESSAGES 
    colord.mo               gnome-mime-data.mo    newt.mo
    diffutils.mo            gnome-vfs-2.0.mo      polkit-gnome-1.mo
    GConf2.mo               gtksourceview-2.0.mo  shared-mime-info.mo
    gimp20-libgimp.mo       libbonobo-2.0.mo      system-config-printer.mo
    gimp20.mo               libgnome-2.0.mo       vte-0.0.mo
    gimp20-script-fu.mo     libgnome-keyring.mo   vte-2.90.mo
    gimp20-std-plug-ins.mo  libwnck.mo            xdg-user-dirs.mo
    gimp20-tips.mo          Linux-PAM.mo

What do you mean by "non-packetable"?

infirit, 2015-07-16:
The parent language code is ``he`` and applies to all it's children.

LordBlick, 2015-07-16:
Nevermind, my rpm macro %find_lang from some reason does not accepts that locale, although any other does.

closed 2015-07-16 by LordBlick