blueman-project/blueman #323: unable to connect to keyboard

psiloi, 2015-07-16:
I'm under linux mint 17.2 but got the same problem with 17.1. I am with either cinnamon or mate (this one preferably), 64 bit install.

I can't connect my Kensington key folio pro 2. The device is recognized, the pairing dialog appear, a number is generated, but I cannot succeed in entering it on the keyboard to make the pairing. After 1 or 2 seconds, a dialog of failure appears. Same if I choose to enter my own code (even something simple and quick to type).

BUT the very first time I tried to make the pairing, it did work. The only time. Removing the device does not alter anytthing. In fact, after deleting it, I scan and find my keyboard and try installing it. Only 1 second after the code is shown does the error message shows up.

If I try it on an other computer with linux mint 17.2 32 bit, there is no problem (KDE or mate). I am using the same bluetooth USB key on both computers.

here is dmes gbluetooth related full listing:
[   26.135522] Bluetooth: Core ver 2.19
[   26.135544] Bluetooth: HCI device and connection manager initialized
[   26.135787] Bluetooth: HCI socket layer initialized
[   26.135792] Bluetooth: L2CAP socket layer initialized
[   26.135803] Bluetooth: SCO socket layer initialized
[   26.323739] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   26.323743] Bluetooth: BNEP filters: protocol multicast
[   26.323753] Bluetooth: BNEP socket layer initialized
[   26.336613] Bluetooth: RFCOMM TTY layer initialized
[   26.336625] Bluetooth: RFCOMM socket layer initialized
[   26.336631] Bluetooth: RFCOMM ver 1.11
[ 2446.923046] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[ 2446.923060] Bluetooth: HIDP socket layer initialized

psiloi, 2015-07-16:
an update: I have been able to connect my keyboard on the 64bit system after installing bluez-compat and then typing:

   sudo hidd --connect <mac-address-of-keyboard>

so bluez 4 seems to be OK.

cschramm, 2015-07-16:
Linux Mint ships a very old version of blueman that is not really supported anymore. Can you try a more recent one?

psiloi, 2015-07-16:
Hi,
Thank you for your quick answer. (have you seen my second message latest update BTW?)

I have tried compiling the latest blueman and bluez to get around that old stuff. Bluez 5 I can compile and make intsall fine. But then blueman I cannot. For bluez 4 is installed on my system as far as aptitude/apt-get is concerned, and when I type ./configure on the blueman 2 sources, it tells me to install bluez 5 or to fiddle config variables in an undocumented way and I could not google that connendrum with any success.

===== ./configure output =====
checking for BLUEZ... no
configure: error: Package requirements (bluez >= 4.61 gthread-2.0 >= 2.32) were not met:

No package 'bluez' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables BLUEZ_CFLAGS
and BLUEZ_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
==== end of output ====

BLUEZ_CFLAGS and BLUEZ_LIBS are shell variables, right? But to what value should I set them? I have never before got such a request when compiling stuff!

cschramm, 2015-07-16:
Guess the message is irritating as it checks for bluez and gthread in one step and you probably just need libglib2.0-dev.

psiloi, 2015-07-16:
I have already installed libglib2.0-dev so no it isn't (the messages adout any libdev are clearer than that).

cschramm, 2015-07-20:
PKG_CONFIG_PATH can be used to specify a non-default pkg-config path, but I don't think you have the require library anywhere non-default. Aynway, there are a lot of more up-to-date deb packages, so e.g. just get [1.99.alpha1 from jessie](https://packages.debian.org/jessie/blueman#pdownload). Its dependencies should work fine with Ubuntu 14.04 and thus Linux Mint.

psiloi, 2015-07-21:
Hi,

I have been able to connect to my keyboard with blueZ 4 and blueman 1.99.
Thank you very much.

But I have no success in connecting my audio headset. So I keep being
interested in compiling blueman v2. And My previous questions remain. Could
you help there? I f you recall I compiled blueZ v5 and installed it from
there: ne package is being involved, so how can I manage this for configure
to succeed when it is looking up for blueZ v5+?

regards,

cschramm, 2015-07-21:
Depending on your architecture you should have something like /usr/lib/x86_64-linux-gnu/pkgconfig/bluez.pc for other software to be able to find libbluetooth.

Concerning the audio device issue, you may just be missing pulseaudio-module-bluetooth.

cschramm, 2015-07-21:
A blueman 2.0 deb package is available in Debian testing: https://packages.debian.org/stretch/blueman

You will need obexd-client (>= 0.47) or bluez-obexd, which is not satisfiable in Ubuntu / Linux Mint. You can get that from Debian too or ignore the dependency and lack support for receiving files.

closed 2015-07-21 by cschramm

psiloi, 2015-07-21:
Thanks for your help. I have been able to compile blueman  2.0!

As for my bluettoth headset, I already have the module installed that you
told me about. The problem was that the pulseaudio daemon was not started.

but... there are now major problems:
#1: I can no more connect my keyboard. The input service is no more
offered. my keyboard is instead a human what-ye-call'em and if I try to
connect, I am not given the random keypass to enter at the keyboard and the
blueman programm is frozen while trying to connect. See #2 for important
details
#2: bluetooth is no more a service. It is not listed as a service. So no
daemon started! I tried the linux from scratch way to install it, rebooted,
but it did no good. I reinstalled bluez v4 and kept the now current config
file and rebooted.I encountered problem #1 so completely disintall bluez v4
via synaptic, renstalled it (no more question about config ???) and
rebooted but nothing changed.

How can I get back a woring bluetooth config? Or a complete wipeout of
bluetooth?

regards,

cschramm, 2015-07-21:
> bluetooth is no more a service. It is not listed as a service.

Where? In systemctl?

psiloi, 2015-07-21:
systemctl ???
where what? what are you telling me to locate?

cschramm, 2015-07-21:
You wrote about some list where the bluetoothd service is missing. I'm just wondering what you're referring to.

psiloi, 2015-07-21:
the list I got by so doing:

$ sudo service --status-all

cschramm, 2015-07-21:
So it's a SysVinit tool you're talking about. Did you actually try starting the service? What are you told?

psiloi, 2015-07-21:
Oh, since I mess with it according to
http://www.linuxfromscratch.org/blfs/view/svn/general/bluez.html the
service is listed and functional.

The only problem is that the bluetooth experience is despicable. I cannot
anymore connect my keyboard (bluez v5 + blueman v2). the keyboard is
listed. If i attempt a pairing, a random number option is chosen. This
random number I am not shown, and blueman stalls to death (I must kill it
when it is on the trying to connect screen).

So I'd like to wipe out bluetooth support from my system before
reinstalling bluez v4 and blueman 1.99.

regards,

psiloi, 2015-07-21:
I tried a full wiping and reinstalling it is sort of working. first with
stock bluez and blueman of linux mint.

well, it seems something has gone wrong: blueman-manager has an exception
with python:

Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 27, in <module>
    from gi.repository import GObject
  File "/usr/lib/python2.7/dist-packages/gi/__init__.py", line 32, in
<module>
    raise ImportError('When using gi.repository you must not import static '
ImportError: When using gi.repository you must not import static modules
like "gobject". Please change all occurrences of "import gobject" to "from
gi.repository import GObject".

The proposed change is no solution, the same error does occur.

I tried blueman 1:99 but it's also broken:

_________
Load
(/usr/local/lib/python2.7/dist-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services']
_________
__load_plugin
(/usr/local/lib/python2.7/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'>
_________
__load_plugin
(/usr/local/lib/python2.7/dist-packages/blueman/main/PluginManager.py:133)
loading <class
'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
_________
pa_context_event
(/usr/local/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:341)
1
blueman-manager version 2.0 starting
Stale PID, overwriting
_________
pa_context_event
(/usr/local/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:341)
2
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:105)
org.bluez owner changed to  :1.3
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 604, in
msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/bin/blueman-manager", line 126, in on_bluez_name_owner_changed
    self.Config = Config()
TypeError: __init__() takes at least 2 arguments (1 given)
_________
pa_context_event
(/usr/local/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:341)
3
_________
pa_context_event
(/usr/local/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:341)
4
_________
on_pa_ready
(/usr/local/lib/python2.7/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:27)
connected
_________
<lambda>
(/usr/local/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:353)
1

====

So I wipe all bluez and blueman with dpkg -r

now I install bluez 5.32 from source

than install blueman 1.99 with dpkg -i

and reboot

problem : bluetooth is not a service available in my machine (not listed
with service --status-all).

OK, this is where I was when I trid the linux from scratch script. I won't
do it again. any idea on how to do it then?

regards,

2015-07-21 15:21 GMT+02:00 Jean-Pierre Rivière <jn.pierre.riviere@gmail.com>
:

> Oh, since I mess with it according to
> http://www.linuxfromscratch.org/blfs/view/svn/general/bluez.html the
> service is listed and functional.
>
> The only problem is that the bluetooth experience is despicable. I cannot
> anymore connect my keyboard (bluez v5 + blueman v2). the keyboard is
> listed. If i attempt a pairing, a random number option is chosen. This
> random number I am not shown, and blueman stalls to death (I must kill it
> when it is on the trying to connect screen).
>
> So I'd like to wipe out bluetooth support from my system before
> reinstalling bluez v4 and blueman 1.99.
>
> regards,
>
>
>
>> So it's a SysVinit tool you're talking about. Did you actually try
>> starting the service? What are you told?
>>
>>
>
>
>
> Jean-Pierre

cschramm, 2015-07-21:
Maybe the version you built interferes with the old one from the repository. Current blueman (and any 1.99 version) does not have any static gobject usage.

psiloi, 2015-07-21:
that's why I decided to use bluez 5.32 in my latest tries.

I have applied the gnom patch to v5.32 now. the bluetooth service is now
existent! Great, but alas it cannot be started :-(

my linux knowledge has reached its boundaries there. I don't know what to
do further.

the service --status-all gives:
 [ - ]  bluetooth

and a service bluetooth start:
start: Job failed to start

service bluetooth restart:
stop: Unknown instance:
start: Job failed to start

regards,

cschramm, 2015-07-21:
> my linux knowledge has reached its boundaries there. I don't know what to do further.

Just use binary packages. That's what they are for.

psiloi, 2015-07-21:
regards,
they lust are a nightmare to install with dpkg because all the system has
then to be upgraded.

and I am unable to revert to a clean state.

I probably will have to make a complete system reinstall :-(