blueman-project/blueman #325: blueman-applet 2.0 crashes on suspend

City-busz, 2015-07-17:
It happens always on suspend. Distribution is Arch Linux. The log:
```
blueman-applet version 2.0 starting
Stale PID, overwriting
/usr/lib/python3.4/site-packages/gi/overrides/__init__.py:313: Warning: The property GSettings:schema is deprecated and shouldn't be used anymore. It will be removed in a future version.
  return super_init_func(self, **new_kwargs)
_________
Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
['ExitItem', 'NMDUNSupport', 'AppIndicator', 'DhcpClient', 'StandardItems', 'KillSwitch', 'Menu', 'PowerManager', 'Headset', 'RecentConns', 'NMPANSupport', 'ShowConnected', 'AuthAgent', 'TransferService', 'NetUsage', 'Networking', 'DiscvManager', 'DBusService', 'SerialManager', 'PPPSupport', 'GameControllerWakelock', 'StatusIcon'] 
_________
Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3, introspection typelib not found 
_________
get_interface_version (/usr/lib/python3.4/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3.4/site-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
[... 1771 lines cut ...]
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 170 in initialize
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py", line 328 in add_item
  ...
Interrupted (core dump)

cschramm, 2015-07-22:
Please install the dconf CLI and post the output of `dconf read /org/blueman/plugins/recentconns/recent-connection`.

City-busz, 2015-07-22:
```
$ dconf read /org/blueman/plugins/recentconns/recent-connections
'eNp900tT00AAB/DKU4qi+EJQMCpqQcGEtrTEB1aKtFYrCmh8oIQmZTPTpvzzcMY6OO5hvegH8Noj45G7l1y9On4Kv4PbFMaBaZPD7mYfyf7+m3xuy4VWEYps4VBE6QiFQo5R1tG2kPpd7OlY+zXu95kq72tXRnkz45ZVU8iajm4V1YIupPX3Bq+WdMuvI5lsegwdSjefqmqapds2OgvaumuHlxzLMDfC6FL6+aAoytK8nJ6T50U5lZCTMXQzHKboUTr5aNlw9DLCef85G7axYaol9OZC/m5c19BwhHT5+xH5JUlTsYl6Y0KqF8l6IYpJMV6cWY/Gius4ytBHccx/tFoyVBvH+fJefpeLx0VDWPjgWgb6GU5QnPTfYRQqJk7xSfUlm6Ri6jjNcIZiYJembvIEcJY0NAnumJHTUTmRkKMJWZzGIMMQxTmli49qfkY4n3e3MBwhjXAHv/79SNqVkXqmXq3k7VRVxxAylWrVqxW9bcvbtg1hsWI6GCGdpFVoFxgEiosknCc8HrI/FWk6MJVLDJcpRkmzLK4wXKW4RvZHEGEYoxgnrdnXGW5QTBDfOrln/ZQf/s6tfXxJytUMb0coVizLq9m4GWATGSSKqeY2UQ20RRliFPGmtmmGBEXygG2GQaa4FWC7zXCH4m7DNrtrCw/8/KFxW/0l/Ccw1BI/NMvBvQBYiuE+xVwLmBQISzPMUzxoCltgyFBkD8AeMuQoHgXAHjPkKZ40YIt7sKE/2W8cNsyXLHs1q2Dz86ryoyP/v9ZIeiU/hqcB1GcMSxTLLajRQOoKw3OKF02pCsNLilcHqK8Z3lCsBlDfMryjWKtT9S9QJ/8BjvpZHA=='

closed 2015-07-22 by cschramm

City-busz, 2015-07-22:
It fixes the problem. Thank you very much!

cschramm, 2015-07-22:
:smiley: You're welcome.

commit 9619d3b5 (2015-07-22): Fix infinite loop if RecentConns has only invalid items

Closes #325
commit 21934db6 (2015-07-22): Fix infinite loop if RecentConns has only invalid items

Closes #325