blueman-project/blueman #327: Blueman no longer works

elieobeid7, 2015-07-21:
manjaro latest update, can't enable blueman for some reason, error is

>    Connection Failed: Traceback (most recent call last):
>       File "/usr/lib/python3.4/site-packages/dbus/service.py", line 707, in _message_cb
>         retval = candidate_method(self, *args, **keywords)
>       File "/usr/lib/python3.4/site-packages/blueman/main/DbusService.py", line
> 36, in wrapper
>         return method(*args[1:], **kwargs)
>       File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/DBusService.py",
> line 76, in connect_service
>         self.Applet.Plugins.RecentConns.notify(service)
>       File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py",
> line 261, in notify
>         for i in RecentConns.items:
>     TypeError: 'NoneType' object is not iterable

cschramm, 2015-07-22:
It looks like RecentConns may be uninitialized in case of the notify method, but this is a really odd situation as it means neither a name owner change for the org.bluez bus occurred, nor was an adapter added.

Please provide a the full output when starting blueman-applet. (https://github.com/blueman-project/blueman/wiki/Troubleshooting)

elieobeid7, 2015-07-22:
`blueman-applet`gives

    (blueman-applet:31871): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:2:19: Theming engine 'adwaita' not found
    blueman-applet version 2.0 starting
    There is an instance already running

Though gnome-themes-standard is installed

cschramm, 2015-07-22:
You need to kill the running instance to start a new one, see https://github.com/blueman-project/blueman/wiki/Troubleshooting

elieobeid7, 2015-07-22:
sorry

    blueman-applet:1811): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:2:19: Theming engine 'adwaita' not found
    blueman-applet version 2.0 starting
    Stale PID, overwriting
    /usr/lib/python3.4/site-packages/gi/overrides/__init__.py:313: Warning: The property GSettings:schema is deprecated and shouldn't be used anymore. It will be removed in a future version.
      return super_init_func(self, **new_kwargs)
    _________
    Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
    ['NMDUNSupport', 'PowerManager', 'DhcpClient', 'AppIndicator', 'ExitItem', 'TransferService', 'GameControllerWakelock', 'Menu', 'PPPSupport', 'ShowConnected', 'Networking', 'StandardItems', 'AuthAgent', 'DBusService', 'SerialManager', 'KillSwitch', 'DiscvManager', 'NMPANSupport', 'StatusIcon', 'RecentConns', 'Headset', 'NetUsage'] 
    _________
    Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
    Unable to load plugin module AppIndicator
    cannot import name AppIndicator3, introspection typelib not found 
    _________
    get_interface_version (/usr/lib/python3.4/site-packages/blueman/bluez/BlueZInterface.py:13)
    Detected BlueZ 5 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Headset.Headset'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
    _________
    get_interface_version (/usr/lib/python3.4/site-packages/blueman/bluez/obex/Base.py:20)
    Detected BlueZ integrated obexd 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Menu.Menu'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Networking.Networking'> 
    _________
    load_nap_settings (/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py:36)
    Loading NAP settings 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    Not loading DhcpClient because it's conflict has higher priority 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
    _________
    update_menuitems (/usr/lib/python3.4/site-packages/blueman/plugins/applet/DiscvManager.py:123)
    warning: Adapter is None 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
[... 47 lines cut ...]
    _________
    on_registered (/usr/lib/python3.4/site-packages/blueman/bluez/obex/AgentManager.py:18)
    /org/blueman/obex_agent 
    _________
    RequestPowerState (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:131)
    Requesting True 
    _________
    on_power_state_change_requested (/usr/lib/python3.4/site-packages/blueman/plugins/applet/KillSwitch.py:119)
    True 
    _________
    UpdatePowerState (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:159)
    off True 
    foff False 
    on False 
    current state False 
    new state False 
    _________
    check (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:119)
    callbacks done 
    _________
    set_adapter_state (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:90)
    True 
    _________
    set_adapter_state (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:90)
    Exception occurred org.bluez.Error.Blocked: Blocked through rfkill

elieobeid7, 2015-07-22:
what does bug means? it means no answers and i have to wait till the next release of blueman? Because as far as I see all tickets labeled as bugs don't have answers.

cschramm, 2015-07-22:
I have to look into it, but it seems like it's a bug if blueman is started with a hard kill switch enabled.

On the other hand, how do you trigger the error? It seems like you're trying to connect a service while bluetooth is blocked. That won't work anyway.

elieobeid7, 2015-07-22:
I'm trying to connect to my jbl2 speaker like i always did http://www.jbl.com/bluetooth-speakers/JBL+FLIP+II.html

for some reason, starting yesterday i guess, I'm not able to enable bluetooth through blueman

cschramm, 2015-07-22:
You probably have a hardware kill switch set. The strange thing is that blueman should show a disabled state and you should not be able to connect to anything via the GUI. How do you connect? Via the "Recent Connections" item in the applet menu?

elieobeid7, 2015-07-22:
thx for telling me that, i ran

sudo rfkill unblock bluetooth

and it seems to work

cschramm, 2015-07-22:
I'd still like to take a closer look at this, but unfortunately I don't see a way how you tried to connect to the service.

If bluetooth is unavailable the applet should look like [this](http://i.imgur.com/0LKq9Fv.png) and the manager should show [this](http://i.imgur.com/oEnA3Qo.png).

Can you please describe what you did to produce the error?

elieobeid7, 2015-07-22:
true, it looked like you said, but when i click "enable", blueman listed my devices like so

http://screenshots.ubuntu.com/screenshots/b/blueman/2372_large.png 

but the icon is always like so

http://i.imgur.com/0LKq9Fv.png

so basically it could only list my devices but not enable bluetooth. As for "bluetooth adaptors", when you open it, sometimes it looks normal, sometimes it looks like an empty gtk window,  can't accurately describe it, didn't take pictures.

I'm not so sure how to reproduce it, but I'll take a guess, I have a broadcom BCM4313 network card and it sucks, sometimes it fails to connect to Wifi. so, to fix it, I disable the wifi, then disable the network, then enable both. My guess is, after doing this multiple times, the bluetooth got disabled, and blueman is still running and somehow it couldn't re-enable it.

cschramm, 2015-07-22:
So the manager opens also it failed to enable bluetooth. I'll get that fixed.

cschramm, 2015-07-22:
I found an issue in our rfkill code with Python 3. If you update your local `blueman/plugins/mechanism/RfKill.py` accordingly, it should work. :smiley:

closed 2015-07-22 by cschramm

elieobeid7, 2015-07-22:
awesome!! thank you

commit 6f19e021 (2015-07-22): Fix unblocking rfkill for Python 3

Closes #327
commit 8eb34991 (2015-07-22): Fix unblocking rfkill for Python 3

Closes #327