blueman-project/blueman #329: Pairing prompt is always displayed via dialog instead of notification

monsta, 2015-07-23:
System: LMDE 2 (Linux Mint Debian Edition), based on Debian Jessie
Desktop: MATE 1.10
Blueman version: 2.0-1 (installed from Debian Testing)
BlueZ version: 5.23-2+b1

In order to reproduce https://github.com/mate-desktop/mate-notification-daemon/issues/51, I've installed blueman, borrowed an USB bluetooth adapter from a coworker and tried to pair the adapter with my old phone (Nokia E52). Unfortunately, I can't get pairing prompt to be displayed via notification. I always get the dialog instead, even though the corresponding option is set:
```
monsta@asylum ~ $ gsettings get org.blueman.general notification-daemon
true
```
I've posted the output of ```blueman-applet``` [here](http://pastebin.com/NfEqFbE5). The last error message was shown when I let the pairing prompt dialog exit by timeout.

I've also made a script which tries to imitate the notification process that's happening in blueman code:

```
#!/usr/bin/python

from gi.repository import Notify
from blueman.gui.Notification import Notification

def on_action(n, action):
    print action

print Notify.get_server_caps()

actions = [["ok", _("OK")], ["cancel", _("Cancel")]]

Notification("Test", "test message", 0, actions, on_action)
```
This script showed the notification bubble with two buttons just fine. And its output showed me that server capabitilies include "actions":
```
monsta@asylum ~ $ ./notify_test.py 

** (notify_test.py:9675): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
['actions', 'body', 'body-hyperlinks', 'body-markup', 'icon-static', 'sound']

```

I'm completely confused. :confused:

cschramm, 2015-07-23:
Confuses me as well. I'd add a debug output with values of `Config('org.blueman.general')['notification-daemon']` and `cls.actions_supported()` (and `actions`) to Notification.py:160 to see if any of them is unexpected.

monsta, 2015-07-23:
Ok, so putting this code there:
```python
    def __new__(cls, summary, message, timeout=-1, actions=None, actions_cb=None, pixbuf=None, status_icon=None):
        dprint ("config setting: ", Config('org.blueman.general')['notification-daemon'])
        dprint ("actions: ", actions)
        dprint ("actions supported: ", cls.actions_supported())
        if not Config('org.blueman.general')['notification-daemon'] or (actions and not cls.actions_supported()):
...
```
showed this result:
```
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
config setting:  True 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
actions:  None 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
actions supported:  True 
```

cschramm, 2015-07-23:
:confused: So it should go into the else branch. Can you verify?

If that's the case, I think not blueman, but your notification daemon uses a fallback. Is that possible?

monsta, 2015-07-23:
> So it should go into the else branch. Can you verify?

Yes, it goes to else branch.
```python
        if not Config('org.blueman.general')['notification-daemon'] or (actions and not cls.actions_supported()):
            # Use fallback as user does not want to use a notification daemon or we have to show actions and the
            # notification daemon does not provide them
            dprint ("if branch")
            klass = _NotificationDialog
        else:
            dprint ("else branch")
            klass = _NotificationBubble
```
```
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
else branch 
```

> If that's the case, I think not blueman, but your notification daemon uses a fallback. Is that possible?

Hmm... I don't think that mate-notification-daemon is able to use such fallbacks.

Also I wonder why ```actions``` is ```None``` here. The dialog has a PIN prompt and two buttons - I presume the notification bubble would have the same elements (and handlers for them)?

cschramm, 2015-07-23:
Hrm... `actions` is strange, yeah... Are you sure you're looking at the right notification?

monsta, 2015-07-23:
Well there's only one ```Notification.py``` file, right? I've added debug messages only there and not in any other files.
Before that, it says that it's in ```RequestPinCode```:
```
_________
RequestPinCode (/usr/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py:159)
Agent.RequestPinCode 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
config setting:  True 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
actions:  None 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
actions supported:  True 
_________
__new__ (/usr/lib/python2.7/dist-packages/blueman/gui/Notification.py:159)
else branch 
```
The pairing prompt dialog looks like this:

[image]

cschramm, 2015-07-23:
Oh, that one. That's a dialog, not a notification, no matter what. Not sure if notifications support more complex input than buttons, but at least we do not use it. :wink:

monsta, 2015-07-23:
Oh damn :laughing: 

Hmm but how do I reproduce https://github.com/mate-desktop/mate-notification-daemon/issues/51 then...

> On trying to configure the connection to another device via bluetooth, I am
asked to confirm the connection and password via a notification. The problem
ist that the notification does not stay in the upper right corner, but it is
moving rather quickly to the left. It's so fast I can't read it, I had to make
screenshots of it. But I never managed to click the "confirm" button, which is
in the notification.

Which notification did the reporter mean? I didn't get anything like that...

cschramm, 2015-07-23:
That depends on the type and capabilities of the pairing device and probably also on which side initialized the pairing.

The notification must be the one from `blueman.main.applet.BluezAgent.RequestConfirmation`. You should be able to just hard-code some `alias` and `passkey` and use the rest of the method's code (leaving out the `pixbuf` and `status_icon` parameters).

monsta, 2015-07-24:
Ok thanks, I used that code. I couldn't get the notification bubble to move somewhere as described in that report, but I'll ask some more questions there.
I think this report can be closed now. Thanks for the help. :smile:

closed 2015-07-24 by cschramm