blueman-project/blueman #330: Another bug: blueman isn't starting

elieobeid7, 2015-08-05:
I'm on Manjaro XFCE 64 bit latest version, blueman was working, i received a notification to update my system, can't remember if there was a blueman update, I updated the system and restarted the pc, blueman stops working. blueman-apelet gives,

    ▶ blueman-applet      
    
    (blueman-applet:2052): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:2:19: Theming engine 'adwaita' not found
    blueman-applet version 2.0 starting
    Stale PID, overwriting
    /usr/lib/python3.4/site-packages/gi/overrides/__init__.py:313: Warning: The property GSettings:schema is deprecated and shouldn't be used anymore. It will be removed in a future version.
      return super_init_func(self, **new_kwargs)
    _________
    Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
    ['NMDUNSupport', 'PowerManager', 'DhcpClient', 'AppIndicator', 'ExitItem', 'TransferService', 'GameControllerWakelock', 'Menu', 'PPPSupport', 'ShowConnected', 'Networking', 'StandardItems', 'AuthAgent', 'DBusService', 'SerialManager', 'KillSwitch', 'DiscvManager', 'NMPANSupport', 'StatusIcon', 'RecentConns', 'Headset', 'NetUsage'] 
    _________
    Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
    Unable to load plugin module AppIndicator
    cannot import name AppIndicator3, introspection typelib not found 
    _________
    get_interface_version (/usr/lib/python3.4/site-packages/blueman/bluez/BlueZInterface.py:13)
    Detected BlueZ 5 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    Not loading DhcpClient because it's conflict has higher priority 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Networking.Networking'> 
    _________
    load_nap_settings (/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py:36)
    Loading NAP settings 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Menu.Menu'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.Headset.Headset'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
    _________
    update_menuitems (/usr/lib/python3.4/site-packages/blueman/plugins/applet/DiscvManager.py:123)
    warning: Adapter is None 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
    _________
    get_interface_version (/usr/lib/python3.4/site-packages/blueman/bluez/obex/Base.py:20)
    Detected BlueZ integrated obexd 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    Not loading PPPSupport because it's conflict has higher priority 
    _________
    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
    _________
    _on_obex_owner_changed (/usr/lib/python3.4/site-packages/blueman/plugins/applet/TransferService.py:170)
    obex owner changed: :1.47 
    _________
    on_dbus_name_owner_change (/usr/bin/blueman-applet:94)
    org.bluez owner changed to :1.1 
    _________
    set_nap (/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py:65)
    set nap False 
    _________
    update_menuitems (/usr/lib/python3.4/site-packages/blueman/plugins/applet/DiscvManager.py:123)
    warning: Adapter is None 
    _________
    initialize (/usr/lib/python3.4/site-packages/blueman/plugins/applet/RecentConns.py:144)
    rebuilding menu 
    _________
    enumerate_connections (/usr/lib/python3.4/site-packages/blueman/plugins/applet/ShowConnected.py:50)
    Found 0 existing connections 
    _________
    on_registered (/usr/lib/python3.4/site-packages/blueman/bluez/obex/AgentManager.py:18)
    /org/blueman/obex_agent 
    _________
    RequestPowerState (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:131)
    Requesting True 
    _________
    on_power_state_change_requested (/usr/lib/python3.4/site-packages/blueman/plugins/applet/KillSwitch.py:119)
    True 
    _________
    UpdatePowerState (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:159)
    off True 
    foff False 
    on True 
    current state True 
    new state False 
    _________
    UpdatePowerState (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:159)
    Signalling False 
    _________
    check (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:119)
    callbacks done 
    _________
    set_adapter_state (/usr/lib/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:90)
    True 

when I click on bluetooth adaptors, it asks me if I want to enable bluetooth, I say yes, and it shows this button

http://i.imgur.com/9Hs4hob.png

When I click "bluetooth manager" it looks like this

http://i.imgur.com/Gj5lx0r.png

What's happening now reminds me of this

https://github.com/blueman-project/blueman/issues/327

bluetooth adaptors and bluetooth manager looks the same, back then I forgot to take screenshots, can't tell if the issues are related though, I don't think so.

elieobeid7, 2015-08-05:
after restarting the pc several times and turning off and on the network and wifi, it's working again, I suspect it has to do with rfkill too, but can't tell what's the problem

cschramm, 2015-08-05:
Sounds like #327. Maybe you just lost the https://github.com/blueman-project/blueman/commit/8eb34991bf84b6868f295fa9a705ba0783e749aa change?

elieobeid7, 2015-08-05:
yeah lost it, but noticed that before opening this topic and fixed it, and there was still an error, so i guess there's something else.

elieobeid7, 2015-08-05:
If I have to guess, I'd say it has to do with booting the system, sometimes it works on boot sometimes i have to restart the pc or turn off the network and wifi, so it's not consistent

closed 2015-09-13 by cschramm