blueman-project/blueman #331: No pin requested after dropping BlueZ4

infirit, 2015-08-08:
I am not seeing a pin request for my mobile phone any-more since https://github.com/blueman-project/blueman/commit/c57bd1ecf6f340698066e40ced5894910c6d469e. Is this correct?

On the phone I only get a request to confirm the pair with no pin number to verify. Neither the phone nor blueman seems to care and pair just fine like this.

I notices as I was working on replacing the GtkAlignment in the assistant and thought I messed something up, it turns out.. it wasn't me (this time :smile_cat:).

cschramm, 2015-08-09:
Hmm... That commit should not have changed any behavior for BlueZ 5... :confused:

infirit, 2015-08-09:
Noticed AuthAgent plugin was disabled which may be the reason. But when I enable it I get he below. I'll confirm if this plugin fails also just before and just after the BlueZ4 removal.

```
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py", line 63, in Load
    self.__load_plugin(self.__classes[name])
  File "/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py", line 163, in __load_plugin
    inst._load(self.user_data)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/AppletPlugin.py", line 56, in _load
    self.Applet.DbusSvc.add_definitions(self)
  File "/usr/lib64/python3.4/site-packages/blueman/main/DbusService.py", line 28, in add_definitions
    raise MethodAlreadyExists
blueman.main.DbusService.MethodAlreadyExists
```

infirit, 2015-08-09:
Yep, the plugin fails when I am on c57bd1ecf6f340698066e40ced5894910c6d469e.

cschramm, 2015-08-09:
Can you check the value of `name` in DbusService.py", line 28? Is it `SetTimeHint`?

infirit, 2015-08-09:
> Is it SetTimeHint?

It is indeed.

infirit, 2015-08-09:
The below fixes the problem for me, not sure it is the correct one :smile_cat: 

```diff
diff --git a/blueman/plugins/applet/AuthAgent.py b/blueman/plugins/applet/AuthAgent.py
index 9943ebb..fb87a5c 100644
--- a/blueman/plugins/applet/AuthAgent.py
+++ b/blueman/plugins/applet/AuthAgent.py
@@ -14,10 +14,9 @@ class AuthAgent(AppletPlugin):
     __author__ = "Walmis"
     __depends__ = ["StatusIcon"]
 
-    _agent = None
-    _last_event_time = 0
-
     def on_load(self, applet):
+        self._agent = None
+        self._last_event_time = 0
         self.Applet = applet
 
     @dbus.service.method('org.blueman.Applet', in_signature="u")
```

cschramm, 2015-08-09:
OK... I don't have the slightest idea what's the difference, but if it helps...

infirit, 2015-08-09:
I suspect it just hides the real problem, timing..?

Also disabling/enabling fails for me. When enabled I can disable normally and see the plugin instance being deleted. But then when enabling I see the following. After that I get the same initial traceback..

```python
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py", line 63, in Load
    self.__load_plugin(self.__classes[name])
  File "/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py", line 163, in __load_plugin
    inst._load(self.user_data)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/AppletPlugin.py", line 58, in _load
    self.on_manager_state_changed(applet.Manager != None)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/AuthAgent.py", line 32, in on_manager_state_changed
    self._agent = BluezAgent(self.Applet.Plugins.StatusIcon, self._last_event_time)
  File "/usr/lib64/python3.4/site-packages/blueman/main/applet/BluezAgent.py", line 49, in __init__
    Agent.__init__(self, '/org/blueman/agent/bluez_agent')
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/Agent.py", line 45, in __init__
    dbus.service.Object.__init__(self, dbus.SystemBus(), obj_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 485, in __init__
    self.add_to_connection(conn, object_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 576, in add_to_connection
    self._fallback)
KeyError: "Can't register the object-path handler for '/org/blueman/agent/bluez_agent': there is already a handler"
```

cschramm, 2015-08-10:
That error must mean that the plugin instance (class BluezAgent) is either *not* deleted when disabling or created twice when re-enabling.

In any case, I cannot see any connection to the removal of BlueZ 4 support... :confused:

cschramm, 2015-08-10:
I can confirm the issues on my side.

cschramm, 2015-08-10:
`AuthAgent` calls `del self._agent` in `on_unload`, but `BluezAgent`'s `__del__` is not called (every time). It looks like there's another reference to the object left.

cschramm, 2015-08-10:
This just does not work. I looks like there are references in the dbus handling, so we just need to unregister the agent explicitly instead of just `del`ing it.

infirit, 2015-08-10:
Just to be sure I went back to the commit before c57bd1ecf6f340698066e40ced5894910c6d469e. I moved the agents and last_event_time variables out of on_load and things still work fine after that, loading and unloading also no problems.

Also I noticed that when I have AuthAgent disabled initially I can enable it fine. But it will not load properly when enabled at startup which to me suggest moving the attributes changes the timing and hides the "real" problem.

cschramm, 2015-08-10:
I'll prepare a fix later today.

cschramm, 2015-08-10:
A patch to first unregister the agent explicitly and destroying it afterwards is at https://github.com/blueman-project/blueman/commit/619a236e6102569b19b70a3884df759a9154f4e1, but it does not solve the issues as it looks like the instance still has some reference.

It may solve you're issue with loading the plugin on startup though, at least I cannot reproduce it.

infirit, 2015-08-10:
>It may solve you're issue with loading the plugin on startup though

It does load now when blueman starts, which is most important for 99.9% of users :smile:.

cschramm, 2015-08-10:
I think I got it resolved now: The call to remove_from_connection was removed when dropping BlueZ 4. Please try https://github.com/blueman-project/blueman/commit/d3c9856e847e121e5b8d7184f631b1e6a04442c8.

infirit, 2015-08-10:
All is well now :+1:

closed 2015-08-10 by cschramm