blueman-project/blueman #333: blueman-applet hanging with 'Bad rfkill event size'

mgorny, 2015-08-08:
After starting blueman-applet with current git (e657a16d10262d818278c6946472083e312750c6):

```
blueman-applet version 2.0 starting
Stale PID, overwriting
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
['ExitItem', 'ShowConnected', 'AuthAgent', 'DhcpClient', 'NMPANSupport', 'DBusService', 'DiscvManager', 'NetUsage', 'Headset', 'KillSwitch', 'Menu', 'Networking', 'PowerManager', 'PPPSupport', 'NMDUNSupport', 'RecentConns', 'SerialManager', 'StandardItems', 'StatusIcon', 'TransferService', 'GameControllerWakelock', 'AppIndicator'] 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/Networking.py:34)
Loading NAP settings 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because its conflict has higher priority 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
Failed to load AuthAgent
_agent 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because its conflict has higher priority 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:112)
warning: Adapter is None 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
on_dbus_name_owner_change (/usr/bin/blueman-applet:86)
org.bluez owner changed to :1.2 
_________
set_nap (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/Networking.py:63)
set nap False 
_________
initialize (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:138)
rebuilding menu 
_________
_on_obex_owner_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/TransferService.py:172)
obex owner changed: :1.40 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
enumerate_connections (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/ShowConnected.py:54)
Found 0 existing connections 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
on_registered (/usr/lib64/python2.7/site-packages/blueman/bluez/obex/AgentManager.py:15)
/org/blueman/obex_agent 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:72)
Bad rfkill event size 
[...]
```

with new 'Bad rfkill event size' events appearing very quickly, practically hanging it up completely.

blueman: git (e657a16d10262d818278c6946472083e312750c6)
BlueZ: 5.33
Distribution: Gentoo
Desktop environment: Xfce 4.12
Kernel: 4.1.0-pf1

mgorny, 2015-08-08:
Curious enough, this doesn't happen when blueman-applet is started first by Xfce's session manager. Only when i kill that instance and start another one on the console. Tried restarting bluetoothd but then bluetoothd hangs... looks suspicious.

cschramm, 2015-08-09:
So bluetoothd has the same issue? Sounds like something's going really wrong on your system then...

Could you try some older commit (like the 2.0 release) and maybe bisect where the issue shows up if it's not reproducible with that commit?

mgorny, 2015-08-09:
Well, got some more info now. However, I believe that above all blueman should be able to handle the 'Bad rfkill event size' error more gracefully than getting into deadloop reading from /dev/rfkill.

First things first, my device is some old Broadcom USB dongle:

```
Bus 002 Device 004: ID 0a5c:2001 Broadcom Corp. Bluetooth Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.01
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0a5c Broadcom Corp.
  idProduct          0x2001 Bluetooth Device
  bcdDevice            0.a0
  iManufacturer           1 Broadcom Corp.      
  iProduct                2 BCM2033             
  iSerial                 3 20330006AB806EC1
  bNumConfigurations      1
[...]
```

I can paste the full `lsusb -v` output if you want but it's huge and doesn't seem really relevant. The point is, I doubt the device really has any kind of rfkill.

Some observations:
* on bootup `/dev/rfkill` does not exist.
* starting bluetoothd doesn't create it.
* starting blueman causes `/dev/rfkill` to be created.
* once `/dev/rfkill` is created, both blueman & bluetoothd deadloop on restart reading from it.
* if I remove `/dev/rfkill`, both restart fine again. However, the device gets recreated after restarting either.

So my guess would be: blueman causes bluez to create `/dev/rfkill` but does not use it in this case. However, after restarting either of them, the device gets used and produces cruft. And I suspect it produces cruft because I have no rfkill devices.

mgorny, 2015-08-09:
Oh, one minor note: when first starting blueman after boot, the icon says that rfkill is blocked. However, after restarting either blueman or bluetoothd, the icon says it is unblocked.

cschramm, 2015-08-09:
We do `open('/dev/rfkill', 'w')` to disable a killswitch, so blueman would create it if the file does not exist. See #17 for a discussion of what causes /dev/rfkill to be absent.

You're using Gentoo, so I guess you configured and built your kernel yourself, right? Can you confirm it lacks the rfkill subsystem?

The question then is if blueman is able to work without (or if we could make it so). Please try to disable the PowerManager plugin and see where that gets you.

mgorny, 2015-08-09:
Ok, so I see now where the problem lies. I didn't notice `/dev/rfkill` is actually created as a regular file rather than a device node. So that's where the problem lies, and what breaks other software. And yes, I have the rfkill subsystem disabled because obviously I don't have any rfkill devices. And yes, blueman works fine with lack of/broken rfkill, as long as it doesn't try to `open()` the regular file with some data in it.

commit 7da2dcd1 (2015-08-10): Handle when rfkill subsystem is not available gracefully

Check for the /dev/rfkill device and raise ImportError if not
found to stop the plugins from loading.

fixes #333
commit 311e2c36 (2015-08-19): RfKill: open /dev/rfkill r+b to avoid creating

Open /dev/rfkill in r+b mode rather than wb since the latter creates the
file if it does not exist. And we certainly can not create /dev/rfkill
as a regular file since this is never going to work and is going to
break a lot of other programs that assume /dev/rfkill is a valid device
node.

Fixes #333
Closes #336
commit 5662bc61 (2015-10-05): RfKill: open /dev/rfkill r+b to avoid creating

Open /dev/rfkill in r+b mode rather than wb since the latter creates the
file if it does not exist. And we certainly can not create /dev/rfkill
as a regular file since this is never going to work and is going to
break a lot of other programs that assume /dev/rfkill is a valid device
node.

Fixes #333
Closes #336
commit fa8c0ba1 (2015-10-05): Handle when rfkill subsystem is not available gracefully

Check for the /dev/rfkill device and raise ImportError if not
found to stop the plugins from loading.

fixes #333
closed 2015-08-10 by infirit