blueman-project/blueman #339: ShowConnected does not change icon

infirit, 2015-08-10:
Class ``PropertiesBase`` emits the the ``property-changed`` signal just fine as I see them in the log. But then for some reason the signal is lost between ``PropertiesBase`` -> ``Device`` -> ``ShowConnected``. The problem was introduced in 4d12346508adf224ad25a484ee0ae9623c4410ad and 969228c1cbef651616e125b8394db43e7c70ce8f does not solve the problem for ``ShowConnected``. The signal connects properly in ``ShowConnected`` so we are connecting to the correct signal I believe.

On startup the state is properly read in  ``enumerate_connections`` and the icon is updated accordingly.

I am at a loss why the signal is lost :confused:, perhaps others know what is going on?

infirit, 2015-08-10:
I can connect to the ``property-changed`` signal and have it do something in the ``Device`` class which confuses me even more.. Where is the signal gone to in ``ShowConnected``.

cschramm, 2015-08-10:
bb823ab :smiley:

infirit, 2015-08-10:
Perfect :smiley_cat:

closed 2015-08-10 by infirit