blueman-project/blueman #341: SerialManager: AttributeError: 'Device' has no attribute 'Address'

yars068, 2015-08-14:
Hi now,
As you noticed, at end of [these log](http://pastebin.com/1Se9GJk7), seen 
```
_________
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:28)
None Connected 0
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 59, in _on_device_property_changed
    self.terminate_all_scripts(device.Address)
AttributeError: 'Device' object has no attribute 'Address'
```
That is unrelated to #340, and for that reason, is another issue. Currently, I have no idea on how to fix it :(

cschramm, 2015-08-15:
We possibly have issues wherever https://github.com/blueman-project/blueman/commit/4d12346508adf224ad25a484ee0ae9623c4410ad drops `path_keyword` and starts relying on the object provided by the glib signal. If this is used on some "Any" object which does not have a path, it also does not have a path and in consequence no properties when the handler is called.

I identified issues at the following places:

BluemanApplet._on_adapter_property_changed
GameControllerWakelock._on_device_property_changed
NetUsage._on_network_property_changed
SerialManager._on_device_property_changed

So it looks like DiscvManager, GameControllerWakelock, NetUsage, and SerialManager are actually broken in master.

cschramm, 2015-08-15:
I hacked a patch. Completely untested yet. https://github.com/blueman-project/blueman/commit/98abc016842ad0e537ba40059ee84bb9f532cdd2

yars068, 2015-08-16:
Please see the [log](http://pastebin.com/SMiu04qh) with having this patch. Sending and receiving still works, but we have some tracebacks.

cschramm, 2015-08-18:
OK, I just checked the patch. It makes those handlers work correctly and fixes the `Address` property issue. I could not replicate your new traceback (I think the second one is just the consequence of the first one). How did you create it? Just by discovering a new device? Also, are you sure that it does not exit without my patch?

yars068, 2015-08-18:
It appeared whien I tried send/receive file. That's works but only one time, then I get `AttributeError: 'NoneType' object has no attribute 'get_path'`. With my patch this is not appeared, because I  tried sending and receiving many times. With my patch, I have only UnicodeErrors when I tried with a file that have non-ASCII name (characters is out of range(128)).

cschramm, 2015-08-18:
So that exception also occurs in current master?

> With my patch this is not appeared

Your code breaks and as a consequence disables the SerialManager plugin, so just disabling it should have the same effect then. Unfortunately I cannot spot any connection between the traceback and the SerialManager plugin... :confused:

yars068, 2015-08-18:
With current master SerialManager works, but I have
```
Destination file exists, renaming to: 20150818130152_05 - A Place for my Head.mp3 
adding action
_________
_on_properties_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/obex/Transfer.py:26)
/org/bluez/obex/server/session3/transfer2 Transferred 0 
_________
_on_properties_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/obex/Transfer.py:26)
/org/bluez/obex/server/session3/transfer2 Status complete 
_________
_on_properties_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/obex/Transfer.py:26)
/org/bluez/obex/server/session3/transfer2 Transferred 0 
_________
_on_session_removed (/usr/lib64/python2.7/site-packages/blueman/bluez/obex/Manager.py:45)
/org/bluez/obex/server/session3 
_________
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:28)
None Connected 0 
_________
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:28)
None Connected 0 
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 59, in _on_device_property_changed
    self.terminate_all_scripts(device.Address)
AttributeError: 'Device' object has no attribute 'Address'
_________
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:28)
/org/bluez/hci0/dev_C4_43_8F_28_B2_98 Connected 0 
_________
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:28)
/org/bluez/hci0/dev_C4_43_8F_28_B2_98 Connected 0 
```
when tried to receive file.

cschramm, 2015-08-18:
Sounds like there's a bug in my patch then... Strange that I cannot replicate this...

Maybe the issue was even covered by the Address property issue.

Can you try replacing the `.Address` call in master's `SerialManager.py` with `.get_properties()['Address']`. I think this should also fix the issue. Let's see if you get the `get_path` error then.

yars068, 2015-08-18:
Yes, that solves the issue, when I receive a file from phone, but when I send file to phone, I get [this](http://pastebin.com/QiSDFc1X).

cschramm, 2015-08-18:
So it's not related to my patch. I'll add it to master then and look into your issue.

commit d8a61134 (2015-08-15): Add path to property-changed signal

Fixes #341
closed 2015-08-19 by cschramm