blueman-project/blueman #345: Missing manual PIN entry when pairing

mr700, 2015-08-17:
I have a bluetooth printer and it has predefined fixed key. I can only "Use Random Key" or "Proceed Without Pairing". Blueman used to have manual PIN input, that's how I used my GPS receiver. Please add this option again.

cschramm, 2015-08-18:
This slipped through, but it should work although it's irritating. If you choose the random passkey option or just pair in the manager window without using the setup dialog you should get a request for the PIN code (the AuthAgent plugin needs to be enabled).

I'll rework the scheme in the setup dialog to improve this.

Alecs69, 2015-11-15:
Same problem. On the headset fixed PIN and does not provide manual input. AutnAgent included. When connecting the keyboard is also a problem, since it is necessary to dial PIN on the keyboard. But what key was sent is not known.

closed 2015-11-26 by cschramm

zarkone, 2021-06-13:
`bluetoothctl` shows this number -- so in my case, I used both tools because I couldn't enter commands in bluetoothctl without keyboard: 

[image]

commit 4fc3a8fa (2015-11-26): [assistant] Rename "random passkey" to generic "pair"

Closes #345