blueman-project/blueman #346: AttributeError: 'NoneType' object has no attribute 'get_iter'

cschramm, 2015-08-19:
Follow-up for the issue reported by @yars068 in #341.

````
File "blueman/gui/DeviceList.py", line 410, in find_device_by_path
    iter = self.get_model().get_iter(path)
AttributeError: 'NoneType' object has no attribute 'get_iter'
````

Seen reportedly when sending a file.

yars068, 2015-08-21:
I noticed that error appears when I send file to device using status icon (right click on blueman's tray icon > Send file to  device...), but not when I doing that using blueman-manager (in that case all works fine).

infirit, 2015-10-08:
Confirmed, I had a quick look and it seems the model does not exist any-more after the screen with the devicelist. I remember this working when I tried it last. I'll try to bisect it and see where it got broken.

cschramm, 2015-10-14:
@infirit: Found anything? I have no idea why the model gets lost there...

infirit, 2015-10-14:
It goes back a long time. I think the reason is that we explicitly destroy the ``DeviceSelectorDialog`` which contains the devicelist after we selected the device.

@yars068 can you try if the below works?

```diff
diff --git a/apps/blueman-sendto b/apps/blueman-sendto
index 3f99d1e..2a76680 100755
--- a/apps/blueman-sendto
+++ b/apps/blueman-sendto
@@ -364,7 +364,6 @@ class SendTo:
     def select_device(self):
         d = DeviceSelectorDialog()
         resp = d.run()
-        d.destroy()
         if resp == Gtk.ResponseType.ACCEPT:
             sel = d.GetSelection()
             if sel:
```

infirit, 2015-10-14:
It fixed the problem for me so I pushed it as 50940ff7dfa16d8fa56103cf5d6f5748d20fe728. If it causes undesirable side effects let me know.

closed 2015-10-14 by cschramm

yars068, 2015-10-20:
@infirit > It fixed the problem for me

Confirmed.