blueman-project/blueman #347: Functions: dprint fails on data type conversion

yars068, 2015-08-19:
The issue is in `s += (str(a) + " ")`. So, that lead to that error:
```
_on_action (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/TransferService.py:48)
accept 
var 'a' is <type 'unicode'>
var 'a' is <type 'unicode'>
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/gui/Notification.py", line 137, in on_action
    actions_cb(n, action)
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/TransferService.py", line 57, in _on_action
    self._agent.reply(self.transfers[self._pending_transfer['transfer_path']]['path'])
  File "/usr/lib64/python2.7/site-packages/blueman/bluez/obex/Agent.py", line 67, in reply
    dprint(self._agent_path, reply)
  File "/usr/lib64/python2.7/site-packages/blueman/Functions.py", line 69, in dprint
    s += (str(a) + " ")
UnicodeEncodeError: 'ascii' codec can't encode characters in position 24-29: ordinal not in range(128)
```

yars068, 2015-08-19:
Solution is 4c8d482f.

infirit, 2015-08-19:
That solution will fail on python3 where the ``unicode`` datatype does not exist. How does this work for you?

```diff
diff --git a/blueman/Functions.py b/blueman/Functions.py
index b34ccb9..1b1b9c9 100644
--- a/blueman/Functions.py
+++ b/blueman/Functions.py
@@ -65,7 +65,7 @@ def dprint(*args):
 
         s = ""
         for a in args:
-            s += (str(a) + " ")
+            s += ("%s " % a)
         co = sys._getframe(1).f_code
 
         fname = BOLD(co.co_name)
```

yars068, 2015-08-19:
That works fine, good :+1:

infirit, 2015-08-19:
>But for what reason was used ```str``` here?

It will retrieve the object string representation (it's __str__) and/or convert codepoints to bytes. 

```python
>>> [1,2,3,4,5].__str__()
'[1, 2, 3, 4, 5]'
>>> str([1, 2, 3, 4, 5])
'[1, 2, 3, 4, 5]'
```

yars068, 2015-08-19:
Thanks you for consult.

infirit, 2015-08-20:
>Maybe we need to use __repr__() method instead of str() call?

No, repr is something different and not what we want to see, below an example.

``python
>>> 'abc'.__repr__()
"u'abc'"
``

closed 2015-08-20 by infirit

infirit, 2015-08-20:
>possibly it can cause problem when dprint will get the non-string data

Any object will return a string and if not implemented __explicitly__ you will get something like ``MyClass`` below. It is up to the object/class to override the ``__str__`` method with something useful, for example ``MyClassWithStr``.

```python
>>> class Myclass(object):
	pass

>>> x = Myclass()
>>> x.__str__()
'<__main__.Myclass object at 0x7f06260bd550>'

>>> class MyclassWithStr(object):
	__name__ = "MyClassWithStr"
	def __str__(self):
		return self.__name__

	
>>> y = MyclassWithStr()
>>> y.__str__()
'MyClassWithStr'
```

commit 7a2e20e4 (2015-08-20): Functions: Do not call str to get string representation

Instead let python handle it with string formatting and avoid
UnicodeEncodeError.

fixes #347
commit 2ed9dc1d (2015-10-05): Functions: Do not call str to get string representation

Instead let python handle it with string formatting and avoid
UnicodeEncodeError.

fixes #347