blueman-project/blueman #348: Bluetooth not visible in Linux mint

AnksArora, 2015-08-21:
I am able to start my bluetooth but its not visible to any of outside world's devices. Configuration file also has right settings
# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

cschramm, 2015-08-21:
Please issue the `show` command in `bluetoothctl` and see what value the Discovery property shows.

...and please fix your formatting here.

yars068, 2015-08-21:
Please issue the blueman-applet's version. I have blueman-applet 2.0, so it have that settings. Do left click on it's tray icon,  then in "Bluetooth Devices" window go to "Adapter" > "Preferences" and select preferred discoverable timeout. It is possible to select "Hide", "Always visible" and "Temporary visible", in the last case it is possible to set preferred timeout up to 30 minutes.

AnksArora, 2015-08-24:
[image]
No adapters found when i try to set up a new device for file transmission..is there any package or plugin missing ??

yars068, 2015-08-24:
To perform operations with bluetooth, your bluetooth adapter should be supported by BlueZ and Linux kernel (and also, the kernel driver may need a firmware), so check the appropriate packages. That is unrelated to blueman as it is a user-space application.

infirit, 2015-08-24:
@AnksArora, see https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth

It sound like your adapter is not recognized by BlueZ. If so there is very little we can do in blueman.

closed 2015-09-13 by cschramm