blueman-project/blueman #350: segfault in latest Fedora release (git0a5defd)

ashleysommer, 2015-08-24:
Just updated to blueman version `2.1-0.1.git0a5defd.fc23` in Fedora and immediately started getting this segfault when attempting to pair any device, or marking any device as Trusted.

Downgrading back to previous version `2.0-11.fc23` fixed the issue.

```
Aug 24 10:54:55 ashxps systemd-coredump[2195]: Process 2181 (blueman-manager) of user 1000 dumped core.                                               
                                               Stack trace of thread 2181:
                                               #0  0x00007fda12d031d2 invalid_closure_notify (libgobject-2.0.so.0)
                                               #1  0x00007fda12cf1b30 g_closure_invalidate (libgobject-2.0.so.0)
                                               #2  0x00007fda12cf175b g_closure_unref (libgobject-2.0.so.0)
                                               #3  0x00007fda12d036c5 g_signal_handler_disconnect (libgobject-2.0.so.0)
                                               #4  0x00007fda1241bd30 ffi_call_unix64 (libffi.so.6)
                                               #5  0x00007fda1241b79b ffi_call (libffi.so.6)
                                               #6  0x00007fda1189b384 pygi_invoke_c_callable (_gi.so)
                                               #7  0x00007fda1189cf18 pygi_function_cache_invoke (_gi.so)
                                               #8  0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #9  0x00007fda1e0e6560 PyEval_EvalFrameEx (libpython2.7.so.1.0)
                                               #10 0x00007fda1e0e96b4 PyEval_EvalCodeEx (libpython2.7.so.1.0)
                                               #11 0x00007fda1e0e85c6 PyEval_EvalFrameEx (libpython2.7.so.1.0)
                                               #12 0x00007fda1e0e96b4 PyEval_EvalCodeEx (libpython2.7.so.1.0)
                                               #13 0x00007fda1e0755ac function_call (libpython2.7.so.1.0)
                                               #14 0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #15 0x00007fda1e0e2ac7 PyEval_CallObjectWithKeywords (libpython2.7.so.1.0)
                                               #16 0x00007fda1189a800 pygi_signal_closure_marshal (_gi.so)
                                               #17 0x00007fda12cf20b5 g_closure_invoke (libgobject-2.0.so.0)
                                               #18 0x00007fda12d04499 signal_emit_unlocked_R (libgobject-2.0.so.0)
                                               #19 0x00007fda12d0d1a0 g_signal_emit_valist (libgobject-2.0.so.0)
                                               #20 0x00007fda12d0d54f g_signal_emit (libgobject-2.0.so.0)
                                               #21 0x00007fda10b1d324 gtk_menu_shell_button_release (libgtk-3.so.0)
                                               #22 0x00007fda10afc1da _gtk_marshal_BOOLEAN__BOXEDv (libgtk-3.so.0)
                                               #23 0x00007fda12cf22e4 _g_closure_invoke_va (libgobject-2.0.so.0)
                                               #24 0x00007fda12d0ca43 g_signal_emit_valist (libgobject-2.0.so.0)
                                               #17 0x00007fda12cf20b5 g_closure_invoke (libgobject-2.0.so.0)
                                               #18 0x00007fda12d04499 signal_emit_unlocked_R (libgobject-2.0.so.0)
                                               #19 0x00007fda12d0d1a0 g_signal_emit_valist (libgobject-2.0.so.0)
                                               #20 0x00007fda12d0d54f g_signal_emit (libgobject-2.0.so.0)
                                               #21 0x00007fda10b1d324 gtk_menu_shell_button_release (libgtk-3.so.0)
                                               #22 0x00007fda10afc1da _gtk_marshal_BOOLEAN__BOXEDv (libgtk-3.so.0)
                                               #23 0x00007fda12cf22e4 _g_closure_invoke_va (libgobject-2.0.so.0)
                                               #24 0x00007fda12d0ca43 g_signal_emit_valist (libgobject-2.0.so.0)
                                               #25 0x00007fda12d0d54f g_signal_emit (libgobject-2.0.so.0)
                                               #26 0x00007fda10c52c14 gtk_widget_event_internal (libgtk-3.so.0)
                                               #27 0x00007fda10af936e propagate_event (libgtk-3.so.0)
                                               #28 0x00007fda10afb1a8 gtk_main_do_event (libgtk-3.so.0)
                                               #29 0x00007fda106453d2 gdk_event_source_dispatch (libgdk-3.so.0)
                                               #30 0x00007fda129f2e9a g_main_context_dispatch (libglib-2.0.so.0)
                                               #31 0x00007fda129f3230 g_main_context_iterate.isra.29 (libglib-2.0.so.0)
                                               #32 0x00007fda129f3552 g_main_loop_run (libglib-2.0.so.0)
                                               #33 0x00007fda10afa355 gtk_main (libgtk-3.so.0)
                                               #34 0x00007fda1241bd30 ffi_call_unix64 (libffi.so.6)
                                               #35 0x00007fda1241b79b ffi_call (libffi.so.6)
                                               #36 0x00007fda1189b384 pygi_invoke_c_callable (_gi.so)
                                               #37 0x00007fda1189cf18 pygi_function_cache_invoke (_gi.so)
                                               #38 0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #39 0x00007fda1e0e6a68 PyEval_EvalFrameEx (libpython2.7.so.1.0)
                                               #40 0x00007fda1e0e96b4 PyEval_EvalCodeEx (libpython2.7.so.1.0)
                                               #41 0x00007fda1e0755ac function_call (libpython2.7.so.1.0)
                                               #42 0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #43 0x00007fda1e05f95c instancemethod_call (libpython2.7.so.1.0)
                                               #44 0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #45 0x00007fda1e0e2ac7 PyEval_CallObjectWithKeywords (libpython2.7.so.1.0)
                                               #46 0x00007fda1e060660 PyInstance_New (libpython2.7.so.1.0)
                                               #47 0x00007fda1e050b03 PyObject_Call (libpython2.7.so.1.0)
                                               #48 0x00007fda1e0e6a68 PyEval_EvalFrameEx (libpython2.7.so.1.0)
                                               #49 0x00007fda1e0e96b4 PyEval_EvalCodeEx (libpython2.7.so.1.0)
                                               #50 0x00007fda1e0e97d9 PyEval_EvalCode (libpython2.7.so.1.0)
                                               #51 0x00007fda1e102bdf run_mod (libpython2.7.so.1.0)
                                               #52 0x00007fda1e103db2 PyRun_FileExFlags (libpython2.7.so.1.0)
                                               #53 0x00007fda1e104fc7 PyRun_SimpleFileExFlags (libpython2.7.so.1.0)
                                               #54 0x00007fda1e1171e1 Py_Main (libpython2.7.so.1.0)
                                               #55 0x00007fda1d33f580 __libc_start_main (libc.so.6)
                                               #56 0x0000555f02c34839 _start (python2.7)
```

cschramm, 2015-08-24:
I think the issue is in blueman/gui/DeviceList.py, line 309 `existing_dev.disconnect_signal('property-changed')`. Could you try commenting that (`# existing_dev.disconnect_signal('property-changed')`) and see if it still crashes?

ashleysommer, 2015-08-25:
@cschramm Thanks for your suggestion.
I commented that line in blueman/gui/DeviceList.py as you suggested, but I still get the same failure, with the same stacktrace.

I updated to the next new fedora release 2.1-0.2.git7a2e20e.fc23 and it has the same problem.

cschramm, 2015-08-26:
OK, then I cannot guess where it's coming from.

Please run the program in gdb and issue `py-bt` to get the Python stack when it crashes. I think you need to have a package called python-debuginfo.

closed 2015-09-13 by cschramm