blueman-project/blueman #351: Awkwardness in getting connected to a paired device

DavidGriffith, 2015-08-26:
I used to be able to pair my S5 with my laptop and then whenever I wanted to send over photos, I'd turn on bluetooth for both devices and share the files over.  Now it seems that I must manually tell my laptop to connect with the S5 or else the S5 will time out.  It's like this: start bluetoothctl, then type "connect 4C:XX:XX:XX:XX:XX".  Then the Blueman rune will turn green and I can proceed.  I don't know if I'm being dense or what, but shouldn't Blueman have an option to explicitly connect to a known device so I can avoid monkeying with bluetoothctl.

As always, I'm using the Master branch from the repo.

cschramm, 2015-08-26:
What do you see when you right-click on the device in blueman-manager? What UUIDs are reported for the device (should be in the blueman-applet output)?

DavidGriffith, 2015-08-26:
Audio Source, Audio Profile, Send a File, Pair, Untrust, Setup, Rename Device, Remove, Disconnect.

Selecting "Audio Source" does work, but I don't want audio shared.  I cannot get this phone connected without also exporting its audio to the laptop running Blueman.

It reports this:
_on_property_changed (/usr/local/lib/python2.7/dist-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_4C_A5_6D_76_70_8C Connected 0

cschramm, 2015-08-26:
> Selecting "Audio Source" does work, but I don't want audio shared. I cannot get this phone connected without also exporting its audio to the laptop running Blueman.

That is, unfortunately, a shortcoming of BlueZ 5. If you use the generic connect method a specific subset of services is connected at once and everything more specific is totally unusable at least via the D-Bus API.

As a consequence, service entries like "Audio Source" just connect all those basic services. `connect 4C:A5:6D:76:70:8C` in `bluetoothctl` should do the same (audio is connected too then, isn't it?).

If you cannot spot the UUIDs in the output, you can also see them with `info 4C:A5:6D:76:70:8C` in `bluetoothctl`.

What happens if you try to send a file without connecting?

DavidGriffith, 2015-08-26:
Selecting "Audio Source" does connect everything up.  Is this a bug in BlueZ 5, a misfeature, or what?

Ah, here's the UUID: 00001105-0000-1000-8000-00805f9b34fb

infirit, 2015-08-26:
I don't see this with my galaxy nexus. It connects automatically and disconnects when done. It does not seem to matter if the phone is paired or trusted, the same (dis)connect behaviour.. BlueZ 5.33

cschramm, 2015-08-26:
It's an unfortunate design choice that BlueZ 5 only (really) supports connecting "everything" up. It actually gets much more unfortunate as "everything" does not mean everything here (see example below).

The reason we have service specific entries in our menu is a) historic as it was different with BlueZ 4 and b) due to the face that everything does not mean everything in BlueZ 5. Here's an example:

Menu item | BlueZ 4 | BlueZ 5
--- | --- | ---
audio source | audio source | "everything" (here: audio source and input)
input | input | "everything" (here: audio source and input)
network | network | network

As blueman 2.1 will not support BlueZ 4 anymore, we could think of melting "audio source" and "input" from the example into a single item. That actually fits the BlueZ 5 awkwardness much better as it's not really possible to get the state for a single service (we only get a single binary state for "everything" even if only a subset is actually connected), so the current entries may be wrong if a lower layer decides to do something unexpected.

Regarding your issue, the UUID you gave is an OBEX service, so sending files should work. I guess that's already exposed in unconnected state as otherwise e.g. an audio service would be present too. So I still expect it to work without connecting. Again: What happens if you try to send a file without connecting?

DavidGriffith, 2015-08-26:
If I try to send a file without connecting, I get "File not sent." on the phone.  The laptop doesn't even notice.

cschramm, 2015-08-26:
That sounds like a signal handling issue then (it may of course also be an issue of the phone). Anything happening in the blueman-applet output when you try to send? How about the other way (sending a file with blueman)?

DavidGriffith, 2015-08-26:
When I try to send from the phone without explicitly connecting and it fails, nothing happens.  And now for more confusion.  I just now logged out and logged back into my laptop.  Now if I try to send without explicitly connecting, it works.  After a few hours, then it stops working.  I've tried putting the laptop in and out of suspent and toggle bluetooth to get it to stop working, but I'm at a loss to describe just when and how it stops working.

cschramm, 2015-08-26:
Please check if obexd is running when it does not work. It should be started by blueman-applet via D-Bus (if not already running) and never terminate. You can also check if restarting one of blueman-applet / bluetoothd / obexd has the same effect as logging in again.

closed 2015-12-06 by cschramm