blueman-project/blueman #352: Merge menu entries of basic services

cschramm, 2015-08-26:
See https://github.com/blueman-project/blueman/issues/351#issuecomment-134919867

With BlueZ 5 audio and input service items actually do the same and do not necessarily report the state of the specific service, so they should probably be merged into a single menu item, displaying the service name and providing to connect / disconnect depending on the device state.

infirit, 2015-10-28:
There is a ``ConnectProfile(string uuid)`` option in the device-api, maybe this is the one we should use to connect specific things?

cschramm, 2015-10-28:
That method is unfortunately useless by design, see http://marc.info/?l=linux-bluetooth&m=142038630619231

infirit, 2015-10-28:
Hmm, I see. So if I understand correctly, we need a generic "connect" and another for NAP if available (eg a phone)? Anything else is up to other application  to manage.

cschramm, 2015-10-29:
Nearly. The network services' connect method is in blueman.services.meta.NetworkService. There's also a special method in blueman.services.meta.SerialService. All other services use the generic connect method in blueman.Service. They currently all have their own menu entry as they were connected and disconnected individually with BlueZ 4, but now it's not possible to get the services' state anymore and it would fit the BlueZ 5 way better to use just a single item for them.

infirit, 2017-01-21:
PR #575

infirit, 2017-05-13:
Got a cheap(ish) 4.0 mouse from logitech and I have no connect button.

```
00001800-0000-1000-8000-00805f9b34fb Generic Access
00001801-0000-1000-8000-00805f9b34fb Generic Attribute
0000180a-0000-1000-8000-00805f9b34fb Device Information
0000180f-0000-1000-8000-00805f9b34fb Battery Service
00001812-0000-1000-8000-00805f9b34fb Human Interface Device
00010000-0000-1000-8000-011f2000046d Proprietary
```

For some reason we are not catching 1812, reopening for now so I don't forget to look at this.

cschramm, 2017-05-13:
#699 I guess?

infirit, 2017-05-13:
Yup 😄

closed 2017-05-13 by infirit