blueman-project/blueman #355: open /dev/rfkill r+b results in an error

gsxj2014, 2015-09-09:
Distro : Arch Linux
```
$ uname -r
4.1.6-1-ARCH
```

```
Python 3.4.3 (default, Mar 25 2015, 17:13:50) 
[GCC 4.9.2 20150304 (prerelease)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open('/dev/rfkill', 'r+b')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
io.UnsupportedOperation: File or stream is not seekable.
```

cschramm, 2015-09-13:
What type does the file have? For me it's character special file (`c` in `ls -l`). Please try `open('/dev/rfkill', 'r+b', buffer=0)`.

gsxj2014, 2015-09-15:
```
$ ls -l /dev/rfkill 
crw-rw-r-- 1 root rfkill 10, 55 Sep 15 07:12 /dev/rfkill
```

`open('/dev/rfkill', 'r+b', buffering=0)` seems to work

```
Python 3.4.3 (default, Mar 25 2015, 17:13:50) 
[GCC 4.9.2 20150304 (prerelease)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> open('/dev/rfkill', 'r+b', buffering=0)
<_io.FileIO name='/dev/rfkill' mode='rb+'>
```

commit 2a812a86 (2015-09-15): Disable buffering for /dev/rfkill

Fixes #355
commit 8307c744 (2015-10-05): Disable buffering for /dev/rfkill

Fixes #355
closed 2015-09-15 by cschramm