blueman-project/blueman #358: Device was removed once and it doesn't appear anymore

HumanBass, 2015-09-22:
Mint 17.2

I have a bose mini soundlink that was detected just fine initally. I synced but there was some kind of error message that made me want to it again from start. So I removed...and then it doesn't appear anymore even when I put in detection mode. My cellphone anlid other computer detects it just fine, except my mint machine. It is like it was blacklisted. Is there any way reset this "blacklist"?

cschramm, 2015-09-23:
There's no such thing in blueman and as far as I know it's the same for BlueZ. You could check `bluetoothctl` during discovery to see what's going on on the BlueZ layer. If the device shows up there, it's a blueman issue.

closed 2015-10-05 by cschramm

Noino, 2018-03-01:
I had this problem. Fixed by removing the file 
/var/lib/bluetooth/<bt mac>/cache/<device mac>