blueman-project/blueman #359: bluetooth devices no longer discovered since wifi support enabled on Lenovo G45

eltonwoo, 2015-09-29:
I am an end user reporting the above problem. Previously, I was able to pair a bluetooth mouse on my machine. However, the Lenovo's Realtek / Qualcomm card did not support wifi. After successfully enabling wifi, I am no longer able to detect any bluetooth devices (as a comparison, I can detect these same devices with my Asus EEE 1000 PC running the same  distro. Both machines have all current updates.

Details:
"problem" machine: *Lenovo 45-80(80E301GUUS)*
OS: Linux Mint 17.2  'Rafaela"
Desktop: XFCE / Gnome: 3.8.4
Kernel: 3.16.0-38-generic
Blueman Device Manager 1.23
Network controller: Qualcomm Atheros D evice 0041 (rev. 20) / Lenovo Device 3545
Ethernet controller: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 10) / Lenovo Device 3812

Previously the Realtek card in this machine would not connect via wifi. Now that the problem is finally resolved, bluetooth does not seem to function. It scans, but finds nothing. I have various bluetooth devices (mice, keyboard, smartphone,headset,netbook, none are discovered. 

Link to the thread: http://forums.linuxmint.com/viewtopic.php?f=53&t=204566

I would be glad to provide any further system information if requested.

respectfully
elton[dot]woo[at]gmail[dot]com

cschramm, 2015-09-30:
Just to mention it: This is not a blueman issue, but BlueZ.

Do you see the controller in bluetoothctl?

How did you solve your WiFi issue? By installing a backported kernel module?

eltonwoo, 2015-09-30:
Could you kindly re-forward / redirect my report to the BlueZ developers? This is  somewhat above the level of my linux skills. WRT the wifi issue: _yes_ I have gotten help. (Kindly refer to the lnuxmint forum link that I have provided). Thank you.

respectfully....

cschramm, 2015-09-30:
I did not mean that in the I-dont-care-about-your-issue-as-its-not-mine sense. You can reach the BlueZ project via its users mailing list: http://www.bluez.org/development/lists/ As it's most probably an issue with your third party kernel module, they might not be able and / or willing to help you as well...

If you answer my questions, we can try to find out in what situation you actually are. I did not read the entire forum thread, but to me it looks like you now have a replacement driver module for the chip that got WiFi working but simultaneously broke Bluetooth. So the most fundamental question - which I already asked - is if your system actually detects the bluetooth device or not.

eltonwoo, 2015-09-30:
I didn't mean to infer that you 'didn't care' ... my apologies if the tone of my response led you to believe that. Yes, I am getting help in that forum (if you wish to follow and in the thread, I would certainly appreciate it!). Your hypothesis is correct: I did have bluetooth functioning until _after_ the wifi issue was resolved. In short, one problem fixed, and then another popped up. _LOL_

 My main reason for posting _here_ was to inform the developers of the problem. I feel reasonably certain that I'm not the only linux user ('newbie') who has such a Lenovo machine with that 'incompatible' Realtek / Qualcomm card. 

To save yourself time, a look at the last / most recent page of the thread should bring you up-to-date.

So (asking your advice) should I go ahead and report this in the bluez forum? As indicated, I am already  getting help in the Linux Mint forum. Again, thanks for your kind reply.

cschramm, 2015-09-30:
Hmm... the thread looks like you did not install any module, but just copy a new firmware, probably overwriting another (builtin?) one that made the bluetooth function work before, but not the WiFi function. So obviously the new firmware achieves the opposite. Maybe the ath10k guys can help.

eltonwoo, 2015-10-01:
Well it looks like it. BTW, the thread header has been changed to: "Atheros QCA6164 0168c:0041 ver20 [SOLVED]". How do I report this to the ath10k team? I don't want to go running all over the place and get involved in several discussions, though. The only reason that I created an account was to report the bluetooth problem ... and I landed in the wrong forum (as you indicated it should have been the blueZ team).

I have been told that full support for the card in the Lenovo is supposed to be in the 4.3 kernel. so.... either I wait, or just live with the fact  that I have wifi working (at least). IMVHO, as long as you can connect to the Internet, you can get help with anything. Thus I can live with not having the bluetooth capability (though it _would be nice_ if I had both wifi *and* bluetooth working in LInux. <shrug>

cschramm, 2015-10-01:
I'm not 100 % sure if you did or did not install the backported ath10k driver. https://wireless.wiki.kernel.org/en/users/Drivers/ath10k/firmware clearly states that support for your device needs at least the version from kernel 4.0. Not sure what the changes in 4.3 are about, but I think backports also incorporate development versions, so those changes should already be included.

eltonwoo, 2015-10-01:
I'm at ernel 3.16... and backports was mentioned in the instructions that I was given. The person who was guiding / instructing me in this has not added and further comment to the thread, so I guess I'll have to wait till I get to kernel 4.0....

cschramm, 2015-10-02:
You seem to have installed the backported ath10k driver from Sep 9th [here](http://forums.linuxmint.com/viewtopic.php?p=1070459#p1070459) unless you did another "clean install" of your system afterward, but the WiFi device would not work at all otherwise. So the only change in linux-next that you definitely do not have is [this](https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/drivers/net/wireless/ath/ath10k?id=76d164f582150fd0259ec0fcbc485470bcd8033e).

To me it looks like the firmware that makes your WiFi chip work with the recent driver breaks its bluetooth function. I'd [ask for help at Linux Wireless](https://wireless.wiki.kernel.org/en/users/support) as it may be a generic side effect they may not even be aware of.

closed 2015-10-02 by cschramm

eltonwoo, 2015-10-02:
That post I made on 24 Sept. was _after_ my  last clean install.

This is the first (and probably the last) time that I have a bought a Lenovo. It's because some people told me that the Lenovos are the choice of linux users. If I knew about this "whitelist" thing, and all the headaches involved in getting their machines to work with linux, I wouldn't have wasted my money (and *time*).

 Ok. I'll see if /how to report to the Linux Wireless forum without having to subscribe to yet another list...