blueman-project/blueman #360: Cannot switch between 2 adapters

phaethon, 2015-10-01:
I have two bluetooth adapters. They both show up under menu Adapter. I can see only devices paired to one adapter. If I try to switch to the other adapter window briefly blinks, but still shows devices from the first adapter.
Ubuntu 15.04

infirit, 2015-10-03:
Can you run through the [troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting) page and provide the versions and output of ``blueman-applet`` and ``blueman-manager`` run from the console.

phaethon, 2015-10-05:
As for versions I use standard Ubuntu AMD64 15.04 packages.
blueman, 1.99~alpha1-1ubuntu1
bluez, 4.101-0ubuntu25

phaethon, 2015-10-05:
This is blueman-manager output when trying to change from one device to the other:
```
Loading configuration plugins
Using GConf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['PulseAudioProfile', 'Services'] 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
1 
blueman-manager version 1.99.alpha1 starting
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
2 
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:105)
org.bluez owner changed to  :1.2 
Using GConf config backend
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 771, in Introspect
    reflection_data += self.__class__._reflect_on_method(func)
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 333, in _reflect_on_method
    in_sig = tuple(Signature(func._dbus_in_signature))
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 521, in _signal_func
    def _signal_func(self, message):
TypeError: sighandler() takes no arguments (2 given)

_________
get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 4 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
hci1 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
No such adapter 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
None 
_________
on_adapter_changed (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerToolbar.py:74)
toolbar adapter /org/bluez/964/hci0 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
add_device (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:310)
adding new device 
_________
do_cache (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:489)
Caching new device 00:1B:66:02:1C:F4 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Fake False 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Trusted 1 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Paired 1 
_________
init_services (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:73)
Loading services 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
Sennheiser MM400-X 
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
3 
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
4 
_________
on_pa_ready (/usr/lib/python2.7/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:21)
connected 
_________
<lambda> (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:318)
1 
_________
on_adapter_selected (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerMenu.py:192)
selected /org/bluez/964/hci1 
_________
RemoveDevice (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:420)
<Device object at 0x7fb1b9416f50 (blueman+main+Device+Device at 0x25d5d20)> 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
hci1 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
No such adapter 
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
None 
_________
on_adapter_changed (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerToolbar.py:74)
toolbar adapter /org/bluez/964/hci0 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
add_device (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:310)
adding new device 
_________
do_cache (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:489)
Caching new device 00:1B:66:02:1C:F4 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Fake False 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Trusted 1 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:288)
row update event Paired 1 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/964/hci0/dev_00_1B_66_02_1C_F4 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/964/hci0/dev_00_1B_66_02_1C_F4 
_________
init_services (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:73)
Loading services 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
Sennheiser MM400-X 
```

cschramm, 2015-10-05:
It says it cannot find your hci1 adapter. Please open a python shell and show the result of the following statements:

````python
from blueman.bluez.Manager import Manager
map(lambda adapter: adapter.get_object_path(), Manager().list_adapters())
````

phaethon, 2015-10-05:
```
In [2]: map(lambda adapter: adapter.get_object_path(), Manager().list_adapters())
_________
get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 4 
Out[2]: 
[dbus.ObjectPath('/org/bluez/964/hci0'),
 dbus.ObjectPath('/org/bluez/964/hci1')]
```

closed 2015-10-05 by cschramm

phaethon, 2015-10-05:
This fixes blueman-manager to work with both adapters.

cschramm, 2015-10-05:
:smiley:

commit d133e19b (2015-10-05): Fix adapter detection

The prefix of the object path may be different

Closes #360
commit 98834300 (2015-10-05): Fix adapter detection

The prefix of the object path may be different

Closes #360