blueman-project/blueman #362: Blueman freezes when connecting/disconnecting Sixaxis DS3

zaps166, 2015-10-01:
As in title, Blueman freezes when connecting/disconnecting Sixaxis DS3. I'm using Arch/Manjaro and I tried git version and stable 2.0 version. When I have enabled Bluetooth and I press the "PS" button, the gamepad automatically connects and it works, but blueman-applet and/or blueman-manager always freezes, and I must kill the processes. The same about disconnect - the blueman-manager freezes, but not always.

cschramm, 2015-10-02:
Please provide some output of blueman to the point where it hangs, see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

zaps166, 2015-10-02:
This is the console output of blueman-applet (gamepad is already paired, I'm pressing the "PS" button to connect):

`_________`
`_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)`
`/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1`

It shows one line, sometimes two, three or four lines of this message.
After some time the blueman sometimes starts to respond, but mostly it freezes forever.

cschramm, 2015-10-02:
Both the applet and the manager window freeze? Any output (also from blueman-manager) when it starts to respond again?

zaps166, 2015-10-02:
### These are outputs from blueman-manager (it also starts blueman-applet)

#### Connect gamepad, doesn't freeze:
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
PLAYSTATION(R)3 Controller 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Connected 1 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
PLAYSTATION(R)3 Controller 
_________
monitor_power_levels (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:149)
starting monitor 
_________
level_setup_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:304)
animating up 

#### Freeze blueman-applet after connection, blueman-manager still responds:
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
PLAYSTATION(R)3 Controller 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Connected 1 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
PLAYSTATION(R)3 Controller 
_________
monitor_power_levels (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:149)
starting monitor 
_________
level_setup_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:304)
animating up 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

#### Disconnecting from blueman-manager (blueman-applet doesn't respond now) from context menu (doesn't disconnect at all), freezes for long time and does nothing:
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 158, in on_disconnect
    appl.disconnect_service(service.device.get_object_path(), service.uuid, port)
  File "/usr/lib/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
[... cut ...]
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Connected 0 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
PLAYSTATION(R)3 Controller 
_________
level_setup_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:304)
animating down 
_________
update (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:150)
stopping monitor (not connected)

zaps166, 2015-10-02:
### I've tried many times to pair the gamepad using blueman and it is possible very rare - only when the blueman-applet doesn't freeze.
### Probably the output can differ when I try to do this second time

#### Pairing OK:
_________
_on_interfaces_added (/usr/lib/python3.4/site-packages/blueman/bluez/Manager.py:38)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C 
_________
__init__ (/usr/lib/python3.4/site-packages/blueman/main/Device.py:23)
caching initial properties 
_________
add_device (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:276)
adding new device 
_________
do_cache (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:418)
Caching new device 38:C0:96:5F:13:6C 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Paired 0 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Class 1288 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Class 1288 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Icon input-gaming 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Icon input-gaming 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Connected 1 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Connected 1 
_________
monitor_power_levels (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:149)
starting monitor 
_________
level_setup_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:304)
animating up 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Modalias usb:v054Cp0268d0100 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Modalias usb:v054Cp0268d0100 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)

#### Pairing failed, blueman-applet freezes before notification about accepting device:
_________
_on_interfaces_added (/usr/lib/python3.4/site-packages/blueman/bluez/Manager.py:38)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C 
_________
__init__ (/usr/lib/python3.4/site-packages/blueman/main/Device.py:23)
caching initial properties 
_________
add_device (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:276)
adding new device 
_________
[... 82 lines cut ...]
row update event Modalias usb:v054Cp0268d0100 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C Modalias usb:v054Cp0268d0100 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Modalias usb:v054Cp0268d0100 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
_on_property_changed (/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0/dev_38_C0_96_5F_13_6C UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event UUIDs dbus.Array([dbus.String('00001124-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)

cschramm, 2015-10-03:
If you follow the bluetoothctl commands from the wiki while blueman is running, does it freeze as well? If not, please try again but do not disable the agent in bluetoothctl. blueman should handle the authorization then, but I guess it will freeze as you do not seem to get the authorization screen in any case.

Other things you can try:

Disable the following plugins to see if it keeps the applet from freezing:

* GameControllerWakelock
* NetUsage
* PPPSupport
* RecentConns
* SerialManager
* ShowConnected

Get some debug messages from bluetoothd. Maybe we can spot the event that causes the applet to freeze (the authorization request?). To do this, stop the bluetooth service (make sure blueman, bluetoothctl and similar are not running; they would restart it automatically) and start bluetoothd manually adding the -d option. You will get debugging messages in the syslog (daemon.log if that exists on Arch) then.

infirit, 2015-10-03:
So I have been the same but never really investigated (sorry).

One thing I noticed when running ``blueman-manager`` from a shell is that ``args2`` in the on_success function in ManagerDeviceMenu.py around ~113 is always an empty tuple for me with my mobile and the game controller. Below is a workaround, maybe it is useful for a serial device in which case we can commit the workaround but I have none so I hope @cschramm can look at this :smile:.
```
diff --git a/blueman/gui/manager/ManagerDeviceMenu.py b/blueman/gui/manager/ManagerDeviceMenu.py
index 207d2e1..e979cf3 100644
--- a/blueman/gui/manager/ManagerDeviceMenu.py
+++ b/blueman/gui/manager/ManagerDeviceMenu.py
@@ -103,7 +103,8 @@ class ManagerDeviceMenu(Gtk.Menu):
         device = service.device
 
         def success(*args2):
-            dprint("success", args2)
+            if args2: dprint("success", args2)
+            else: dprint("success")
             prog.message(_("Success!"))
 
             if isinstance(service, SerialPort) and SERIAL_PORT_SVCLASS_ID == uuid128_to_uuid16(service.uuid):
```

After disabling the gamecontrollerwakelock plugin (and after workaround) things work properly again. I think the main reason is that xdg-screensaver never return and ``killall xdg-screensaver`` brings things back to live. The problem is with subprocess not ending but I have some an idea how to fix this. Hopefully I'll be able to post a patch a bit later.

zaps166, 2015-10-03:
Yes, "GameControllerWakelock" has a bug.
I'm using "XResetScreenSaver(Display *display)" in my software to stop the screen saver on X11, maybe this should be better. It should be called in the loop with some interval.

infirit, 2015-10-03:
@zaps166 can you please try https://github.com/infirit/blueman/commit/fbaa92757c4caf88d00e0fb919640d07fe057f96

infirit, 2015-10-03:
@zaps166 can you please try https://github.com/infirit/blueman/commit/bde1b109607bc93832850ece3ad4fa68cd0833a5

ps: ignore the previous message

zaps166, 2015-10-03:
I tried and blueman-applet doesn't freeze, but it creates two xdg-screensaver processes and sleep process (50 sec) inside one of the xdg-screensaver process. When the gamepad is disconnected, the processes dissapears after sleep timeout, but if I do something like this: connect, disconnect, connect, disconnect, etc, I have many sleep and xdg-screensaver processes and all of them dissapears after a timeout when the gamepad is disconnected. Is this correct?
What about "XResetScreenSaver()" in loop? I don't know Python, but this function should reset the screen saver timeout. Can it be called from Python?

infirit, 2015-10-03:
Please test and report back how this pull works for ppl. https://github.com/blueman-project/blueman/pull/363

> two xdg-screensaver processes and sleep process (50 sec) inside one of the xdg-screensaver process. When the gamepad is disconnected, the processes dissapears after sleep timeout, but if I do [...]

This is normal and how ``xdg-screensaver`` works, we call it only twice, once on connect and again on disconnect. It will clean things up once timeout is reached.

>What about "XResetScreenSaver()" in loop? I don't know Python, but this function should reset the screen saver timeout. Can it be called from Python?

One can use ctypes or cython to wrap/call c/cpp functions from python. But I would prefer to stick to the xdg tools for this as it they are supposed to work properly across desktops.

zaps166, 2015-10-03:
OK, but is creates more then one xdg-screensaver process. If you connect the gamepad multiple times in short time, then multiple processes will be open and they remains until gamepads are connected.

I connected and disconnected the gamepad many times in a short time:
```
top -b | grep xdg-screensaver
20444 tomo      20   0   16,4m   2,3m   0,0  0,0   0:00.00 S  `- xdg-screensaver
20562 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
20654 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
20751 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
20844 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
20946 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
21034 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
21192 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
21296 tomo      20   0   16,4m   2,2m   0,0  0,0   0:00.00 S  `- xdg-screensaver
21301 tomo      20   0   16,4m   2,7m   0,0  0,0   0:00.00 S  `- xdg-screensaver
```

infirit, 2015-10-03:
>OK, but is creates more then one xdg-screensaver process.

This is how ``xdg-screensaver`` works and you can look at it [here](http://cgit.freedesktop.org/xdg/xdg-utils/tree/scripts/xdg-screensaver.in). You get the same result when running it multiple times to suspend and resume from a shell. For example, ``xdg-screensaver suspend <winid> && xdg-screensaver suspend <winid>``. I can't change this :smile:.

>they remains until gamepads are connected

To be sure, you meant **dis**connected here, right? If so then as long as all of them go away on disconnect after the timeout we are good, which they do now for me with the changes proposed in #363.

zaps166, 2015-10-03:
`Until` is difficult to use :smile:

If gamepad is connected then process exists - **OK**.
If gamepad is disconnected then process closes after a while - **OK**.
If you do: connect, disconnect and connect gamepad multiple times in a short time then many processes exists and all of them are open when gamepad is connected - **not OK**, because there are extra processes.
When I connect two gamepads then no extra process is created - **OK**.
When I disconnect one of two gamepads then process disappear - **not OK**, because one gamepad is still connected.

infirit, 2015-10-03:
>If gamepad is connected then process exists - OK.
>If gamepad is disconnected then process closes after a while - OK.

Good
>If you do: connect, disconnect and connect gamepad multiple times in a short time then many processes exists and all of them are open when gamepad is connected - not OK, because there are extra processes.

I do not see how I can fix that in blueman. We call ``xdg-screensaver`` once to suspend and once to resume. Then repeat for a next connect->disconnect.

What ``xdg-screensaver`` does is create a tempfile and checks it every 50 seconds. The problem is that there is no randomness (like timestamp) in the created tempfile so while the **resumed** command sleeps one can startup new processes and both will see the tmpfile and keep running. IMHO this is a design flaw/bug in ``xdg-screensaver``.
>When I connect two gamepads then no extra process is created - OK.

Good.
>When I disconnect one of two gamepads then process disappear - not OK, because one gamepad is still connected.

Ok, this is not right. Looking at the code it is a bit icky and imo overly complicated. I'll have a look later and add to the afore mentioned pull..

zaps166, 2015-10-03:
Ok!

infirit, 2015-10-03:
I pushed the new [commit](https://github.com/infirit/blueman/commit/47ebc4d1d4039fd53306dd4c9d0859e95084d2ba) to  #363. I think I have all bases covered now in a much simpler way. Can you please try it out and keep track of the number of locks that ``blueman-applet`` now prints to stdout when run from a shell.

zaps166, 2015-10-03:
Number of locks: 0
Number of locks: 1 - one gamepad
Number of locks: 2 - two gamepats
Number of locks: 1 - one gamepad
Number of locks: 0
It looks OK, but `_on_property_changed` shows me now 32 times :smile: and after connect/disconnect the number of `_on_property_changed` still increases! Now is 36 times.
And second issue: connect gamepad (the led on the gamepad is ON), then disable bluetooth and the gamepad is not automatically disconnected - the led remains ON for a few minutes.

infirit, 2015-10-03:
>``but _on_property_changed``shows me now 32 times

That is quite normal, I see the same.

>then disable bluetooth and the gamepad is not automatically disconnected - the led remains ON for a few minutes.

Sound like this is not related to blueman freezing or the misbehaving plugin?

I'll push the commits to master in a bit.

@cschramm, please can you look into the ``args2`` being an empty tuple? I don't have any serial device to check if that makes a difference. The relevant comment is [this](https://github.com/blueman-project/blueman/issues/362#issuecomment-145230188)

zaps166, 2015-10-03:
> Sound like this is not related to blueman freezing or the misbehaving plugin?

No, it's not related to #362 and it does not work with my USB key on PC, on laptop is OK.
Thanks for Bugfix!

cschramm, 2015-10-04:
@infirit Yes, connecting a serial port service args2 could e.g. be `(dbus.String(u'/dev/rfcomm1'),)`. But what's the problem there? `dprint("success", ())` should not lead to problems. Does it?

infirit, 2015-10-04:
It does as dprint tries to use the values inside the tuple when creating the string, see below. I can also change dprint ro skip this bit
```python
set_op (/usr/lib64/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:85)
op: regenerating instance <ManagerDeviceMenu object at 0x7f81f580b318 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x233acd0)> 
/usr/lib64/python3.4/site-packages/gi/overrides/GObject.py:541: Warning: /var/tmp/portage/dev-libs/glib-2.44.1/work/glib-2.44.1/gobject/gsignal.c:2569: instance '0x233acd0' has no handler with id '1279'
  return func(*args, **kwargs)
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib64/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 114, in success
    dprint("success", args2)
  File "/usr/lib64/python3.4/site-packages/blueman/Functions.py", line 68, in dprint
    s += ("%s " % a)
TypeError: not enough arguments for format string
```

infirit, 2015-10-04:
Assuming we always get an iterable with strings what this? Works for me.
```diff
diff --git a/blueman/gui/manager/ManagerDeviceMenu.py b/blueman/gui/manager/ManagerDeviceMenu.py
index 8f2f79f..e5960d0 100644
--- a/blueman/gui/manager/ManagerDeviceMenu.py
+++ b/blueman/gui/manager/ManagerDeviceMenu.py
@@ -111,7 +111,7 @@ class ManagerDeviceMenu(Gtk.Menu):
         device = service.device
 
         def success(*args2):
-            dprint("success", args2)
+            dprint("success", " ".join(args2))
             prog.message(_("Success!"))
 
             if isinstance(service, SerialPort) and SERIAL_PORT_SVCLASS_ID == uuid128_to_uuid16(service.uuid):
```

cschramm, 2015-10-04:
I see. Yes, the solution makes sense to me.

cschramm, 2015-10-04:
We could also change the line in the dprint function to `s += ("%s " % str(a))`. I think that fits better what was meant there.

infirit, 2015-10-04:
The problem with calling ``str`` it gives bytes under python2 and codepoints under python3. And mixing codepoints and bytes under python2 is problematic (encode/decode errors).

I pushed the patch as I put them in my comment.

commit fc7bc968 (2015-10-03): GameControllerWakelock: Replace subprocess with our Gio launch function

Fixes #362
commit f7328c61 (2015-12-10): GameControllerWakelock: Replace subprocess with our Gio launch function

Fixes #362
closed 2015-10-04 by infirit