blueman-project/blueman #366: Cannot pair with legacy BT device who doesn't support SSP

tiancj, 2015-10-09:
run bluetoothd and blueman-applet via cli, sudo bluetoothd -dddd -C -E -n, and blueman-applet.
the logs of bluez are as follows:
bluetoothd[15667]: src/adapter.c:pin_code_request_callback() hci1 08:00:00:35:8E:DD
bluetoothd[15667]: src/device.c:device_create() dst 08:00:00:35:8E:DD
bluetoothd[15667]: src/device.c:device_new() address 08:00:00:35:8E:DD
bluetoothd[15667]: src/device.c:device_new() Creating device /org/bluez/hci1/dev_08_00_00_35_8E_DD
bluetoothd[15667]: src/device.c:new_auth() Requesting agent authentication for 08:00:00:35:8E:DD
bluetoothd[15667]: src/agent.c:agent_ref() 0x894890: ref=2
bluetoothd[15667]: src/agent.c:agent_ref() 0x894890: ref=3
bluetoothd[15667]: Agent /org/blueman/agent/bluez_agent replied with an error: org.freedesktop.DBus.Python.KeyError, Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/local/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py", line 170, in RequestPinCode
    self.ask_passkey(device, dialog_msg, notify_msg, False, True, ok, err)
  File "/usr/local/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py", line 128, in ask_passkey
    alias = self.get_device_alias(device_path)
  File "/usr/local/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py", line 101, in get_device_alias
    name = props["Name"]
KeyError: u'Name'
bluetoothd[15667]: src/adapter.c:btd_adapter_pincode_reply() hci1 addr 08:00:00:35:8E:DD pinlen 0
bluetoothd[15667]: src/agent.c:agent_unref() 0x894890: ref=2
bluetoothd[15667]: src/agent.c:agent_unref() 0x894890: ref=1
bluetoothd[15667]: src/adapter.c:connect_failed_callback() hci1 08:00:00:35:8E:DD status 5
bluetoothd[15667]: plugins/policy.c:conn_fail_cb() status 5
bluetoothd[15667]: src/adapter.c:bonding_attempt_complete() hci1 bdaddr 08:00:00:35:8E:DD type 0 status 0x5
bluetoothd[15667]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
bluetoothd[15667]: src/device.c:device_cancel_authentication() Canceling authentication request for 08:00:00:35:8E:DD

So, if we change name = props["Name"] to name = props.get("Name"), this error doesn't occur. And the logs of BlueZ are:

bluetoothd[16377]: Agent /org/blueman/agent/bluez_agent replied with an error: org.freedesktop.DBus.Python.TypeError, Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/local/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py", line 173, in RequestPinCode
    self.dialog.present_with_time(self.time_func())
TypeError: 'int' object is not callable

tiancj, 2015-10-09:
basic information:
blueman: 2.0
BlueZ: 5.34
Distribution: Ubuntu 14.04,

cschramm, 2015-10-14:
I added fixes in #368.

We were not yet prepared to not find a Name property. I added a fall-back to the Address property.

The time_func stuff was just wrong (since 1.99.alpha2), but I think it does no harm as `present` is called before, so it does not matter if the `present_with_time` call actually happens.

closed 2015-10-14 by tiancj