blueman-project/blueman #370: Question: which command in terminal to enable serial connection with bt-gps, instead of GUI

marksev1, 2015-10-18:
Hi, so I'm sending NMEA sentences from my phone's gps to the netbook via bluetooth.

In Blueman I always press setup on the nexus5 and then I choose serial and then forward and it opens the serial connection /dev/rfcomm1 I think and everyting works. I'm getting NMEA sentences there.

Now my question is how I could do this setup part from the terminal same as when I choose serial via the GUI of Blueman. I've found that there is some "bluetooth-assistent" but I didn't find any additional options there to choose serial.

Thanks!

cschramm, 2015-10-19:
blueman creates an rfcomm device using libbluetooth, see [here](https://github.com/blueman-project/blueman/blob/3189ea39022d7d46bad95c7d70c04fd36eeb9562/module/libblueman.c#L329). Additionally, the blueman-rfcomm-watcher tools keeps the device opened so that it does not vanish due to inactivity.

BlueZ ships a tool called `rfcomm` you can use to do the same on the command line (on some distributions this may be in a separate package).

marksev1, 2015-10-20:
Thanks! So "sudo rfcomm connect 0" should prolly do it? For /dev/rfcomm0. I'm not quite sure what the difference between rfcomm connect, and rfcomm listen are?

cschramm, 2015-10-20:
It's basically a client-server scheme. You listen for connections on the server side and connect to a listening server from the client side.

closed 2015-10-21 by cschramm