blueman-project/blueman #376: Unable to find phone

thomasnaser, 2015-10-23:
Just installed Ubuntu 15.10 and still unable to connect to my Android phone - Dual booting with Windows XP (yeah I know)which has no problem. The chip is the Broadcom BCM43142.
Lspci shows that the chip is there and running, the kernel (4.2.0-16-generic)
dump that the chip is there and working, but I still can't connect in either direction.
The phone will connects when I run Windows but it doesn't find the same chip when I try in Ubuntu.

infirit, 2015-10-23:
Please see https://github.com/blueman-project/blueman/wiki/Troubleshooting and report back.

yars068, 2015-10-23:
@thomasnaser, please try with right-click on blueman tray icon and select "Add new device..." option, then follow instructions in the appeared dialog (that is blueman-assistant). 
@infirit, adding with this way works fine for me, but with blueman-manager (which will appeared by left-click) I have same problem. Please see [log](http://pastebin.com/WWcXyiAC). I just removed my phone in the blueman-manager, then on the phone, turned on visible mode, and clicked "Search" button... New devices are not appears.

yars068, 2015-10-24:
After e5e68ed, blueman-manager works fine now for me.

infirit, 2015-10-28:
It would be nice to know if version 2.0.1  [1] is working for you @yars068.

[1] https://github.com/blueman-project/blueman/releases/tag/2.0.1

closed 2015-11-03 by cschramm