blueman-project/blueman #377: blueman-adapters: New adapters fail to show

infirit, 2015-10-23:
Does this work for others? Removing seems to be ok.

infirit, 2015-10-23:
Ok, I tracked it down to [``on_adapter_added``](https://github.com/blueman-project/blueman/blob/master/apps/blueman-adapters#L85) which never calls ``add_to_notebook``.  It seems the local connected ``on_property_changed`` never gets the signal, thus failing to add. I can make it work with the below patch but I am not too familiar with the Bluez.Manager signals so I am not sure this is the right way to go :)

```diff
diff --git a/apps/blueman-adapters b/apps/blueman-adapters
index 8be339b..5e5ed3b 100755
--- a/apps/blueman-adapters
+++ b/apps/blueman-adapters
@@ -60,7 +60,8 @@ class BluemanAdapters:
             self.manager.connect_signal('adapter-added', self.on_adapter_added)
             adapters = self.manager.list_adapters()
             for adapter in adapters:
-                self.add_to_notebook(adapter)
+                path = adapter.get_object_path()
+                self.on_adapter_added(self.manager, path)
         except Exception as e:
             print(e)
             self.manager = None
@@ -83,13 +84,8 @@ class BluemanAdapters:
         Gtk.main_quit()
 
     def on_adapter_added(self, _manager, adapter_path):
-        def on_property_changed(_adapter, name, value, _path):
-            if name == 'Powered' and value:
-                adapter.disconnect_signal(sig)
-                self.add_to_notebook(adapter)
-
         adapter = Bluez.Adapter(adapter_path)
-        sig = adapter.connect_signal('property-changed', on_property_changed)
+        self.add_to_notebook(adapter)
 
     def on_dbus_name_owner_change(self, owner):
         print('org.bluez owner changed to '+owner)
```

yars068, 2015-10-23:
Unfortunatelly, I cannot check it, because I have a only one adapter :(

cschramm, 2015-10-24:
To me it looks like the current code adds all adapters on start and additional adapters as soon as they get powered. We have to check why the signal does not appear. It should come from `blueman.bluez.manager._on_interfaces_added`.

What seems strange to me is that initial adapters are added in any case while new adapters are required to be powered first. Not sure why that's the case...

infirit, 2015-10-25:
>What seems strange to me is that initial adapters are added in any case while new adapters are required to be powered first

Yes, removing the ``Powered `` restriction in ``on_adapter_added`` will make the new adapter tabs show up. I traced it to https://github.com/blueman-project/blueman/commit/c9b08dcf0ee4d995404c9cbf9d09fc1bd379260a but the commit message does not mention why. If not objection I will push the below.

````diff
diff --git a/apps/blueman-adapters b/apps/blueman-adapters
index 8be339b..67ec4bc 100755
--- a/apps/blueman-adapters
+++ b/apps/blueman-adapters
@@ -84,9 +84,8 @@ class BluemanAdapters:
 
     def on_adapter_added(self, _manager, adapter_path):
         def on_property_changed(_adapter, name, value, _path):
-            if name == 'Powered' and value:
-                adapter.disconnect_signal(sig)
-                self.add_to_notebook(adapter)
+            adapter.disconnect_signal(sig)
+            self.add_to_notebook(adapter)
 
         adapter = Bluez.Adapter(adapter_path)
         sig = adapter.connect_signal('property-changed', on_property_changed)
````

cschramm, 2015-10-25:
That makes no sense to me. Why wait for any (!) property change before adding it?

I link the second junk of your initial diff much better.

But in any case, I'd really like to know why the signal does not show up (maybe it just doesn't in case the adapter is already powered initially). If you do not look into that, please keep this issue open and I will as soon as I'm back (will be away next week).

infirit, 2015-10-25:
>That makes no sense to me. Why wait for any (!) property change before adding it?

Looking at the log when I remove this is that I see the actions (hidden/visible/temp) being applied twice. My guess is that this connect/disconnect is working around this.  I think this can be done in another (easier) way so I'll try that out before pushing anything.

>why the signal does not show up

I think we get more than one, see above, but I am not sure why atm.

>I link the second junk of your initial diff much better.

lolz auto-correct strikes again :stuck_out_tongue_winking_eye:

cschramm, 2015-10-26:
:laughing: You know what I meant.

So you say adapter added is called twice? Hrm...

infirit, 2015-10-26:
I am not sure, I saw the ``DiscoverableTimeout`` and ``Discoverable`` being set twice. My guess was we either connect to the signal or get the signal more than once.

However now that I look again I am not sure my guess was correct. I'll look at it some more this week and also post some logs.

infirit, 2015-10-26:
Ok, I pushed a couple commits to fix the problem of new devices not showing. But..

I was able to reproduce why I saw the properties set more than once, it happens when I remove and then re-add a device. The property are set the same number of times I re-add the device. Logs [here](https://gist.github.com/infirit/901c6f67cc7bf4f77ed2). No clue why :confused:..

infirit, 2015-10-27:
More details, I can reproduce it with [this script](https://gist.github.com/infirit/7e3bbd92f63e4e01119d) so it look like the problem goes into the bluez modules. I think it may be better to move this to a new issue if not let me know :smile:.

infirit, 2015-10-27:
Ok, I can fix it by explicitly disconnecting the ``property-changed`` signal when the adapter is removed. See my [example script](https://gist.github.com/infirit/58bfcba5e30f5e1ae48c). I think the Adapter/Manager class should disconnect any signals for me on adapter removal, or am I wrong?

infirit, 2015-10-31:
>The property are set the same number of times I re-add the device.

So we allow more than one Adapter object to be created for the same hci device. This means if something still holds a ref to the old object we may end up with more than one. I suspect this is what happens and by making sure we always use the same object this problem goes away.

We should think about turning it into a singleton where we always return the same object for a hci device regardless of how many times we called ``Bluez.Adapter(path)``. Same applies for ``Bluez.Manager`` I guess.

Closing this one and propose further discussion to continue on https://github.com/blueman-project/blueman/pull/380

closed 2015-10-31 by infirit