blueman-project/blueman #381: My Bluetooth hardware is not detected --- Using Korora 22 in my Dell Inspiron 1464

benausti, 2015-11-02:
Hi,

My bluetooth hardware is not getting detected. I am using Korora 22 in my Dell Inspiron 1464. Screenshot is below. Please advise how to fix it.
[image]

infirit, 2015-11-02:
Can you please go through https://github.com/blueman-project/blueman/wiki/Troubleshooting

And report back with the results/required info?

yars068, 2015-11-02:
Mostly, that means, your hardware does not detected by kernel (its BlueZ stack), so you need driver which supports your BT-adapter. Please see the output of `lspci` or `lsusb` and `dmesg` commands to know it.

benausti, 2015-11-03:
when i run the commands i got the following output. Please help me identifying the issue:

15:33 Austine-Kor@localhost ~ $ su
Password: 
15:33 root@localhost Austine-Kor # lspci
00:00.0 Host bridge: Intel Corporation Core Processor DRAM Controller (rev 12)
00:01.0 PCI bridge: Intel Corporation Core Processor PCI Express x16 Root Port (rev 12)
00:16.0 Communication controller: Intel Corporation 5 Series/3400 Series Chipset HECI Controller (rev 06)
00:1a.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 06)
00:1b.0 Audio device: Intel Corporation 5 Series/3400 Series Chipset High Definition Audio (rev 06)
00:1c.0 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 (rev 06)
00:1c.1 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 2 (rev 06)
00:1c.5 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 6 (rev 06)
00:1d.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 06)
00:1e.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev a6)
00:1f.0 ISA bridge: Intel Corporation HM55 Chipset LPC Interface Controller (rev 06)
00:1f.2 SATA controller: Intel Corporation 5 Series/3400 Series Chipset 4 port SATA AHCI Controller (rev 06)
00:1f.3 SMBus: Intel Corporation 5 Series/3400 Series Chipset SMBus Controller (rev 06)
02:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV710/M92 [Mobility Radeon HD 4330/4350/4550]
02:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] RV710/730 HDMI Audio [Radeon HD 4000 series]
04:00.0 Network controller: Broadcom Corporation BCM4312 802.11b/g LP-PHY (rev 01)
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8101/2/6E PCI Express Fast/Gigabit Ethernet controller (rev 02)
ff:00.0 Host bridge: Intel Corporation Core Processor QuickPath Architecture Generic Non-core Registers (rev 02)
ff:00.1 Host bridge: Intel Corporation Core Processor QuickPath Architecture System Address Decoder (rev 02)
ff:02.0 Host bridge: Intel Corporation Core Processor QPI Link 0 (rev 02)
ff:02.1 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor QPI Physical 0 (rev 02)
ff:02.2 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 02)
ff:02.3 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 02)
15:34 root@localhost Austine-Kor # lsusb
Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0c45:641d Microdia 1.3 MPixel Integrated Webcam
Bus 001 Device 006: ID 413c:8162 Dell Computer Corp. Integrated Touchpad [Synaptics]
Bus 001 Device 005: ID 413c:8161 Dell Computer Corp. Integrated Keyboard
Bus 001 Device 003: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 001 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
15:35 root@localhost Austine-Kor # dmesg
[    0.000000] microcode: CPU0 microcode updated early to revision 0xe, date = 2013-06-26
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.2.3-200.fc22.x86_64 (mockbuild@bkernel01.phx2.fedoraproject.org) (gcc version 5.1.1 20150618 (Red Hat 5.1.1-4) (GCC) ) #1 SMP Thu Oct 8 03:23:55 UTC 2015
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.2.3-200.fc22.x86_64 root=/dev/mapper/korora-root ro rd.lvm.lv=korora/root rd.lvm.lv=korora/swap rhgb quiet LANG=en_US.UTF-8
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'lazy' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000d2000-0x00000000000d3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bf27bfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf27c000-0x00000000bf281fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf282000-0x00000000bf3e5fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf3e6000-0x00000000bf40efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf40f000-0x00000000bf46efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf46f000-0x00000000bf46ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf470000-0x00000000bf4f0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bf4f1000-0x00000000bf70efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf70f000-0x00000000bf716fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf717000-0x00000000bf71efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf71f000-0x00000000bf77dfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf77e000-0x00000000bf79efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bf79f000-0x00000000bf7e1fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf7e2000-0x00000000bf7fefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bf7ff000-0x00000000bf7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf800000-0x00000000bfffffff] reserved
[... cut ...]
[   76.134142] Bluetooth: HCI device and connection manager initialized
[   76.134147] Bluetooth: HCI socket layer initialized
[   76.134150] Bluetooth: L2CAP socket layer initialized
[   76.134160] Bluetooth: SCO socket layer initialized
[   76.208891] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   76.208896] Bluetooth: BNEP filters: protocol multicast
[   76.208902] Bluetooth: BNEP socket layer initialized
[ 1481.440611] perf interrupt took too long (2505 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
15:35 root@localhost Austine-Kor #

infirit, 2015-11-03:
From what I can gather ``BCM2046`` should work in bluez >4.66, but this is broadcom so who knows..

@benausti, still waiting for the info described https://github.com/blueman-project/blueman/wiki/Troubleshooting. If the device is not detected with the command described [here](https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth) there is nothing we can do from blueman :smile:

yars068, 2015-11-03:
Possible, the problem is in firmware, which is too old, and it may doesn't support your combined device. This device is managed by b43 driver, which requires a proper firmware.

closed 2015-11-26 by cschramm